lvgl/lvgl #799: Adding chart area

ali-rostami, 2019-02-06:
Hi, with respect to this [@kisvegabor 's comment](https://github.com/littlevgl/lvgl/issues/340#issuecomment-460853736), I opened this issue.
This is part of what he said in that comment:
>"area chart" is not supported right now but it can be a new char type. e.g. `LV_CHART_TYPE_AREA`. It can use `lv_draw_triangle` to draw the areas.

so, @kisvegabor  would you please explain more. why using ```lv_draw_triangle```?

kisvegabor, 2019-02-06:
I can imagine that the areas are built from triangles:
[image]

All this can be done in a function similar to `lv_chart_draw_lines`: https://github.com/littlevgl/lvgl/blob/master/lv_objx/lv_chart.c#L600

ali-rostami, 2019-02-07:
Hmm, now I got it.
I will work on it soon. thanks.

ali-rostami, 2019-02-15:
yeah, It's working. I will send a PR soon.

kisvegabor, 2019-03-02:
@ali-rostami 
Is there any new here?

ali-rostami, 2019-03-12:
Hi Gabor
Isn't there any other method? I think it's so slow.
This is how I implemented it. -It's just for test- :
```
static void lv_chart_draw_lines(lv_obj_t * chart, const lv_area_t * mask)
{
	lv_chart_ext_t * ext = lv_obj_get_ext_attr(chart);

	uint16_t i;
	lv_point_t p1;
	lv_point_t p2;
	lv_coord_t w = lv_obj_get_width(chart);
	lv_coord_t h = lv_obj_get_height(chart);
	lv_coord_t x_ofs = chart->coords.x1;
	lv_coord_t y_ofs = chart->coords.y1;
	int32_t y_tmp;
	lv_coord_t p_prev;
	lv_coord_t p_act;
	lv_chart_series_t * ser;
	lv_opa_t opa_scale = lv_obj_get_opa_scale(chart);
	lv_style_t style;
	lv_style_copy(&style, &lv_style_plain);
	style.line.opa = ext->series.opa;
	style.line.width = ext->series.width;

	/*Go through all data lines*/
	LL_READ_BACK(ext->series_ll, ser) {
		style.line.color = ser->color;

		p1.x = 0 + x_ofs;
		p2.x = 0 + x_ofs;

		p_prev = ser->start_point;
		y_tmp = (int32_t)((int32_t) ser->points[p_prev] - ext->ymin) * h;
		y_tmp = y_tmp / (ext->ymax - ext->ymin);
		p2.y = h - y_tmp + y_ofs;

		for(i = 1; i < ext->point_cnt; i ++) {
			p1.x = p2.x;
			p1.y = p2.y;

			p2.x = ((w * i) / (ext->point_cnt - 1)) + x_ofs;

			p_act = (ser->start_point + i) % ext->point_cnt;

			y_tmp = (int32_t)((int32_t) ser->points[p_act] - ext->ymin) * h;
			y_tmp = y_tmp / (ext->ymax - ext->ymin);
			p2.y = h - y_tmp + y_ofs;

			if(ser->points[p_prev] != LV_CHART_POINT_DEF && ser->points[p_act] != LV_CHART_POINT_DEF) {
#if 0
				lv_draw_line(&p1, &p2, mask, &style, opa_scale);
#elif 1
				lv_point_t triangle_points[3];
				triangle_points[0] = p1;
				triangle_points[1].x = p2.x;
				triangle_points[1].y = y_ofs + h;
				triangle_points[2].x = p1.x;
				triangle_points[2].y = y_ofs + h;
				lv_draw_triangle(triangle_points, mask, style.line.color);
				triangle_points[2] = p2;
				lv_draw_triangle(triangle_points, mask, style.line.color);
#endif
			}
			p_prev = p_act;
		}
	}
}
```

kisvegabor, 2019-03-13:
It seems good to me. Why do you think it's slow?

kisvegabor, 2019-03-13:
Looks awesome by the way :)

ali-rostami, 2019-03-13:
you know my reference for measuring speed is the lag I see in my graph.
I have a graph which its y comes from a uart with 120 Hz frequency. I set the width and point counts the same number. when I set the type of chart ```LV_CHART_TYPE_COLUMN``` -which can plot for me the area type too by considering the condition I mentioned- There is no lag and it is plotted perfectly. by using the method above there is a bad lag. so that's why I was wondring if there is another way.
>>Looks awesome by the way :)

thanks :)

anyway if it is ok I will go to the next step wich is based on this issue #818 close #799.

kisvegabor, 2019-03-14:
> I set the width and point counts the same number. 

In this case, triangle drawing will be really slow because it will draw 1px wide horizontal lines. 
However, I can update the triangle drawing function in case of right triangles (one angle is 90°). 

> anyway if it is ok I will go to the next step which is based on this issue #818 close #799.

Yes, you can move forward and I'll add the optimizations as I can. But please send a PR for the area drawing first. (It's easier to overview smaller PRs)

ali-rostami, 2019-03-17:
As my PR is merged I close this issue.
thanks, @kisvegabor for your help

closed 2019-03-17 by ali-rostami

kisvegabor, 2019-03-17:
I thank you too. I'll update triangle drawing as I can.

ali-rostami, 2019-03-18:
Oh, great 👍 
-Just as a suggestion- reopen this issue as a reminder if you prefer.

kisvegabor, 2019-03-18:
I opened a new issue for it: #960