lvgl/lvgl #801: VDB and monochrome displays

C47D, 2019-02-06:
Hi,

As mentioned in #796 I am using an 128*64 OLED display based on the SSD1306 controller, I had been able to compile the project but noticed the VDB buffer is being filled with all 0x01's so all the display pages have the first row light up.

Is each byte if the VDB buffer representing a pixel of the OLED display (or any other monochrome display)?

In case you want to take a look at my hideous code (here)[https://github.com/C47D/lvgl_ssd1306] you can find it.

The flush function:
https://github.com/C47D/lvgl_ssd1306/blob/01096003832aa43c9f8353309e5f9d4b728f6009/hello_world_label.cydsn/main.c#L36

Kind regards

kisvegabor, 2019-02-07:
In `lv_conf.h` you have `LV_VDB_PX_BPP       LV_COLOR_SIZE` and `LV_COLOR_DEPTH   1`.

With `LV_COLOR_DEPTH   1` the VDB will still use 1 byte for 1 pixel.

So need to do something like:
```
if(color_p->full) set_px(x,y, 1);
else  set_px(x,y, 0)
```

Have a look at this as an example: https://github.com/littlevgl/lv_drivers/blob/master/display/ST7565.c#L155

C47D, 2019-02-07:
Thanks for the reply @kisvegabor, I will take a look at the example later today, and return with more questions.

C47D, 2019-02-08:
Hi,

I modified the flush function just to print the pixel coordinates:

```c
void ssd1306_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
    DBG_println("x1: %d, x2: %d, y1: %d, y2: %d", __func__, x1, x2, y1, y2);
    
    /* Return if the area is out the screen */
    if ((x2 < 0) || (y2 < 0) || (x1 > LV_HOR_RES - 1) || (y1 > (LV_VER_RES - 1))) {
        lv_flush_ready();
        return;
    }
        
    bool is_set = color_p->full;
    DBG_println("Pixel is set? %s", is_set ? "true" : "false");
    
    lv_flush_ready();
}
```

And got the following output:
`
..\lvgl\lv_objx\lv_label.c: 133: label created
x1: 31372, x2: 0, y1: 127, y2: 0
Pixel is set? true
x1: 31372, x2: 0, y1: 127, y2: 10
Pixel is set? true
x1: 31372, x2: 0, y1: 127, y2: 20
Pixel is set? true
x1: 31372, x2: 0, y1: 127, y2: 30
Pixel is set? true
x1: 31372, x2: 0, y1: 127, y2: 40
Pixel is set? true
x1: 31372, x2: 0, y1: 127, y2: 50
Pixel is set? true
x1: 31372, x2: 0, y1: 127, y2: 60
Pixel is set? true
`

Having such large pixel numbers in x1 seems correct?

Regards

kisvegabor, 2019-02-08:
> Just one quick question, does the vdb[0] byte always store the pixel x0,y0 content, vdb[1] stores x0,y1 pixel content and so on?

vdb[0]  should go to (x1;y1) (paramters in `disp_flush`) on your display.

> DBG_println("x1: %d, x2: %d, y1: %d, y2: %d", __func__, x1, x2, y1, y2);

You didn't assigne the `__func__` variable
DBG_println("func: %s, x1: %d, x2: %d, y1: %d, y2: %d", __func__, x1, x2, y1, y2);

C47D, 2019-02-08:
> You didn't assigne the `__func__` variable
`DBG_println("func: %s, x1: %d, x2: %d, y1: %d, y2: %d", __func__, x1, x2, y1, y2);`

Thanks for pointing that out, i will fix it later today.

> vdb[0] should go to (x1;y1) (paramters in `disp_flush`) on your display.

I will also test this, please excuse me for all this dumb questions.

kisvegabor, 2019-02-09:
> I will also test this, please excuse me for all this dumb questions.

No worries :)

C47D, 2019-02-11:
So, just a follow up,

I changed my program, now the label placed at the middle of the display should say _lvgl_, the new flush function do something similar to the [ST7565 example]( https://github.com/littlevgl/lv_drivers/blob/master/display/ST7565.c#L155), now the display content is placed in a 'proxy' buffer (`ssd_buffer`) and at the end of the flush function the display is updated.

```c
#define SSD1306_VER_RES     64
#define SSD1306_HOR_RES     128

uint8_t ssd_buffer[SSD1306_COLS * SSD1306_PAGES] = {0};

void ssd_set_pixel(uint8_t x, uint8_t y, bool set)
{
    if (set) {
        ssd_buffer[x + ((y / SSD1306_PAGES) * SSD1306_COLS)] |= 1 << (y & 7);        
    } else {
        ssd_buffer[x + ((y / SSD1306_PAGES) * SSD1306_COLS)] &= ~(1 << (y & 7));
    }
}

void ssd_update(void)
{
    ssd_send_buffer(ssd_buffer, sizeof ssd_buffer);
}

void ssd1306_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
    DBG_println("x1: %d, y1: %d, x2: %d, y2: %d, pixel: %s", x1, y1, x2, y2, color_p->full ? "full" : "empty");
    
    if (0 > x2) {
        return;
    }
    
    if (0 > y2) {
        return;
    }
    
    if (SSD1306_HOR_RES - 1 < x1) {
        return;
    }
    
    if (SSD1306_VER_RES - 1 < y1) {
        return;
    }
    
    uint8_t act_x1 = x1 < 0 ? 0 : x1;
    uint8_t act_y1 = y1 < 0 ? 0 : y1;
    uint8_t act_x2 = x2 > SSD1306_HOR_RES - 1 ? SSD1306_HOR_RES - 1 : x2;
    uint8_t act_y2 = y2 > SSD1306_VER_RES - 1 ? SSD1306_VER_RES - 1 : y2;
    
    for (uint8_t y = act_y1; y <= act_y2; y++ ) {
        for (uint8_t x = act_x1; x <= act_x2; x++) {
            DBG_println("x: %d, y: %d, %s", x, y, color_p->full != 0 ? "set" : "clear");
            if (color_p->full != 0) {
                ssd_set_pixel(x, y, 1);
            } else {
                ssd_set_pixel(x, y, 0);
            }
            
            color_p++;
        }
        
        color_p += 128;
    }
    
    ssd_send_buffer(ssd_buffer, sizeof ssd_buffer);
    lv_flush_ready();
}
```

And the display always end up being like this:
[image]

I'm having a hard time trying to understand how lvgl works, so i need a little bit more time, i will try again next weekend.

All the code is in [1].

[1] [lvgl_ssd1306](https://github.com/C47D/lvgl_ssd1306)

embeddedt, 2019-02-11:
@C47D You said your display is monochrome at the top. Is it supposed to be showing color like that?

C47D, 2019-02-11:
@embeddedt Yes, just 2 pages (16 pixels) of the display are yellow, like [this](https://www.ebay.com/itm/128X64-OLED-LED-Display-Module-For-Arduino-SSD1306-0-96-I2C-IIC-Blue-Yellow-/152159605952?_ul=MX) one.

kisvegabor, 2019-02-12:
`color_p += 128;` is not required because you already have `color_p++;`  one line above which will go through all pixels.

Please verify that your driver works without lvgl. Just try to set some pixels and see if it works.

C47D, 2019-02-15:
Hi,

Sorry for the late update but i had been busy, i tried removing `color_p += 128` but it didn't helped.

> Please verify that your driver works without lvgl. Just try to set some pixels and see if it works.

I did so, i can set, clear and toggle pixels using a coordinate system (0 <= x <= 127 and 0 <= y <= 63).

I may be try lvgl again the next week.

Thanks for the help 👍

kisvegabor, 2019-02-18:
If still not working please post your new `disp_flush` function and and a screenshot.

C47D, 2019-02-18:
Hi,

I will update the github project as soon as i get home.

Regards

- carlos

C47D, 2019-03-01:
Will open a topic in the forum for this issue.

closed 2019-03-01 by C47D

kisvegabor, 2019-03-01:
Awesome, thank you!