lvgl/lvgl #802: LV_COLOR_16_SWAP=1 brakes build

ik-browselabs, 2019-02-07:
Hi,

I get the following error when I try to define LV_COLOR_16_SWAP as 1.

`./lvgl/lv_objx/lv_canvas.c: In function ‘lv_canvas_mult_buf’:
./lvgl/lv_objx/lv_canvas.c:344:33: error: ‘lv_color_t {aka union <anonymous>}’ has no member named ‘green’; did you mean ‘green_h’?
             canvas_buf_color[j].green = (uint16_t) ((uint16_t) canvas_buf_color[j].green * copy_buf_color[j].green) >> 8;
                                 ^~~~~
                                 green_h
./lvgl/lv_objx/lv_canvas.c:344:84: error: ‘lv_color_t {aka union <anonymous>}’ has no member named ‘green’; did you mean ‘green_h’?
             canvas_buf_color[j].green = (uint16_t) ((uint16_t) canvas_buf_color[j].green * copy_buf_color[j].green) >> 8;
                                                                                    ^~~~~
                                                                                    green_h
./lvgl/lv_objx/lv_canvas.c:344:110: error: ‘lv_color_t {aka union <anonymous>}’ has no member named ‘green’; did you mean ‘green_h’?
             canvas_buf_color[j].green = (uint16_t) ((uint16_t) canvas_buf_color[j].green * copy_buf_color[j].green) >> 8;
                                                                                                              ^~~~~
                                                                                                              green_h
`

I get the error when I do the clean build, i.e.:
- just clone the repo
- copy lv_conf_templ.h to lv_conf.h
- enable first ifdef
- change LV_COLOR_16_SWAP from 0 to 1
- and try a clean build.

If I get LV_COLOR_16_SWAP back to 0, then everything builds fine.

Best regards

embeddedt, 2019-02-07:
@ikolak I tried this just now with the stable `master` branch and it worked fine. Can you update and see if that fixes your problem?

ikolak, 2019-02-07:
@embeddedt, thanks for the reply.

I've tested it once more - but the error is still here.

Here are my steps:
`ikolak@box:~$ mkdir test_lvgl`
`ikolak@box:~$ cd test_lvgl/`
`ikolak@box:~/test_lvgl$ git clone https://github.com/littlevgl/lvgl.git`

> Cloning into 'lvgl'...
> remote: Enumerating objects: 5, done.
> remote: Counting objects: 100% (5/5), done.
> remote: Compressing objects: 100% (4/4), done.
> remote: Total 12995 (delta 0), reused 3 (delta 0), pack-reused 12990
> Receiving objects: 100% (12995/12995), 10.77 MiB | 1.99 MiB/s, done.
> Resolving deltas: 100% (10209/10209), done.

`ikolak@box:~/test_lvgl$ cp lvgl/lv_conf_templ.h ./lv_conf.h`

Enable first #ifdef so lv_conf.h is used in build 
`ikolak@box:~/test_lvgl$ sed -i '/Set it to "1" to enable content/c\#if 1' ./lv_conf.h`

If I try to compile problematic file now - it goes smoothly
`ikolak@box:~/test_lvgl$ gcc -c lvgl/lv_objx/lv_canvas.c`

Buf if I change LV_COLOR_16_SWAP to 1
`ikolak@box:~/test_lvgl$ sed -i '/LV_COLOR_16_SWAP/c\#define LV_COLOR_16_SWAP 1' ./lv_conf.h`

then I get the compile error after issuing
`ikolak@box:~/test_lvgl$ gcc -c lvgl/lv_objx/lv_canvas.c`

If I get it back to 0, compile is ok.

Can you please test this like this?

Thanks in advance

kisvegabor, 2019-02-07:
Hi,

`LV_COLOR-16_SWAP` wasn't really handled correctly. Thank you for pointing this out.

I pushed the fix to `master`.

Is it ok now?

ikolak, 2019-02-07:
Hi @kisvegabor 

Compile time is ok now. 

I haven't tested it yet, but I don't see you handling .green color when `LV_COLOR_16_SWAP = 1` - didn't go into the code deeply; I'll test and report you results.

Best regards

embeddedt, 2019-02-07:
@ikolak When swapping a 16-bit color, there's no need to touch the green part.

Memory layout before swap: `RRRRRGGGGGGBBBBB`
Memory layout after    swap: `BBBBBGGGGGGRRRRR`

As you can see, the green part of the color can remain unchanged. Is that what you're referring to?

ikolak, 2019-02-07:
@embeddedt yes, but in the code, if `LV_COLOR_DEPTH == 16` and `LV_COLOR_16_SWAP == 1` then `canvas_buf_color[j].green` never gets populated, right?

kisvegabor, 2019-02-07:
@embeddedt Not really:
Before swap: `RRRRRGGG` `GGGBBBBB`
After swap:  `GGGBBBBB` `RRRRRGGG` 

It's important when you send 16-bit data on a 8-bit interface. This case the LSB will be sent first (Little endian byte order) but the displays usually wait for the MSB first.

`canvas_buf_color[j].green` is never populated with `LV_COLOR_16_SWAP == 1`. Instead `green_h` and `green_l` is populated.

embeddedt, 2019-02-07:
@kisvegabor Oops, I misunderstood the meaning of swap in this case. Thanks for the clarification!

ikolak, 2019-02-07:
@kisvegabor - got it now, I've checked the code - union is under the hood so it gets (`green_h` and `green_l`) populated.

Anyhow, I'll check it later today and inform you. Thanks.

ikolak, 2019-02-07:
@kisvegabor it's working now, thanks!

closed 2019-02-07 by ikolak