lvgl/lvgl #803: Label with static buffer - memory management issues

rrroonn, 2019-02-07:
Hi,

I am using a static text buffer (buff) to hold the underlying data for a label object.

using: `lv_label_set_static_text()` and initially padding (buff) with spaces to expand the label to its max size.

The label object seems to assume that setting a label's text is only done once, because any other calls that change the text cause the object to try to  attempt to free up the memory - even though it was not allocated by lvgl.

That would be ok, but changing the text in (buff) and calling 

`lv_obj_invalidate()`

doesn't cause the label to be refreshed with the new data.

Again, maybe its just an inappropriate use of a label.

embeddedt, 2019-02-07:
@rrroonn Try calling `lv_label_set_static_text` again instead of `lv_obj_invalidate`.

kisvegabor, 2019-02-07:
It should also work: 
```c
static char buf[] = {"abc             "};
lv_obj_t * label;

static void label_task(void * p)
{
    buf[2]++;
    if(buf[2] > 'z') buf[2] = 'a';
    lv_obj_invalidate(label);
}

    label = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_static_text(label, buf);

    lv_task_create(label_task, 300, LV_TASK_PRIO_MID, NULL);
```

But without `lv_label_set_static_text` the label's size is not updated. So it can happen that the last character will wrap and go out of the label.

So as @embeddedt suggested it''s better to call `lv_label_set_static_text`.

closed 2019-02-07 by rrroonn