lvgl/lvgl #804: revisit themes

puzrin, 2019-02-07:
I suggest to make built-in themes more close to some of well known UI guidelines. That should take much less efforts than polish existing ones.

Styles to consider:

1. [Material Design](https://material.io/) by Google.
2. [UX/UI Guidelines for Windows Phone 8](https://blogs.msdn.microsoft.com/africaapps/2014/03/08/uxui-guidelines-for-windows-phone-8/)
   - Very simple and nice. Especially nice for small monochrome oled-s.
   - More links: [1](https://www.slideshare.net/NeelimaSalvi/windows-phone-ui-design-guidelines)
   - Note, Metro was abandoned in favour of [Fluent](https://www.microsoft.com/design/fluent/). But imho it's still nice for small displays.

---

What is expected:

1. Check technical limitations (styles may require complex animations, difficult to code manually).
2. Tweak default styles for existing components.
3. Add references into existing themes doc.
4. Tweak main demo to follow UI guidelines.

kisvegabor, 2019-02-07:
Thanks for the links!

With [lv_theme_material](https://littlevgl.com/demo-theme-material) my intention was to "copy" Goolges material design.

But there is no "Windows Phone" theme right now.

If you have ideas to improve the existing themes please send a PR. 

> 1. Check technical limitations (styles may require complex animations, difficult to code manually)

We can add an `anim` part to the themes with `anim_in_left`, `anim_out_bottom` etc predefined animations. Here I wrote about how the extend the themes: #806

> 2. Tweak default styles for existing components.

By "default" do you mean without themes?

seyyah, 2019-02-07:
I would like to present a small additional personal opinion. ngx-admin (Nebular) theme is very nice/popular/open source dashboard for IoT [1, 2]. This resembles LittlevGL' Nemo theme.

In addition to Flat UI Color Palette is very nice [3]. We often use it as a default color palette in our web projects.

## :camera: Screenshots
Nebular dashboard
[image]

Flat UI Color
[image]

## :clipboard: Reference
1. http://akveo.com/ngx-admin/pages/iot-dashboard
2. https://github.com/akveo/nebular
3. https://flatuicolors.com/palette/defo

kisvegabor, 2019-02-08:
@seyyah Thank you!

It would be nice to have a nice collection of ideas and references. 
Hopefully, somebody will be interested in improving the current theme(s).

But we should be careful because if 5 new themes would be added then 5 more themes needed to be maintained when a new object type comes. So it would be better to have a few but high-quality themes.

puzrin, 2019-02-08:
A quick summary "what is wrong with current material theme"

1. First, and most important. Demo's elements layout is random. Align/spacing does not follow MD guidelines. It's very confusing, even if components are right.
2. Elemetns
   - [Buttons](https://material-components.github.io/material-components-web-catalog/#/component/button): secondary buttons a usually without background.
   - [Slider](https://material-components.github.io/material-components-web-catalog/#/component/slider): style mismatch at all (custom used)
   - [Tabs](https://material-components.github.io/material-components-web-catalog/#/component/tabs): background & bottom shadow not needed
   - Other: need recheck
   - Keyboard: everything not like in mobiles :)
3. MD uses 12 [or 8] colors everywhere. Those variables should be defined somehow for fast palette switch. User should not tweak element's color explicit.
4. Fonts. Several font styles/sizes preferable. See MD components demos.
5. Interaction. [Riples](https://material-components.github.io/material-components-web-catalog/#/component/ripple) would be nice

In other words, expected approach is ~ like this:

1. Select UI guideline to follow in default demos/elements (IMHO [Material Design](http://material.io/) is nice)
2. Provide palettes/themes

Now we have (2) without (1). That's the most exact description instead of explaining zillions of small differences.

kisvegabor, 2019-02-09:
Thank you for the list!

puzrin, 2019-03-09:
open please

puzrin, 2019-03-15:
After diving into settings menu, found a small thing, doable "right now" with minimal efforts. I suggest switch to 8-colors palette, like in Material Design.

What that means:

- Define 8 variables (names like in MD)
- Use those in all themes or in one candidate to "next modern theme".

Or, as alternative - announce current themes as legacy (for compatibility) and create new palette-based theme with 3 alternate palettes.

embeddedt, 2019-03-15:
> announce current themes as legacy (for compatibility)

Check with @kisvegabor, but I'm not sure this is necessary. Since it's `v6.0` we could remove the old theme system and replace it with what you are suggesting.

At the same time, of course, we should try to reduce the migration workload on the user.

puzrin, 2019-03-15:
If themes can be dropped at all - that would be ideal. I don't know your current sutuation and try to avoid too aggressive suggestions, when i'm not ready to participate myself.

I think, when someone start make settings menu, it become more clear what to do with current themes.

kisvegabor, 2019-03-17:
> After diving into settings menu, found a small thing, doable "right now" with minimal efforts. I suggest switch to 8-colors palette, like in Material Design.

Now themes have only one "Hue" and its "Saturation" and  "Brightness" is modified. 

Having a palette would really give more freedom. I played with it [here](https://www.materialpalette.com/red/lime) but didn't found the origin of every color (e.g. the white background color)

In [this tool](https://material.io/tools/color/#!/?view.left=0&view.right=0&primary.color=F06292&secondary.color=F06292) it's clear that MD also derives the lighter and darker colors. 

I've checked the UIs collected by @seyyah [here](https://github.com/littlevgl/lvgl/issues/805#issuecomment-462172375) and I couldn't imagine how to use MD's palette to set the color of all those elements. 

Probably with shouldn't use MD's palette directly in LittelvGL but customize it a little to our needs.

I'm just thinking loudly about the possible palette color names :
1. background (screen, window etc background)
2. background_text (text color on background)
3. primary (buttons colors, slider indicator)
4. primary_text (text on primary color)
5. accent (toggled button)
6. accent_text
7. Something for list elements, keyboard color. It should be less colorful. I don't know a general name
8. text for 7)

puzrin, 2019-03-17:
> I've checked the UIs collected by @seyyah here and I couldn't imagine how to use MD's palette to set the color of all those elements.

Let's clarify.

- Having nice screens does not mean they follow MD guidelines. There is no guarantee that you can do everything possible in this world with 8 colors.
- State of art is to not invent custom things, but follow guideline and use existing features.
- It's not possible to "follow standard partially". You can "follow standard" or "not follow standard". There are no third alternative. It's normal to not follow standard if you don't like it. But anyone should not be confused by illusion of "partial following".

If problem is only in values of 8 colours:

- I don't like default MD palette too.
- It should not be a problem to search ready MD palettes in internet.
- As plan B, i can as designer to help with palettes, in the end, when everything else ready.

kisvegabor, 2019-03-18:
I didn't mean to "partially follow MD" just get some inspiration from it. :) The number of colors is just a coincidence.  We should really name the colors differently than MD does to avoid the illusion of using an MD palette.

If we agree on using palettes in themes (I support the idea) we should define the meaning of the colors. Probably my initial suggestion above is not good but can be a good starting point.

puzrin, 2019-03-18:
> We should really name the colors differently than MD does to avoid the illusion of using an MD palette.

Why?  I don't understand. If we follow MD guideline, it's logical to use the same coloring principles => logical to use the same names (or very similar).

puzrin, 2019-03-19:
https://material.io/develop/web/docs/theming/ see class names (9 total), there are some for background too. I can not comment tool from your link - not familiar. But tool != guideline.

> I'm not sure we should follow MD guideline.

We can not follow MD if you don't like it, but we should follow one of existing guideline. Following guideline consist of 2 important parts:

1. use existing solutions
2. avoid a lot of errors (many people forget about this)

"partial following" makes (2) impossible. And that trash all possible benefits about development time.

kisvegabor, 2019-03-20:
> https://material.io/develop/web/docs/theming/ see class names (9 total), there are some for background too. I can not comment tool from your link - not familiar. But tool != guideline.

These 9 colors are only for texts. 

Once we find documentation where all palette colors are described including texts, button, background (somehow didn't find such a description yet) and it fits well to LittlevGL, I'm fine with the MD palette too.. Following the standard has advantages as you pointed out.

puzrin, 2019-03-20:
> These 9 colors are only for texts.

Why? See table with class description. ~ Every second is for background, that' said explicit.

puzrin, 2019-03-20:
From https://www.materialpalette.com/indigo/yellow, download result

```css
.dark-primary-color    { background: #303F9F; }
.default-primary-color { background: #3F51B5; }
.light-primary-color   { background: #C5CAE9; }
.text-primary-color    { color: #FFFFFF; }
.accent-color          { background: #FFEB3B; }
.primary-text-color    { color: #212121; }
.secondary-text-color  { color: #757575; }
.divider-color         { border-color: #BDBDBD; }
```

Names seems expressive enougth. Values helps to understand where to apply.

kisvegabor, 2019-03-20:
> Why? See table with class description. ~ Every second is for background, that' said explicit.

Oh, I checked the "Colors" section. It really looks better. So three 4 main types (+4 for the texts' color on them):
1. primary: in LittelvGL for buttons, slider indicators, window headers
2. secondary: in LittelvGL for toggled buttons
3. surface: in LittelvGL for keyboard buttons, message box background
4. background: in LittelvGL screen, window etc. background

puzrin, 2019-03-20:
> How does it sound?

i could not find what is used as accent color (as in https://github.com/littlevgl/lvgl/issues/804#issuecomment-474719612)

In global, i'm ok with approach [ bgtype, on-bgtype ]

puzrin, 2019-03-20:
https://material.io/develop/android/theming/color/ - here is link with colour names for android (site docs are splitted by platforms)

kisvegabor, 2019-03-21:
I think accent color == secondary color.

It's hard to call MD palette to a standard thing because I've never seen two palette description twice yet.
It's really rather like guidance and everybody shapes it according to his needs. E.g. at some places there are 3 text colors (primary, secondary, hint) but others have one for each color. So it seems we have some freedom but should follow the original idea.

seyyah, 2019-03-21:
Another popular front-end framework is Twitter Bootstrap. It have a color (size etc) variables and default values,

```scss
$white:    #fff;
$gray-100: #f8f9fa;
$gray-200: #e9ecef
...
$blue:    #007bff;
...
$primary:       $blue;
$secondary:     $gray-600;
$success:       $green;
$info:          $cyan;
$warning:       $yellow;
$danger:        $red;
$light:         $gray-100;
$dark:          $gray-800;
...
$body-bg:                   $white;
$body-color:                $gray-900;
```

## :clipboard: Reference
1. https://github.com/twbs/bootstrap/blob/master/scss/_variables.scss

seyyah, 2019-03-21:
I'm not professional front-end developer. I'm working mainly backend developer.

Nebular have a great/rich UI toolkits [4]. It based on Angular. IoT dashboard is very impressive for us [3]. Theme (color) variables is here [2]. StackBlitz Seed Project/demo is in [1].

```scss
color-bg: #3d3780,
color-bg-active: #494299,
color-fg: #a1a1e5,
color-fg-heading: #ffffff,
color-fg-text: #d1d1ff,
color-fg-highlight: #00f9a6,

color-gray: rgba(81, 113, 165, 0.15),
color-neutral: transparent,
color-white: #ffffff,
color-disabled: rgba(255, 255, 255, 0.4),

color-primary: #7659ff,
color-success: #00d977,
color-info: #0088ff,
color-warning: #ffa100,
color-danger: #ff386a,

link-color: #00f9a6,
link-color-hover: #14ffbe,
separator: #342e73,
layout-bg: #2f296b,

scrollbar-fg: #554db3,
scrollbar-bg: #332e73,

menu-fg: color-fg,
menu-bg: color-bg,
menu-active-fg: color-white,
menu-group-fg: color-white,
```

## :clipboard: Reference
1. https://littlevgl-nebular.stackblitz.io
2. https://github.com/akveo/nebular/blob/master/src/framework/theme/styles/themes/_cosmic.scss
3. http://akveo.com/ngx-admin/pages/iot-dashboard
4. https://github.com/akveo/nebular

puzrin, 2019-03-21:
I would avoid bootstrap. It's too web-ish and less detailed about interactions.

seyyah, 2019-03-21:
The bootstrap may help us for (color) naming conventions/standard.

puzrin, 2019-03-21:
@seyyah if we plan to use MD, it's more logical to search in MD demos/docs.

seyyah, 2019-03-21:
> if we plan to use MD, it's more logical to search in MD demos/docs.

OK

kisvegabor, 2019-03-22:
@seyyah Thank you for the list of color names. Independently from the thems we can introduce new `LV_COLOR_...`s which have a meaningful purpose. E.g. `LV_COLOR_DANGER`

@puzrin 
Could you try to rework the colors of `lv_theme_material` with a palette discussed above? With `lv_test_theme1` you can test the result.

puzrin, 2019-03-22:
@kisvegabor i'd like to avoid this, if possible. Prefer to care about font convertor & pio demo.

kisvegabor, 2019-03-22:
@puzrin 
No problem. I can't immerse in either so left this open for someone to play with it.

**The task in summary**
1. Create a palette of 8 colors:
  * primary: in LittlevGL for buttons, slider indicators, window headers
  * primary text: text on primary color
  * secondary: in LittlevGL for toggled buttons
  * secondary text: text on secondary color
  * surface: in LittlevGL for keyboard buttons, message box background
  * surface text: text on surface color
  * background: in LittlevGL screen, window etc. background
  * background text: text on background color

2. Update a theme (e.g. [lv_theme_material](https://github.com/littlevgl/lvgl/blob/master/lv_themes/lv_theme_material.c)) to use the colors from the palette. 

3. Test the result with [lv_test_theme1](https://github.com/littlevgl/lv_examples/blob/master/lv_tests/lv_test_theme/lv_test_theme_1.c).

puzrin, 2019-03-24:
I could suggest a bit different approach. If someone creates "settings menu demo", i could polish it's look, and create/test theme for it.

kisvegabor, 2019-03-25:
Themes in LittlevGL are general things. You should be able to apply them to any of your UI to get a consistent design. If something is general it can't be perfect for every use case, but it could be a good starting point)

The settings demo should behave the same way. Instead of creating a specific design for it "manually" we should apply a theme to let the user change the design with one line of code,

puzrin, 2019-03-25:
The difference is in feedback and motivation. I think, you agree, all of us can have own reasons to participate. And that can be not only "creating great GUI lib".

For me, best motivation is to see result in one of my projects. From this point of view:

- Settings menu can be used immediately. Then there will be reasons to improve result via theme tune
- Standalone theme is too "abstract" (and will not cover animation effects). Result will not be useable immediately in my projects (and no way to check if it's enougth).

While final result may be the same, sequence is different and in first case i'm more interested to participate.

embeddedt, 2019-03-25:
I agree with @kisvegabor.

> Settings menu can be used immediately. Then there will be reasons to improve result via theme tune. Standalone theme is too "abstract".

The general theme included in LittlevGL should be generic (or abstract). That way it works for more people. Then the settings demo itself should include code that modifies the default styles to be more suitable to that specific application.

puzrin, 2019-03-25:
@embeddedt we speak about different things. See what is https://en.wikipedia.org/wiki/Software_prototyping#Evolutionary_prototyping.

I prefer to have certain guarantees about result. There should be some way to verify everything works as expected. One of possible way - create a real prototype, that user can touch with hands. Experimenting with theme is not enougth.

I don't say that's the best/only way of development. Only describe personal preferences, when i can participate.

embeddedt, 2019-03-25:
> One of possible way - create a real prototype, that user can touch with hands. Experimenting with theme is not enough.

I don't disagree. We should definitely create a realistic and well-designed example. I just think the built-in theme system should be generic and not designed for one application. Specific modifications belong with the prototype, not LittlevGL.

puzrin, 2019-03-25:
> I don't disagree. We should definitely create a realistic and well-designed example. I just think the built-in theme system should be generic and not designed for one application.

Sure. I only suggested to change the order or implementation. With demo available first - it's more easy to verify candidate for built-in theme.

kisvegabor, 2019-03-26:
Sure. I only suggested to change the order or implementation. With demo available first - it's more easy to verify candidate for built-in theme.

I agree, besides `lv_test_theme1` and `lv_test_theme2`, the settings demo can be a good application to test the themes.

I'll promote that topic too as help wanted. I still have some work with `dev-6.0`, but if nobody volunteers I can add the demo later.

puzrin, 2019-05-02:
not stale

kisvegabor, 2019-05-24:
not stale

UtkarshBhatthere, 2019-05-26:
Hey, I went through all of your discussions (Truthfully I skipped some parts of the thread :D). Is there anything I can help with ? (My experiences to lvgl are limited to just using the API's but i can cope I guess.)

embeddedt, 2019-05-26:
@UtkarshBhatthere The main tasks to be accomplished are;

1. Create a settings demo with a similar user interface to Android. puzrin created a live HTML example, but I can't seem to find the link. @puzrin can you post the link here again?
2. Modify the theme system so that the settings demo complies with Material Design.

puzrin, 2019-05-26:
@UtkarshBhatthere it would be nice to have something "easy to touch with hands". From my point of view, first steps could be:

1. Revive PlatformIO demo. See https://github.com/littlevgl/pc_simulator_sdl_platformio/tree/vit
   - it worth add minimal STM32F429I-DISCO support (just to check hardware is separated from other code)
2. Implement settings menu demo, according to screenshots https://github.com/littlevgl/lv_examples/issues/26#issuecomment-473279547

Then it will be more easy to decide, how to integrate that into core.

kisvegabor, 2019-05-27:
> Create a settings demo with a similar user interface to Android. puzrin created a live HTML example, but I can't seem to find the link. @puzrin can you post the link here again?

Creating the settings demo is a different issue: https://github.com/littlevgl/lv_examples/issues/26

Here the goal is to add palette support to the themes. Now only one `hue` value is used, but it would be more flexible to use a palette of 8 colors. 

I summarized the palette colors in [this comment](https://github.com/littlevgl/lvgl/issues/804#issuecomment-475818517).

UtkarshBhatthere, 2019-05-30:
> > Create a settings demo with a similar user interface to Android. puzrin created a live HTML example, but I can't seem to find the link. @puzrin can you post the link here again?
> 
> Creating the settings demo is a different issue: [littlevgl/lv_examples#26](https://github.com/littlevgl/lv_examples/issues/26)
> 
> Here the goal is to add palette support to the themes. Now only one `hue` value is used, but it would be more flexible to use a palette of 8 colors.
> 
> I summarized the palette colors in [this comment](https://github.com/littlevgl/lvgl/issues/804#issuecomment-475818517).

I am so sorry i couldn't reply earlier guys. I'll get on to the palette and report you guys with the updates soon. Just to be sure, you are asking for 8 Color based palettes that can be 'chosen' to give a theme like feel to the UI as in 'dark mode' etc.

kisvegabor, 2019-05-31:
> Just to be sure, you are asking for 8 Color based palettes that can be 'chosen' to give a theme like feel to the UI as in 'dark mode' etc.

Yes. For example, the `lv_theme_xxx_init()` can have a prototype like 
```c
/*In lv_theme.h*/
typedef struct {
  lv_color_t primary;
  lv_color_t primary_text;
  ....
}lv_theme_palette_t

/*The init functions*/
lv_theme_xxx_init(const lv_theme_palette_t * palette, const lv_font_t * font);
```

closed 2019-06-28 by stale[bot]