lvgl/lvgl #805: [request] "Setting menu" demo

puzrin, 2019-02-07:
A lot of small embedded devices have interface like this:

1. Main screen
2. Settings menu with tons of params (mostly numeric, sometime text)

IMHO it would save a lot of time, if everyone could take ready example, and copy-paste code with controls of required type.

What is expected from menu:

- Good design
- Proper interaction with different input devices (touch/buttons)
- Code blocks for popular controls, ready to copy-paste
  - Example: "need to set speed between [1.0 ... 5.0] m/sec with step 0.1"

kisvegabor, 2019-02-07:
Awesome idea! it"s really something which would useful fora  lot of people!

It turned out that you have designer skills. Can you create a layout design?

Finally, the user only needs to apply a theme to change the default (plain) appearance.

puzrin, 2019-02-07:
> It turned out that you have designer skills. Can you create a layout design?

I have no designer skills. Only some experience to see obvious problems in design/usability, and where to take ready solutions when available. My skills are more close to "software architect"

If you give me some time, i will prepare list of exact links to UI guidelines, where to read about each menu element.

seyyah, 2019-02-07:
> It turned out that you have designer skills. Can you create a layout design?

I'm not very good, but I try still to do a pencil sketch/moqup and share it.

kisvegabor, 2019-02-08:
@puzrin 
Right now, I have no time to immerse in it so deep. Anyway, a list of links would be useful.

@seyyah 
Thank you, waiting for that!

puzrin, 2019-02-08:
@kisvegabor give me 1 week. I'm trying to understand, what can be useable for programmers, to quickly create sketches interface from ready material components. Something like https://material.io/collections/developer-tutorials/#web

puzrin, 2019-02-08:
_Tech note._

https://stackblitz.com/edit/mdc-web-quick-start-demo

Nice online IDE. Seems demo can be published on web. That's ~ the same as offline (node.js + webpack), but result can be seen immediately, without install. + anyone can fork & edit.

_investigating..._

seyyah, 2019-02-10:
[1] maybe inspired us. I am using Pinterest daily and I opened a board [2] and gathering inspirational work here as I meet.

What should be included in the "settings menu"?

What do you think about "home automation demo" idea?

## :camera: Pins
| | A | B | C | D |
|-|-|-|-|-|
|1| [image] | [image] | [image] | [image] |
| 2 |[image] | [image] | [image] | [image] |
| 3 | [image] | [image] | [image] | [image] |

## :clipboard: Reference
1. https://pttrns.com/?scid=16
2. https://tr.pinterest.com/5eyyah/uiux-littlevgl/

puzrin, 2019-02-11:
> What should be included in the "settings menu"?

- integers with different change methods (+/-, slider)
- the same, but floats with limited decimal part + step to change
- no keyboard
- acceleration on long press

> What do you think about "home automation demo" idea?

I'd say, screens look nice, but there are nothing special in details. All this things are described in UI guidelines. IMO words "home automation" may sound magically, but have nothing special/specific behind. If you follow UI guidlines, it's not a problem to create zillions of such demos. I'd suggest to take existing demo, decide what guideline to follow and fix all deviations.

Scope of this issue is a bit different - provide "almost ready" interface solution for simple projects.

kisvegabor, 2019-02-12:
@seyyah It's really inspiring, thank you!

> What should be included in the "settings menu"?
>> * no keyboard

@puzrin Typing a name for something (room, device name, wifi network etc) is quite commont. IMO we should have a one line text area + keayboard in the example 

> I'd say, screens look nice, but there are nothing special in details. All this things are described in UI guidelines. IMO words "home automation" may sound magically, but have nothing [...]

The demo should use the very default lvgl style (white background, blueish buttons) but can be customized by [applying a theme](https://littlevgl.com/themes). That's the key of the themes. There is a UI with a default appearance but can be completely changed with one line of code. So the themes should follow the UI guidelines.

> Scope of this issue is a bit different - provide "almost ready" interface solution for simple projects.

Yes, in this issue we should focus on:
1. figure out what widgets would be required for a general main screen + settings screen UI.
2. create a layout design
3. implement it in lvgl
4. (create a video for youtube :) )

puzrin, 2019-02-12:
> Typing a name for something (room, device name, wifi network etc) is quite commont. IMO we should have a one line text area + keyboard in the example

In ideal world, setting menu demo should have everything :). But expences to do keyboard "right" will postpone result significantly. I always check several conditions:

- result should be self-sufficient (useable, in this case)
- possible to do in shout time
- extendable (doable with small steps)

I have nothing against keyboard, but suggest postpone it to second iteration. My answer was not "what demo needs at all", but "what demo needs right now, immediately"

puzrin, 2019-02-12:
More comments.

>* figure out what widgets would be required for a general main screen + settings screen UI.

I'd stay aside of main screens. Because variability of main screens is very high, and it's not a big problem to "combine some labels". Trying to generalize - tend to fall into #804.

Suggestion is to focus on settings menu only. It's doable with current possibilities, and result will be valuable.

> 4\. (create a video for youtube :) )

I did not posted anywhere, but suggestion is to make general template for all examples, to simpllify emscripten builds and publish on gh-pages branch.

The reason i did not posted this suggestion - because examples are not self-sufficient. Those are more "code snippets". Personally, i hope #750 to be first example of "demo done right" and land settings menu there.

kisvegabor, 2019-02-13:
> Suggestion is to focus on settings menu only. It's doable with current possibilities, and result will be valuable.

Sound reasonable.

> I did not posted anywhere, but suggestion is to make general template for all examples, to simpllify emscripten builds and publish on gh-pages branch.

I don't understand. In "general template for all examples" which examples do you mean for example?

> The reason i did not posted this suggestion - because examples are not self-sufficient. Those are more "code snippets". Personally, i hope #750 to be first example of "demo done right" and land settings menu there.

I commented on #750

puzrin, 2019-02-13:
> I don't understand. In "general template for all examples" which examples do you mean for example?

We do i18n right now. It would be logical to make example for it. Micro demo with "hello world", and dropdown to select language.

It would be convenient to have template of pio config:
- with settings/deps for SDL build
- with setting for web build (PIO supports additional platforms, but i did not investigated)

embeddedt, 2019-02-13:
@puzrin 

I think I didn't understand what you meant. You said "It would be convenient to have template" but from what I can see we already have one of those. Could you please clarify what you mean by "template"?

> It would be convenient to have template of pio config (or Makefile?):
> 
>  * with settings/deps for SDL build

https://github.com/littlevgl/pc_simulator_sdl_eclipse has a Makefile for those who don't want to use Eclipse.

>  * with setting for web build (PIO supports additional platforms, but i did not investigated)

https://github.com/littlevgl/emscripten builds using a Makefile as well.

puzrin, 2019-02-13:
I mean all such things could be convenient in one place. When new "project" (live demo in this case) required, it's convenient to copy skeleton with all settings/scripts instead of learn/collect details.

Also, i think it would be convenient to make examples via platformio, because those are more easy to install.

embeddedt, 2019-02-13:
> When new "project" (live demo in this case) required, it's convenient to copy skeleton with all settings/scripts instead of learn/collect details.

What details need to be collected? At the moment making a new example merely requires modifying `main.c`.

puzrin, 2019-02-13:
I'd like this things in the same time, on one place:

- install deps via platformio (because user can just open folder and wait until pio does everything).
- able to build for SDL/web/stm32f429-disco in one click.

BrianPugh, 2019-02-13:
@puzrin If you feel like digging through my code; I've setup a pretty good system for a 128x64 screen.

https://github.com/joltwallet/jolt_wallet/tree/joltos

https://gfycat.com/DelayedTatteredCopperbutterfly

kisvegabor, 2019-02-15:
> I'd like this things in the same time, in one place:
> * install deps via platformio (because user can just open folder and wait until pio does everything).
> * able to build for SDL/web/stm32f429-disco in one click.

I'm afraid it won't be that simple with Pio either because you still need to manually install SDL to work in a PC simulator and install the Emscripten SDK to compile javascript.

Despite of it, I believe that Pio is a good direction because it's easy to switch between platforms. Although it wasn't convenient for me to develop the library, it can be for the users who use the library.

puzrin, 2019-02-15:
> I'm afraid it won't be that simple with Pio either because you still need to manually install SDL to work in a PC simulator and install the Emscripten SDK to compile javascript.

Ideal is not achievable, but we still can have some benefits & simplify user's life as max as possible:

- After deps installed once, all demo can be "opened" in PIO with one click (i hope)
- We can provide helper scripts for PIO (in python) and say "just run it to install the rest".
  - That will be OS-transparent (no specific .sh / .bat files)
  - May be we could configure PIO to run this script automatically, if no SDL/emscripten found. Or post clean message what to type in PIO terminal. I did not investigated details, but PIO has some pre-hooks for major comands.

Also note, PIO is NOT 100% free. After 1 month it requires paid account to run tests & bare metal debug (compilation & upload still works without restrictions). But i still don't know better alternative to manage C dependencies for embedded.

embeddedt, 2019-02-15:
> After 1 month it requires paid account to run tests & bare metal debug (compilation & upload still works without restrictions).

As an embedded developer, I can say that I probably wouldn't go for a tool like that. The ability to debug is extremely helpful to me and a huge time saver.

> i still don't know better alternative to manage C dependencies for embedded.

Why do we need to manage those dependencies in the first place? As an example, if I want to build a LittlevGL application with STM32F746G-DISCO, I just have to download the existing Eclipse-based project. What's wrong with that setup?

puzrin, 2019-02-15:
> Why do we need to manage those dependencies in the first place?

Many languages (js, ruby, python, rust, ...) have package managers. This is VERY convenient and simplify collaborative development. But i don't know how to immediately explain benefits to one who never user such things.

I can explain my personal feeling. After coming from languages with package managers, i feel like returned 10 years back :). It may sound strange, but that's the main reason why i wish switch to Rust.

> As an example, if I want to build a LittlevGL application with STM32F746G-DISCO, I just have to download the existing Eclipse-based project. What's wrong with that setup?

If something works - nothing can be wrong with it :). If anyone works with stm32 only, Eclipse-based packs like Attolic can be a very good choice.

I can only list PIO features, useful for me, and you decide if they worth your attention:

- automatic compiler toolchains install (build)
- automatic flashers install (upload)
- support other targets (esp/esp32/avr/...)
- ready ld scripts for "all known" MCUs
- deps management

Note, i'm not sold to PIO and not going to promote it blind. I share both pros and cons, to help all make choice, based on their own preferences.

embeddedt, 2019-02-15:
> I can only list PIO features, useful for me, and you decide if they worth your attention:
> 
>     * automatic compiler toolchains install (build)

That is useful. On popular platforms like ARM and operating systems like Ubuntu, installing a toolchain is usually a single command; however, I've found the steps to be a lot more complicated on Windows and less popular embedded platforms.

> Note, i'm not sold to PIO and not going to promote it blind. I share both pros and cons honest, to help all make choice, based on their own preferences.

From what I can tell, PIO looks like an attempt to unify the IDE ecosystem into one tool that supports many boards. Is that correct?

puzrin, 2019-02-16:
> From what I can tell, PIO looks like an attempt to unify the IDE ecosystem into one tool that supports many boards. Is that correct?

That's a correct statement too.

As opensource developper i don't like hard pinning to any IDE. So, prefer to say "it solves some boring everyday tasks for C programmer/user". See list above.

Or consider it as "C-specific kludge", to make development process more "modern". For example, if we consider Rust, it's installed with one line, has package manager and, as a result - quickly growing ecosystem of packages. It still has no solution for easy flash upload, but i posted this moment to wishlist.

puzrin, 2019-02-24:
Preliminary sketch:

- [online](https://stackblitz.com/edit/lvgl-settings-menu-demo) (site is experimental, may be unstable, not works with firefox)
- Repo: https://github.com/puzrin/lvgl_settings_menu_demo - for local install, works everywhere.

This is what i mean as "settings menu" and how i'd like it too see. Popup dialogs for params change are not ready yet.

kisvegabor, 2019-02-24:
@puzrin Thank you for setting this up!

I'm curiously waiting for the pop-up windows for parameter settings. I think it will be the really interesting part.

puzrin, 2019-03-15:
Some screenshots from designer.

- Everything except checkbox shold be shown as hovering sliding pane when user tap (select) setting from list. The same as here https://github.com/littlevgl/lvgl/issues/805#issuecomment-466734171
- Palette is default Material, don't care about colors now.
- Screenshots are 240*240px, to reflect look on popular LCD-s.
- Should work with both touchscreen and encoder/joystick
  - no need to "change focus on active item" 
- Numeric inputs are incomplete, posted drafts.

### Checkbox setting

[image]

### Replacement for selectbox

(dropdown with single choice):

[image]

### Numeric input with slider

(small range / low precision)

[image]

### Numeric input variants

- Incomplete yet.
- Depend on input type (touch / [encoder/joystick])

[image] [image] [image] [image] [image] [image]

kisvegabor, 2019-03-15:
Have you pushed the updates? I still see the old GUI in  [online](https://stackblitz.com/edit/lvgl-settings-menu-demo) and in [your repo](https://github.com/puzrin/lvgl_settings_menu_demo) the latest commit was 19 days ago.

puzrin, 2019-03-15:
I did not planned to update demo. Are screenshots not enougth?

puzrin, 2019-03-15:
Final iteration (for public review)

- Reviewed by me, but any comments welcome.
- Inputs have variants with/without additional description.
- Input for devises with and without touchscreen looks different.
- No focus change (like tab navigation via multiple fields). Only one item can be changed at once.

## Settings list

See demo https://github.com/littlevgl/lvgl/issues/805#issuecomment-466734171

## Numeric input with slider

- Both touch & encoder/joystick.
- touch: for small range (steps count), because not very precise.

[image] [image]

## Yes/No (checkbox)

- Embedded into settings list, no separate screen.
- If difficult technically - additional screen with sliding switch instead of checkbox.

[image]

## Numeric input, touchscreen

- Touchscreen requires emplicit buttons on display.

[image] [image]

alternate (may be someone like this):

[image] [image]

## Numeric input, encoder

- Encoder - knob with rotation left/right + push.
- No buttons, only arrows to show rotation direction (left/right).
- Acceleration required on fast rotate.

[image]

## Numeric input, joystick

- Joystick - [5-ways tactile switch](https://www.google.com/search?q=5-ways+tactile+switch&tbm=isch)
- Since user can enter meny via "push right", values are changed via push up/down
- Incremental acceleration required on long push

[image] [image]

## "Selectbox"

- Equal to dropdown selector
- Looks like "nested" menu list with radiobuttons and without descriptions
- When user selects something, he returns to parent menu level

[image]

puzrin, 2019-03-15:
Seems finished everything promised, except font builder. Can i help with something else?

Where can i be useful:

- JS tools for LVGL.
- Design/usability recommendations (+ can order sketches from my designer).
- Questions about wrapping projects into platformio.
- Generic questions about architectures and data flows.

What i like to avoid:

- C programming :). I don't like it.
- Advanced webpage interfaces.

kisvegabor, 2019-03-17:
@puzrin Thank you very much! Conceptually looks good to me.

A simple, modular demo can be implemented based on the screenshots.

My view is to make possible to apply themes on this demo too. So let's continue with the discussion of themes in #804