lvgl/lvgl #806: [v6.0] suggestion: make focus style modification mechanism part of the theme

manison, 2019-02-07:
Currently you can define function that will modify style of focused object and you can assign it to group with `lv_group_set_style_mod_cb` regardless of current theme. Since it is responsibility of a theme to define the visual style of objects it would be IMHO better if the theme also defined the visual style for objects when they have the focus.

I suggest moving the style modification function into the theme. For backwards compatibility with v5 the theme's callback could be used by default (instead of current "orangish" default callback) with the possibility to replace it with `lv_group_set_style_mod_cb` as it is now.

Also see #277, #804

kisvegabor, 2019-02-07:
Sounds reasonable. 

`LV_THEME_LIVE_UPDATE` is enabled it's important to have only styles in `lv_theme`. To solve this the currently existing styles should be packed in a `struct` called style `style` It would result in `th->style.btn.rel`

Besides `style` there can be an other `struct` called `group` with `focuse_cb` and `edit_cb` elements.

These callbacks can be assigned when the group is created.

How does it sound to you? Would you like to implement it?

manison, 2019-02-07:
OK. I'll try to create an initial implementation.

manison, 2019-02-11:
In the initial commit https://github.com/manison/lvgl/commit/44e5cc446c9a82043acc3cbaa963d8b409a3c4dc I only moved the styles into the substructure `style` within the `lv_theme_t` structure as you suggested. Please review.

manison, 2019-02-11:
I would like to ask you about a comment in _lv_theme.c_ above the `th_styles` array definition:
> Can't determine the size in compile time because sizeof is not evaluated (should be `sizeof(lv_theme_t) / sizeof(lv_style_t*)

Since `sizeof` is a compile time operator so this expression should be just fine and in fact my compiler does not complain (after the change I use expression `sizeof(((lv_theme_t *)0)->style) / sizeof(lv_style_t*)`). What compiler do you use?

manison, 2019-02-11:
**TODO**
- [ ] Update examples accordingly (_lv_test_theme.c_)

kisvegabor, 2019-02-12:
> I only moved the styles into the substructure `style` within the `lv_theme_t` structure as you suggested. Please review.

Looks good to me! :+1: 

> I would like to ask you about a comment in _lv_theme.c_ above the `th_styles` array definition:

It's also working for me with  `gcc`. I don't remember why I did it this way (I need to add better comments next time). Maybe it's not working in Visual studio?

Anyway, let's use `sizeof` and we will see.

manison, 2019-02-12:
> Maybe it's not working in Visual studio?

Nah, Visual Studio is fine. I'm using it :-)

kisvegabor, 2019-02-12:
Then fine! Maybe I just overcomplicated it :)

manison, 2019-02-13:
@kisvegabor **How to notify all groups about theme changes?**
I introduced function `lv_group_report_style_mod` (here https://github.com/manison/lvgl/commit/327d7c84caea08a3b2a15e5657104c3058fd974e#diff-1ca71106770136ef4cd6600d307f22b2R531) and I need to call it for _all_ groups whenever theme changes (providing `LV_THEME_LIVE_UPDATE` is enabled) so groups can update their style modification functions from changed theme. But since there is no global linked list of all groups I have no way to iterate them. I was thinking about calling the function from `lv_obj_report_style_mod` where the whole object tree is recursively iterated and checking each object's `group_p` and calling the `lv_group_report_style_mod` upon it. This would probably work but since multiple objects can share the same group this would result in group being updated multiple times. It wouldn't hurt since theme changes infrequently, but anyway - what do you think? Any other suggestions?

kisvegabor, 2019-02-13:
Yes, the groups are not added to a linked list. So far it wasn't required. However, having a linked list for them could be a solution now. Instead of `lv_group_t * group = lv_mem_alloc(sizeof(lv_group_t));` in `lv_group_create()` `lv_group_t * group = lv_ll_ins_head(&_lv_group_ll));` can be used.

As `_lv_group_ll` should be marked as `GC_ROOT` (Garbage collector root object) similarly to [this](https://github.com/manison/lvgl/blob/master/lv_misc/lv_task.c#L54)

After that with a `lv_gorup_get_next()` function you can iterate through the linked list. Similarly to [lv_obj_get_child](https://github.com/manison/lvgl/blob/master/lv_core/lv_obj.c#L1368)

manison, 2019-02-13:
I agree the linked list of groups is better long-term solution. Should I create a distinct PR for this or should this go to the "theme style modification" work?

kisvegabor, 2019-02-14:
> Should I create a distinct PR for this or should this go to the "theme style modification" work?

It's fine to have it in the same PR.

manison, 2019-02-15:
> It's fine to have it in the same PR.

OK.

The PR #844 is ready for review.

I tried to implement some focus styles that matches individual themes but it will definitely need some more love from someone with visual feeling :) Probably during #804.

kisvegabor, 2019-02-15:
Thank you very much!

> I tried to implement some focus styles that matches individual themes but it will definitely need some more love from someone with visual feeling :) Probably during #804.

It will be fine for now :)

manison, 2019-02-19:
@kisvegabor I wanted to update the examples but I see you already did it, great :) Thanks.

closed 2019-02-19 by manison

kisvegabor, 2019-02-19:
@manison Yes, I've created the dev-6.0` branch for `lv_examples`.