lvgl/lvgl #807: Extend keypad interface

zavovi, 2019-02-07:
Hi,
I am thinking about using keypad (like old mobile phones - numbers with letters). Is it possible make any interface for use this keypad? I mean, when I click one button twice or three times, then I can change the letter on this button (e.g. num 2 - a, b,c,...). 

Is it any solution for it?

Thank you

kisvegabor, 2019-02-07:
Hi,

For [Button Matrix](https://docs.littlevgl.com/#Button-matrix) you can set an action where you can handle it. You can keep track with number of presses, timeouts etc.

zavovi, 2019-02-07:
Hi, thank you. And how can I use it with external input driver - lv_indev_drv_t ?

zavovi, 2019-02-07:
Maybe, I found the solution, that I remove previous character and show new and it is working for me :-).

I have another question for keypad interface:
How can I put the unicode characters? For example u00E1 ? I tried 
`char c = 'u00E1';`
It is not working.
Thank you.

kisvegabor, 2019-02-08:
> Maybe, I found the solution, that I remove previous character and show new and it is working for me :-).

Great! :)

> How can I put the uUnicode characters? For example u00E1 ?

You nned to pass UTF8 characters:
It should work: `lv_ta_add_text(ta, "Á");` if your IDE/editor has UTF8 encoding
Or `lv_ta_add_char(ta, 0x81C3);`
'Á' in UTF8 is `0xC381` but you need to swap the order --> 0x81C3

See this site for Unicode utf8 converter: http://www.ltg.ed.ac.uk/~richard/utf-8.cgi

As the Unicode issue is not related to the original question I close this issue, If you have further questions about the Unicode character please open a new issue.

closed 2019-02-08 by kisvegabor