lvgl/lvgl #808: Tab view - vertical tabs

zavovi, 2019-02-07:
Hi,
is it possible change the tab view for vertical tabs (on the left/right side)?

Thank you.

kisvegabor, 2019-02-07:
Hi,

not supported directly now. However, besides `TOP`, the `BOTTOM` position is also supported. Similarly it `LEFT` and `RIGHT` can be added.

Would like to implement it and send a PR?

zavovi, 2019-02-21:
Hi, sorry for the late response, I hadn't lot of free time. I tried change your API for tabs and now, I can use it in right and left. There is lot of changes and I don't know, if it is ok for Pull Request now. It is changed mainly for my usage. If you want see it, it is on my fork on branch ZAVAVOV this commit: https://github.com/zavovi/lvgl/commit/effc5f8700dd68c0754e266d459109659fbcb76c

zavovi, 2019-02-22:
The default style couldn't replaced by theme.

kisvegabor, 2019-02-24:
Anyway please send a PR. Probably the default transparent style needs to be reverted but it's nota  big deal.

zavovi, 2019-02-25:
Hi a sent the PR. Here is some example of my usage of yours lib: https://www.facebook.com/zavavov/videos/829343017414636/

kisvegabor, 2019-02-26:
Thank you for the PR!

Nice project! Is it a commercial or a hobby product? Anyway, it'd be great to add it to [LittlevGL references](https://blog.littlevgl.com/2018-12-26/references).

zavovi, 2019-02-27:
It is commercial, but only small pieces. I am making it at home. It is for my command station: http://www.zavavov.cz/en/tcs-en/ and you could add it into your references, after it will be done and I create web page for it. Thank you for your support :-) 

About changes in PR, I will make it, when I will have more time for test it.

kisvegabor, 2019-02-28:
>  It is commercial, but only small pieces. I am making it at home. It is for my command station: http://www.zavavov.cz/en/tcs-en/ and you could add it into your references, after it will be done and I create web page for it

Awesome! I think it would be better if you added the project and wrote a few lines. You can highlight the features better and add the most relevant image. It's not a problem if are not fully ready with it, you can update the references page later.

> About changes in PR, I will make it, when I will have more time for test it..

Okay, thank you. Then I close this issue. Let's continue in the PR.

closed 2019-02-28 by kisvegabor