lvgl/lvgl #809: Horizontal scrollbar

zavovi, 2019-02-07:
Hi,
is it possible use some object with horizontal scrollbar. For example, I want add buttons from left to right and I want move from buttons right to left for see next buttons. How can I make it?

Thank you.

kisvegabor, 2019-02-07:
Hi,

Have look at [lv_page](https://docs.littlevgl.com/#Page).

zavovi, 2019-02-08:
Hi, thank you. I tried it with this code:

        ```
lv_obj_t * page = lv_page_create(screen, NULL);
	lv_obj_set_size(page, 300, 100);
	lv_obj_align(page, NULL, LV_ALIGN_CENTER, 0, 0);
	lv_page_set_sb_mode(page, LV_SB_MODE_AUTO);

	lv_obj_t * label;

	lv_obj_t * btn1 = lv_btn_create(page, NULL);
	lv_obj_align(btn1, NULL, LV_ALIGN_IN_LEFT_MID, 10, 0);
	/*Add a label to the button*/
	label = lv_label_create(btn1, NULL);
	lv_label_set_text(label, "Text1");

	lv_obj_t * btn2 = lv_btn_create(page, NULL);
	lv_obj_align(btn2, btn1, LV_ALIGN_OUT_RIGHT_MID, 10, 0);
	/*Add a label to the button*/
	label = lv_label_create(btn2, NULL);
	lv_label_set_text(label, "Text2");

	lv_obj_t * btn3 = lv_btn_create(page, NULL);
	lv_obj_align(btn3, btn2, LV_ALIGN_OUT_RIGHT_MID, 10, 0);
	/*Add a label to the button*/
	label = lv_label_create(btn3, NULL);
	lv_label_set_text(label, "Text3");

	lv_obj_t * btn4 = lv_btn_create(page, NULL);
	lv_obj_align(btn4, btn3, LV_ALIGN_OUT_RIGHT_MID, 10, 0);
	/*Add a label to the button*/
	label = lv_label_create(btn4, NULL);
	lv_label_set_text(label, "Text4");

	lv_obj_t * btn5 = lv_btn_create(page, NULL);
	lv_obj_align(btn5, btn4, LV_ALIGN_OUT_RIGHT_MID, 10, 0);
	/*Add a label to the button*/
	label = lv_label_create(btn5, NULL);
	lv_label_set_text(label, "Text5");

	lv_indev_t * indev = NULL;

	lv_group_t *group = lv_group_create();
	lv_group_add_obj(group, btn1);
	lv_group_add_obj(group, btn2);
	lv_group_add_obj(group, btn3);
	lv_group_add_obj(group, btn4);
	lv_group_add_obj(group, btn5);

	do
	{
		indev = lv_indev_next(indev);

		if(indev != NULL && indev->driver.type == LV_INDEV_TYPE_ENCODER)
			lv_indev_set_group(indev, group);

	} while(indev);
```
And now, I see two and half buttons. If I use encoder to move right, I see moving the orange frame around the first button, second button and third (half button), but not moved buttons. How can I move with buttons with encoder? 

Thank you.

kisvegabor, 2019-02-09:
You can use the "focus callback":
```c
lv_group_set_focus_cb(g, my_func);

void my_func(lv_group_t * g)
{

    lv_obj_t * f = lv_group_get_focused(g);
    
    /*Get the parent page from the button: button -> page scrollable -> page*/
    lv_obj_t * p = lv_obj_get_parent(f);        /*p = page scrollable*/
    if(p) {
        p  = lv_obj_get_parent(p);              /*p = page*/
        if(p == page) {                         /*the button is on you page*/
            lv_page_focus(page, f, 300);
        }
    }
}
```

zavovi, 2019-02-09:
Many thanks, it seems be good, but the function `lv_page_focus` not working for me.

I added this:
```
lv_page_set_sb_mode(page, LV_SB_MODE_DRAG);
lv_page_set_scrl_fit(page, true, true);		//Scroll object size is fit by objects into
```

kisvegabor, 2019-02-10:
Upps, I really forget to mention this.

Glad to hear that you find it!

zavovi, 2019-02-10:
> Upps, I really forget to mention this.
> 
> Glad to hear that you find it!

Oh, so sorry, I added it, now I can move it with the object only when I add the page into group and I select it. But is not good for me. Still cannot move with page only when focusing the buttons.

kisvegabor, 2019-02-12:
Then add the `page` to the group and use `lv_page_set_arrow_scroll(page, true)` to enable scrolling with arrows.

zavovi, 2019-02-12:
Yes, it is working, but I want move with page by focusing buttons. Why not working function `lv_page_focus` ?

kisvegabor, 2019-02-12:
Please post a simple code what I can test.

zavovi, 2019-02-12:
Hi, I have this code:

```
static lv_obj_t * page;
static void _group_func(lv_group_t * g)
{
	lv_obj_t * f = lv_group_get_focused(g);

	/*Get the parent page from the button: button -> page scrollable -> page*/
	lv_obj_t * p = lv_obj_get_parent(f);        /*p = page scrollable*/
	if(p)
	{
		p  = lv_obj_get_parent(p);              /*p = page*/
		if(p == page)
		{                         /*the button is on you page*/
			lv_page_focus(page, f, 30);
			printf("Page focused\n");
			if(f == NULL)
				printf("f is NULL! \n");
		}
	}
}

static tcs_err_t _app_screen_show_body_tab_trains(lv_obj_t * screen)
{
	page = lv_page_create(screen, NULL);
	lv_obj_set_size(page, 320, 100);
	lv_obj_align(page, NULL, LV_ALIGN_IN_TOP_MID, 0, 0);
	lv_page_set_sb_mode(page, LV_SB_MODE_DRAG);
	lv_page_set_scrl_fit(page, true, true);		//Scroll object size is fit by objects into

	lv_obj_t * label;

	lv_obj_t * btn1 = lv_btn_create(page, NULL);
	lv_obj_align(btn1, NULL, LV_ALIGN_IN_LEFT_MID, 10, 0);
	/*Add a label to the button*/
	label = lv_label_create(btn1, NULL);
	lv_label_set_text(label, "Text1");

	lv_obj_t * btn2 = lv_btn_create(page, NULL);
	lv_obj_align(btn2, btn1, LV_ALIGN_OUT_RIGHT_MID, 10, 0);
	/*Add a label to the button*/
	label = lv_label_create(btn2, NULL);
	lv_label_set_text(label, "Text2");

	lv_obj_t * btn3 = lv_btn_create(page, NULL);
	lv_obj_align(btn3, btn2, LV_ALIGN_OUT_RIGHT_MID, 10, 0);
	/*Add a label to the button*/
	label = lv_label_create(btn3, NULL);
	lv_label_set_text(label, "Text3");

	lv_obj_t * btn4 = lv_btn_create(page, NULL);
	lv_obj_align(btn4, btn3, LV_ALIGN_OUT_RIGHT_MID, 10, 0);
	/*Add a label to the button*/
	label = lv_label_create(btn4, NULL);
	lv_label_set_text(label, "Text4");

	lv_obj_t * btn5 = lv_btn_create(page, NULL);
	lv_obj_align(btn5, btn4, LV_ALIGN_OUT_RIGHT_MID, 10, 0);
	/*Add a label to the button*/
	label = lv_label_create(btn5, NULL);
	lv_label_set_text(label, "Text5");

	lv_indev_t * indev = NULL;

	lv_group_t *group = lv_group_create();

	//lv_group_add_obj(group, page);
	lv_group_add_obj(group, btn1);
	lv_group_add_obj(group, btn2);
	lv_group_add_obj(group, btn3);
	lv_group_add_obj(group, btn4);
	lv_group_add_obj(group, btn5);

	lv_group_set_focus_cb(group, _group_func);

	do
	{
		indev = lv_indev_next(indev);

		if(indev != NULL && indev->driver.type == LV_INDEV_TYPE_ENCODER)
			lv_indev_set_group(indev, group);

	} while(indev);

	return TCS_OK;
}

static tcs_err_t _app_screen_show_body(lv_obj_t * screen)
{
	lv_obj_t *tabview = lv_tabview_create(screen, NULL);

	lv_obj_t *tab1 = lv_tabview_add_tab(tabview, SYMBOL_LOOP);
	lv_obj_t *tab2 = lv_tabview_add_tab(tabview, SYMBOL_HOME);
	lv_obj_t *tab3 = lv_tabview_add_tab(tabview, SYMBOL_SETTINGS);
	lv_tabview_set_tab_act(tabview, 0, false);

	_app_screen_show_body_tab_trains(tab1);
	//_app_screen_show_body_tab_accessories(tab2);
	//_app_screen_show_body_tab_others(tab3);

	return TCS_OK;
}

//This is called from main app and the app_screen_ctx.screen.body is screen
_app_screen_show_body(app_screen_ctx.screen.body);
```

zavovi, 2019-02-12:
Oh, I see it! Your function `lv_page_focus ` is only for Y axis, but my page have scrollbar for X axis....

zavovi, 2019-02-12:
I fixed it in your code: 

```
void lv_page_focus(lv_obj_t * page, const lv_obj_t * obj, uint16_t anim_time)
{
    lv_page_ext_t * ext = lv_obj_get_ext_attr(page);

#if USE_LV_ANIMATION == 0
    anim_time = 0;
#else
    /* Be sure there is no position changing animation in progress
     * because it can overide the current changes*/
    lv_anim_del(page, (lv_anim_fp_t)lv_obj_set_y);
    lv_anim_del(page, (lv_anim_fp_t)lv_obj_set_x);
    lv_anim_del(page, (lv_anim_fp_t)lv_obj_set_pos);
    lv_anim_del(ext->scrl, (lv_anim_fp_t)lv_obj_set_y);
    lv_anim_del(ext->scrl, (lv_anim_fp_t)lv_obj_set_x);
    lv_anim_del(ext->scrl, (lv_anim_fp_t)lv_obj_set_pos);
#endif

    lv_style_t * style = lv_page_get_style(page, LV_PAGE_STYLE_BG);
    lv_style_t * style_scrl = lv_page_get_style(page, LV_PAGE_STYLE_SCRL);

    lv_coord_t obj_y = obj->coords.y1 - ext->scrl->coords.y1;
    lv_coord_t obj_h = lv_obj_get_height(obj);
    lv_coord_t scrlable_y = lv_obj_get_y(ext->scrl);
    lv_coord_t page_h = lv_obj_get_height(page);

    lv_coord_t obj_x = obj->coords.x1 - ext->scrl->coords.x1;
	lv_coord_t obj_w = lv_obj_get_width(obj);
	lv_coord_t scrlable_x = lv_obj_get_x(ext->scrl);
	lv_coord_t page_w = lv_obj_get_width(page);

    lv_coord_t top_err = -(scrlable_y + obj_y);
    lv_coord_t bot_err = scrlable_y + obj_y + obj_h - page_h;
    lv_coord_t left_err = -(scrlable_x + obj_x);
    lv_coord_t right_err = scrlable_x + obj_x + obj_w - page_w;

    /*If obj is higher then the page focus where the "error" is smaller*/

    /*Out of the page on the top*/
    if((obj_h <= page_h && top_err > 0) ||
            (obj_h > page_h && top_err < bot_err)) {
        /*Calculate a new position and let some space above*/
        scrlable_y = -(obj_y - style_scrl->body.padding.ver - style->body.padding.ver);
        scrlable_y += style_scrl->body.padding.ver;
    }
    /*Out of the page on the bottom*/
    else if((obj_h <= page_h && bot_err > 0) ||
            (obj_h > page_h && top_err >= bot_err)) {
        /*Calculate a new position and let some space below*/
        scrlable_y = -(obj_y + style_scrl->body.padding.ver + style->body.padding.ver);
        scrlable_y -= style_scrl->body.padding.ver;
        scrlable_y += page_h - obj_h;
    }
    /*Out of the page on the left*/
    else if((obj_w <= page_w && left_err > 0) ||
            (obj_w > page_w && left_err < right_err)) {
        /*Calculate a new position and let some space above*/
        scrlable_x = -(obj_x - style_scrl->body.padding.ver - style->body.padding.ver);
        scrlable_x += style_scrl->body.padding.ver;
    }
    /*Out of the page on the right*/
    else if((obj_w <= page_w && right_err > 0) ||
            (obj_w > page_w && left_err >= right_err)) {
        /*Calculate a new position and let some space below*/
        scrlable_x = -(obj_x + style_scrl->body.padding.ver + style->body.padding.ver);
        scrlable_x -= style_scrl->body.padding.ver;
        scrlable_x += page_w - obj_w;
    } else {
        /*Already in focus*/
        return;
    }

    if(anim_time == 0) {
        lv_obj_set_y(ext->scrl, scrlable_y);
        lv_obj_set_x(ext->scrl, scrlable_x);
#if USE_LV_ANIMATION
    } else {
        lv_anim_t a;
        a.act_time = 0;
        a.start = lv_obj_get_y(ext->scrl);
        a.end = scrlable_y;
        a.time = anim_time;
        a.end_cb = NULL;
        a.playback = 0;
        a.repeat = 0;
        a.var = ext->scrl;
        a.path = lv_anim_path_linear;
        a.fp = (lv_anim_fp_t) lv_obj_set_y;
        lv_anim_create(&a);

        a.act_time = 0;
        a.start = lv_obj_get_x(ext->scrl);
        a.end = scrlable_x;
        a.time = anim_time;
        a.end_cb = NULL;
        a.playback = 0;
        a.repeat = 0;
        a.var = ext->scrl;
        a.path = lv_anim_path_linear;
        a.fp = (lv_anim_fp_t) lv_obj_set_x;
        lv_anim_create(&a);
#endif
    }
}
```

You can add it into your project.

kisvegabor, 2019-02-13:
Oh, really! 

Can you send a Pull request?

kisvegabor, 2019-03-02:
I implemented it in the `event` branch and will merge it soon to `dev-6.0`.

kisvegabor, 2019-03-02:
Oh sorry, I got confused with the PR. Of course I'll leave your version.

closed 2019-03-02 by kisvegabor