lvgl/lvgl #810: Button groups NEXT and PREV requires double pressing

MichaelVLV, 2019-02-07:
Hi.

I have created screen with 4 buttons and want to navigate between them with hardware buttons.
After implemented this task on MCU, I noticed when I'm switching from neighboring buttons it requires double pressing. I thought that this is some problem with button. But after adding debounce and totaly excluded possible hardware problems, I have recreated part of naviation on PC simulator. 
And problem repeated.

Steps to reproduce:

1. Run program
2, Press NEXT button (switch from BUTTON_1 to BUTTON_2 - ok)
3. Press NEXT button again. And now (2 to 3) it requires double pressing to switch.

I used this project: 
[https://github.com/littlevgl/pc_simulator_sdl_visual_studio](https://github.com/littlevgl/pc_simulator_sdl_visual_studio)

Whole code (can be pasted in PC simulator):

```C
/**
* @file main
*
*/

/*********************
*      INCLUDES
*********************/
#include <stdlib.h>
#include <stdio.h>
#include <Windows.h>
#include <SDL.h>
#include "lvgl/lvgl.h"
#include "lv_drivers/display/monitor.h"
#include "lv_drivers/indev/mouse.h"
#include "lv_drivers/indev/keyboard.h"
#include "lv_examples/lv_apps/demo/demo.h"
#include "lv_examples/lv_apps/benchmark/benchmark.h"
#include "lv_examples/lv_tests/lv_test_theme/lv_test_theme_1.h"
#include "lv_examples/lv_tutorial/10_keyboard/lv_tutorial_keyboard.h"

/*********************
*      DEFINES
*********************/

/**********************
*      TYPEDEFS
**********************/

/**********************
*  STATIC PROTOTYPES
**********************/
static void hal_init(void);
static int tick_thread(void *data);
static void kaypad_create(void);
static lv_res_t keypad_btn_release(lv_obj_t * btn);
static lv_res_t keypad_btn_press(lv_obj_t * btn);

/**********************
*  STATIC VARIABLES
**********************/
static lv_indev_t * kb_indev;
static lv_indev_t * emulated_kp_indev;  /*The input device of the emulated keypad*/
static uint32_t last_key = 0;
static lv_indev_state_t last_state = LV_INDEV_STATE_REL;

typedef enum {
	START_STATE = 0,

	MAIN_MENU,

	SOME_OTHER_MENU_1,
[... 229 lines cut ...]
	kb_drv.read = keyboard_read;
	kb_indev = lv_indev_drv_register(&kb_drv);
#endif

	/* Tick init.
	* You have to call 'lv_tick_inc()' in every milliseconds
	* Create an SDL thread to do this*/
	SDL_CreateThread(tick_thread, "tick", NULL);
}

/**
* A task to measure the elapsed time for LittlevGL
* @param data unused
* @return never return
*/
static int tick_thread(void *data)
{
	while (1) {
		lv_tick_inc(1);
		SDL_Delay(1);   /*Sleep for 1 millisecond*/
	}

	return 0;
}
```

embeddedt, 2019-02-07:
A note from debugging: It seems that everything always (not just one time) works as expected with buttons 1 and 2.

MichaelVLV, 2019-02-22:
Any ideas what going with buttons? Could it be that "swticher" choses labels instead of button objects?

embeddedt, 2019-02-22:
@MichaelVLV No. I'll ask @kisvegabor to comment.

kisvegabor, 2019-02-24:
Hi,

I found the issue. If you copy an object which is added to a group, the new (copied) object will be added to the same group automatically ([code](https://github.com/littlevgl/lvgl/blob/master/lv_core/lv_obj.c#L308)).

I'm not sure which way is more intuitive. (automatic adding or not). What do you think?

embeddedt, 2019-02-24:
@kisvegabor I think the automatic adding makes more sense, since one expects an exact copy of the object. What I think should be done is blocking an object from being added to the same group twice. That looks like the issue in this case.

kisvegabor, 2019-02-24:
@embeddedt I agree!

kisvegabor, 2019-03-02:
I'll fix it soon in `v6.0`

kisvegabor, 2019-03-13:
Fixed in `dev-6.0`. If still not working feel free to reopen this issue.

closed 2019-03-13 by kisvegabor