lvgl/lvgl #812: LV_INDEV_DRAG_THROW = 0 not caught

rrroonn, 2019-02-08:
Possibly an obscure situation, but setting LV_INDEV_DRAG_THROW to 0 can cause this loop in lv_tabview.c to run indefinitely.

```
    while(vect.x != 0)   {
        x_predict += vect.x;
        vect.x = vect.x * (100 - LV_INDEV_DRAG_THROW) / 100;
    }
```
Maybe something to catch at compile time.

embeddedt, 2019-02-08:
@rrroonn `LV_INDEV_DRAG_THROW` is the slow-down percentage (like friction). If it's set to `0` then it would make sense from a theoretical perspective for this loop to run forever. For all practical purposes, however, it would just seem like a bug to the user.

If @kisvegabor agrees I can add a note stating that it must be greater than 0 to `lv_conf_templ.h`.

kisvegabor, 2019-02-09:
> If @kisvegabor agrees I can add a note stating that it must be greater than 0 to `lv_conf_templ.h`.

It would be useful, thank you. In addition
- a `#warning` in `lv_indev.c` 
- and redefining `LV_INDEV_DRAG_THROW` to `1` or `#if`, `#else`, `#endif` around the offending part

would be great!

@embeddedt can you add these too?

embeddedt, 2019-02-10:
@kisvegabor I applied the first two patches.

> redefining `LV_INDEV_DRAG_THROW` to `1` or `#if`, `#else`, `#endif` around the offending part

There are [2 or 3 spots](https://github.com/littlevgl/lvgl/search?q=LV_INDEV_DRAG_THROW&unscoped_q=LV_INDEV_DRAG_THROW) where `LV_INDEV_DRAG_THROW` might potentially be a problem. Wouldn't it be better to put the redefine in `lv_conf_checker.h`?

kisvegabor, 2019-02-12:
> There are [2 or 3 spots](https://github.com/littlevgl/lvgl/search?q=LV_INDEV_DRAG_THROW&unscoped_q=LV_INDEV_DRAG_THROW) where `LV_INDEV_DRAG_THROW` might potentially be a problem.

You are right.

> Wouldn't it be better to put the redefine in `lv_conf_checker.h`?

`lv_conf_checker.h` is generated automatically from the default values in `lv_conf_templ.h`. So it's not designed to add custom code there. Rather the [end of](https://github.com/littlevgl/lvgl/blob/master/lv_conf_templ.h#L382) `lv_conf_templ.h` would be good the throw a warning an redefine the value. What do you think?

embeddedt, 2019-02-12:
@kisvegabor See #827.

kisvegabor, 2019-02-12:
@embeddedt 
Meged, thank you! 

@rrroonn 
Thank you for pointing out this issue!

closed 2019-02-12 by kisvegabor