lvgl/lvgl #813: Button not triggering a callback

ZaneL, 2019-02-08:
Hello,

I'm trying to use this library on a Teensy 3.2 with a ILI9341 display. The display is working nicely and the touch driver is registering clicks, but for some reason the callback isn't being triggered. The on-screen button also isn't being depressed, so I'm not sure what is going on!

Can you take a look at my code and see if I'm missing anything? I've been screwing with this for a long time and I'm not sure what could possibly be wrong.

`/*****************************************************************************
   Includes
 ****************************************************************************/
 
#include <SPI.h>
#include <ILI9341_t3.h>
#include <XPT2046_Touchscreen.h>
#include <lvgl.h>

/*****************************************************************************
   Defines / Global Variables
 ****************************************************************************/

#define LCD_CS_PIN 15
#define LCD_DC_PIN 20
ILI9341_t3 tft = ILI9341_t3(LCD_CS_PIN, LCD_DC_PIN);

/* Touch Controller */
#define TOUCH_CS_PIN 22
#define TOUCH_IRQ_PIN 23
XPT2046_Touchscreen ts(TOUCH_CS_PIN, TOUCH_IRQ_PIN);

IntervalTimer tickTimer;

/*****************************************************************************
   LittleVGL Related Functions
 ****************************************************************************/

 void tick_handler(void)
{
  lv_tick_inc(1);
}

void disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_array)
{
  tft.writeRect(x1, y1, 1 + x2 - x1, 1 + y2 - y1, (const uint16_t *) color_array);
  lv_flush_ready();
}

bool touchpad_read(lv_indev_data_t * data)
{
  static int16_t last_x = 0;
  static int16_t last_y = 0;

  if (ts.tirqTouched()) {
    if (ts.touched()) {
      TS_Point p = ts.getPoint();
      last_x = p.x;
      last_y = p.y;
      data->point.x = last_x;
      data->point.y = last_y;
      data->state = LV_INDEV_STATE_PR;

      return false;
    }
  }

  data->point.x = last_x;
  data->point.y = last_y;
  data->state = LV_INDEV_STATE_REL;

  return false;
}

void hal_init()
{
  lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv);
  disp_drv.disp_flush = disp_flush;
  lv_disp_drv_register(&disp_drv);

  lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  indev_drv.read = touchpad_read;
  lv_indev_drv_register(&indev_drv);
}

static lv_res_t my_click_action(struct _lv_obj_t * obj) 
{
  Serial.println("Button Clicked!");
  return LV_RES_INV;
}
 
static void click_button(void) 
{
  /*Add a button*/
  lv_obj_t *btn1 = lv_btn_create(lv_scr_act(), NULL);
  lv_obj_set_pos(btn1, 2, 2);
  lv_obj_set_size(btn1, 200, 50);
  lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, my_click_action);
 
  /*Add text*/
  lv_obj_t *label = lv_label_create(btn1, NULL);
  lv_label_set_text(label, "Click Me!");
}

/*****************************************************************************
   Setup
 ****************************************************************************/

void setup()
{
  Serial.begin(57600);
  delay(1000);

  tickTimer.begin(tick_handler, 1000); /// 1000us = 1ms

  lv_init();
  hal_init();

  tft.begin();
  tft.setRotation(1);
  tft.fillScreen(ILI9341_WHITE);

  ts.begin();
  ts.setRotation(3);

  // Draw a button, wait for clicks
  click_button();
}

/*****************************************************************************
   Main Loop
 ****************************************************************************/

void loop()
{
  lv_task_handler();
  delay(5); // 5ms
}`

embeddedt, 2019-02-08:
I assume that IntervalTimer is a  Arduino construct. You might want to make sure that it fires in parallel with `loop()`.

ZaneL, 2019-02-08:
When I strip the tick timer stuff out and do everything in the main loop like this, it still has the same problem.

```
/*****************************************************************************
   Main Loop
 ****************************************************************************/

void loop()
{
  lv_task_handler();
  lv_tick_inc(5);
  delay(5); // 5ms
}
```

embeddedt, 2019-02-08:
@ZaneL That's pretty much guaranteed not to work.

The loop and timer need to be in **parallel.** In your case the tick won't get incremented properly if `lv_task_handler` takes more than 0 milliseconds to execute.  You should use an interrupt handler or something similar to that.

ZaneL, 2019-02-08:
Well that first code snippet with the IntervalTimer uses an interrupt-based tick, so I'm assuming that it would be fine. I'll look into it more.

ZaneL, 2019-02-08:
Figured it out. It's because that PJRC touchscreen library outputs values from about 0-4000, so I had to remap the points to values of 0-320 and 0-240 for my screen size.

I'm still working on it, but you can see an example of what needs to be done in the **xpt2046_corr()** function here.

[](https://github.com/littlevgl/lv_drivers/blob/master/indev/XPT2046.c)

closed 2019-02-08 by ZaneL

ZaneL, 2019-02-08:
Here is the working code snippet if anyone needs it:

`/*****************************************************************************
   Includes
 ****************************************************************************/

#include <SPI.h>
#include <ILI9341_t3.h>
#include <XPT2046_Touchscreen.h>
#include <lvgl.h>

/*****************************************************************************
   Defines / Global Variables
 ****************************************************************************/

#define LCD_CS_PIN 15
#define LCD_DC_PIN 20
ILI9341_t3 tft = ILI9341_t3(LCD_CS_PIN, LCD_DC_PIN);

/* Touch Controller */
#define TOUCH_CS_PIN 22
#define TOUCH_IRQ_PIN 23
XPT2046_Touchscreen ts(TOUCH_CS_PIN, TOUCH_IRQ_PIN);

IntervalTimer tickTimer;

/* Screen Setup */
#define XPT2046_HOR_RES 320
#define XPT2046_VER_RES 240
#define XPT2046_XY_SWAP 0
#define XPT2046_X_INV 0
#define XPT2046_Y_INV 0

/* Touch Screen Calibration Values */
#define XPT2046_X_MIN 200
#define XPT2046_Y_MIN 200
#define XPT2046_Y_MIN 200
#define XPT2046_X_MAX 3800
#define XPT2046_Y_MAX 3800

/*****************************************************************************
   LittleVGL Related Functions
 ****************************************************************************/

static void xpt2046_corr(int16_t * x, int16_t * y)
{
#if XPT2046_XY_SWAP != 0
  int16_t swap_tmp;
  swap_tmp = *x;
  *x = *y;
  *y = swap_tmp;
#endif

  if ((*x) > XPT2046_X_MIN)(*x) -= XPT2046_X_MIN;
  else(*x) = 0;

  if ((*y) > XPT2046_Y_MIN)(*y) -= XPT2046_Y_MIN;
  else(*y) = 0;

  (*x) = (uint32_t)((uint32_t)(*x) * XPT2046_HOR_RES) /
         (XPT2046_X_MAX - XPT2046_X_MIN);

  (*y) = (uint32_t)((uint32_t)(*y) * XPT2046_VER_RES) /
         (XPT2046_Y_MAX - XPT2046_Y_MIN);

#if XPT2046_X_INV != 0
  (*x) =  XPT2046_HOR_RES - (*x);
#endif

#if XPT2046_Y_INV != 0
  (*y) =  XPT2046_VER_RES - (*y);
#endif
}

void tick_handler(void)
{
  lv_tick_inc(1);
}

void disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_array)
{
  tft.writeRect(x1, y1, 1 + x2 - x1, 1 + y2 - y1, (const uint16_t *) color_array);
  lv_flush_ready();
}

bool touchpad_read(lv_indev_data_t * data)
{
  static int16_t last_x = 0;
  static int16_t last_y = 0;

  if (ts.tirqTouched()) {
    if (ts.touched()) {
      TS_Point p = ts.getPoint();

      /* Remap points to 320x240 screen */
      last_x = p.x;
      last_y = p.y;
      xpt2046_corr(&last_x, &last_y);

      data->point.x = last_x;
      data->point.y = last_y;
      data->state = LV_INDEV_STATE_PR;

      return false;
    }
  }

  data->point.x = last_x;
  data->point.y = last_y;
  data->state = LV_INDEV_STATE_REL;

  return false;
}

void hal_init()
{
  lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv);
  disp_drv.disp_flush = disp_flush;
  lv_disp_drv_register(&disp_drv);

  lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  indev_drv.read = touchpad_read;
  lv_indev_drv_register(&indev_drv);
}

static lv_res_t my_click_action(struct _lv_obj_t * obj)
{
  Serial.println("Button Clicked!");
  return LV_RES_INV;
}

static void click_button(void)
{
  /*Add a button*/
  lv_obj_t *btn1 = lv_btn_create(lv_scr_act(), NULL);
  lv_obj_set_pos(btn1, 2, 2);
  lv_obj_set_size(btn1, 200, 50);
  lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, my_click_action);

  /*Add text*/
  lv_obj_t *label = lv_label_create(btn1, NULL);
  lv_label_set_text(label, "Click Me!");
}

/*****************************************************************************
   Setup
 ****************************************************************************/

void setup()
{
  Serial.begin(57600);
  delay(1000);

  tickTimer.begin(tick_handler, 1000); /// 1000us = 1ms

  lv_init();
  hal_init();

  tft.begin();
  tft.setRotation(1);
  tft.fillScreen(ILI9341_WHITE);

  ts.begin();
  ts.setRotation(3);

  // Draw a button, wait for clicks
  click_button();
}

/*****************************************************************************
   Main Loop
 ****************************************************************************/

void loop()
{
  lv_task_handler();
  delay(5); // 5ms
}`

embeddedt, 2019-02-08:
@ZaneL Glad to hear you figured it out! I didn't think of the touchscreen library itself being the culprit.