lvgl/lvgl #814: I want design a screen like Grid

TejaswiGunti, 2019-02-08:
[image]
I want to design a screen like above picture.. Could you please help me in this.. I have tried using Button Matrix but could not get the view.. What objects I should use to get the above view..

kisvegabor, 2019-02-10:
have you tried to build it simply from `lv_obj` and `lv_label` objects?

TejaswiGunti, 2019-02-11:
Yeah, @kisvegabor.. I have done it through labels and objects.. But what I actually want is the grid view with all the information in single box. Is there any option to design in such a way?

kisvegabor, 2019-02-12:
You create an `lv_cont` parent and use `LV_LAYOUT_PRETTY` to arrange the items automatically.
More info: https://docs.littlevgl.com/#Container

TejaswiGunti, 2019-02-12:
Thank you.. Can we dynamically change the values in the above picture by using container object? Meanwhile I will try your suggestion..

kisvegabor, 2019-02-12:
> Can we dynamically change the values in the above picture by using container object?

You can change the text of the labels in the boxes. It's not related to the container or the layout.

TejaswiGunti, 2019-02-13:
This worked.. Thank you for your help..

kisvegabor, 2019-02-13:
Great!

closed 2019-02-13 by kisvegabor