lvgl/lvgl #816: v5.3: Night theme has wrong colors for symbol buttons

fpw, 2019-02-09:
When using the night theme and adding symbol buttons to a window (`lv_win_add_btn` with a `SYMBOL_...` constant), the symbols are now black and therefore hard to see on the dark night theme. They were white in v5.2, so I assume this is a bug and not intentional.

As a workaround, I found that adding these settings before loading the night theme helps restoring the old colors, so maybe a fix is to include these modifications in `lv_theme_night_init`:

```
lv_style_plain.image.color = LV_COLOR_MAKE(0xFF, 0xFF, 0xFF);
lv_style_pretty.image.color = LV_COLOR_MAKE(0xFF, 0xFF, 0xFF);
lv_theme_t *theme = lv_theme_night_init(210, LV_FONT_DEFAULT);
```

kisvegabor, 2019-02-10:
Hi,

Thank you for the report! I fixed in the master branch. Is it okay now?

fpw, 2019-02-10:
Almost - pressed and released look fine now, but a toggle button in the toggle state is still black.

kisvegabor, 2019-02-12:
Which toggle button do you mean? The toggle button on the first tab of `lv_test_theme_1` is good at me:
[image]

embeddedt, 2019-02-12:
@kisvegabor I think @fpw means the button created by `lv_win_add_btn`.

kisvegabor, 2019-02-12:
@embeddedt I also thought that but those buttons don't have toggled state by default.

Although you can enable "toggle" manually so it really can be an issue. I pushed a fix.

fpw, 2019-02-12:
@kisvegabor Now it behaves as expected, thank you!

kisvegabor, 2019-02-13:
Great! Thank you for the report and testing!

closed 2019-02-13 by kisvegabor