lvgl/lvgl #818: option for lv_chart_set_next

ali-rostami, 2019-02-10:
Hi
As we know, when we use the  ```lv_chart_set_next``` function, it will shift the whole chart to the left and then add the new value to the most right of the chart. I don't want to update my chart like this.
I want to draw the next value without shifting the whole chart to the left. the x of new point will start from 0 to the mx_x and then start again from point 0. to do this I change this function to this:
```
void lv_chart_set_next(lv_obj_t * chart, lv_chart_series_t * ser, lv_coord_t y)
{
	lv_chart_ext_t * ext = lv_obj_get_ext_attr(chart);
#if 0
	ser->points[ser->start_point] = y;  /*This was the place of the former left most value, after shifting it is the rightmost*/
	ser->start_point = (ser->start_point + 1) % ext->point_cnt;
#else
	static uint16_t i = 0;
	ser->points[i] = y;  /*This was the place of the former left most value, after shifting it is the rightmost*/
	i = (i + 1) % ext->point_cnt;
#endif

	lv_chart_refresh(chart);
}
```
so what do you think about it? I think it would be good to add an option in defines or maybe a parameter to set and define how the ```lv_chart_set_next``` act.

kisvegabor, 2019-02-11:
Instead of a `define` I can imagine a function to change the display mode: `lv_chart_set_mode(chart, LV_CHART_MODE_CIRCULAR/SHIFT)`. This way it can be different for each chart.

ali-rostami, 2019-02-11:
Yep, that's great.
And AFAIK now every time we add new data the whole chart is redrawn again. Although it is necessary for shift mode I think there is no need for this whole refresh in circular mode. Am I right?

kisvegabor, 2019-02-12:
> And AFAIK now every time we add new data the whole chart is redrawn again. Although it is necessary for shift mode I think there is no need for this whole refresh in circular mode. Am I right?

Yes! Instead of `lv_obj_invalidate` `lv_inv_area` can be used for the affected area.

ali-rostami, 2019-02-14:
Dear Gabor, do you have any plan for implementing these?

kisvegabor, 2019-02-15:
Right now, I'm working on the new features of `v6.0` so don't have time to add this. :(

However, seems quite straightforward. Would you like to try it? I can help if you stuck somewhere.

ali-rostami, 2019-02-15:
OK, I will work on it as soon as possible. Of course, I need your help. I will notify you about the parts I couldn't understand.
 thanks

kisvegabor, 2019-03-02:
@ali-rostami 
Were you able to make some progress with it?

ali-rostami, 2019-03-04:
not yet, I will get to this soon.

kisvegabor, 2019-03-04:
Okay. 
Can I assign you this issue?

ali-rostami, 2019-03-04:
yep, no problem.

kisvegabor, 2019-03-04:
Done, thank you!

ali-rostami, 2019-03-07:
> > And AFAIK now every time we add new data the whole chart is redrawn again. Although it is necessary for shift mode I think there is no need for this whole refresh in circular mode. Am I right?
> 
> Yes, good idea! Instead of `lv_obj_invalidate` `lv_inv_area` can be used for the affected area.

Do you mean to use it in lv_chart_refresh function?

kisvegabor, 2019-03-08:
I mean instead if `lv_chart_refresh` you can call `lv_inv_area` in `lv_chart_set_next` to invalidate only a small area.

ali-rostami, 2019-03-08:
Yeah, but the problem is I have no idea how to get the coordination of the area.
How can I get them? How can I get the mask area?

kisvegabor, 2019-03-08:
I suggest adding the feature first and optimize it later.

kisvegabor, 2019-03-24:
Let's continue it in the PR: #951

closed 2019-03-24 by kisvegabor