lvgl/lvgl #819: LV_CHART_TYPE_VERTICAL_LINE chart type

ali-rostami, 2019-02-10:
Hi 
I couldn't understand ```LV_CHART_TYPE_VERTICAL_LINE``` this chart type. can anybody tell me what is this?
thanks

kisvegabor, 2019-02-12:
It is for a special case when `chart width == point count` to draw only vertical lines because it's faster.

I ushed some fixes to `master` so please pull it before testing the chart again.

Example: 
```c
    lv_obj_t * chart = lv_chart_create(lv_scr_act(), NULL);
    lv_chart_set_type(chart, LV_CHART_TYPE_VERTICAL_LINE);
    lv_obj_set_width(chart, 100);
    lv_chart_set_point_count(chart, 100);
    lv_chart_series_t * ser = lv_chart_add_series(chart, LV_COLOR_RED);
    lv_chart_set_series_width(chart, 1);

    lv_chart_set_next(chart, ser, 30);
    lv_chart_set_next(chart, ser, 70);
    lv_chart_set_next(chart, ser, 10);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 20);
    lv_chart_set_next(chart, ser, 30);
    lv_chart_set_next(chart, ser, 70);
    lv_chart_set_next(chart, ser, 10);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 20);
    lv_chart_set_next(chart, ser, 30);
    lv_chart_set_next(chart, ser, 70);
    lv_chart_set_next(chart, ser, 10);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 20);
    lv_chart_set_next(chart, ser, 30);
    lv_chart_set_next(chart, ser, 70);
    lv_chart_set_next(chart, ser, 10);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 20);
    lv_chart_set_next(chart, ser, 30);
    lv_chart_set_next(chart, ser, 70);
    lv_chart_set_next(chart, ser, 10);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 20);
    lv_chart_set_next(chart, ser, 30);
    lv_chart_set_next(chart, ser, 70);
    lv_chart_set_next(chart, ser, 10);
    lv_chart_set_next(chart, ser, 50);
    lv_chart_set_next(chart, ser, 50);
```

[image]

ali-rostami, 2019-02-12:
thanks, yeah now I got it and It's working.

closed 2019-02-12 by ali-rostami

kisvegabor, 2019-02-12:
Great! :)