lvgl/lvgl #821: Dynamically updating of button matrix button attributes

canardos, 2019-02-11:
There is currently no way to dynamically update the inactive (or disabled, long press etc) state of a button in a button matrix without recreating the map.

The matrix is useful for automating the layout of multiple buttons, but setting a button inactive permanently at instantiation time seems of little value (in my use cases at least), and recreating the entire matrix to en/disable single buttons is expensive and requires multiple map arrays.

Because the control byte is stored in the map string which is `const` (as it should be), we can't just update `map_p[btn_idx][0]` and redraw. Even if we could, if the control char wasn't provided initially, then it wouldn't be available for updating.

As a solution, I've modified the button matrix to store the control flags in a separate byte array. To maintain backward compatibility, the control codes in the map are still used to set the initial flags, but after that the flags are modifiable with normal library functions. e.g.:

```
void lv_btnm_set_btn_hidden(const lv_obj_t * btnm, uint16_t btn_idx, bool hidden);
void lv_btnm_set_btn_disabled(const lv_obj_t * btnm, uint16_t btn_idx, bool disabled);
void lv_btnm_set_btn_disable_repeat(const lv_obj_t * btnm, uint16_t btn_idx, bool disabled);

/* OR */

void lv_btnm_set_btn_flags(const lv_obj_t * btnm, uint16_t btn_idx, bool hidden, bool disabled, bool disable_repeat);

/* AND */

void lv_btnm_set_btn_width(const lv_obj_t * btnm, uint16_t btn_idx, uint8_t width);
```

The functionality is useful for my applications. The memory cost is a byte per button, potentially offset as the control character doesn't need to be present.

Let me know if you think this would be useful for the library and I'll organize a pull request.

embeddedt, 2019-02-11:
This would be useful for the issues mentioned in #679.

kisvegabor, 2019-02-12:
@canardos Good idea! It would be more user-friendly to handle the buttons' properties this way. So yes, please, send PR!

@embeddedt It really seems like a good direction to solve other problems too with the button matrix.

kisvegabor, 2019-03-03:
The PR is merged so I close this issue!

Thank you very much @canardos!

closed 2019-03-03 by kisvegabor