lvgl/lvgl #822: BUG: lv_imgbtn_create with copy pointer

seyyah, 2019-02-12:
## :bug: Bug

Initial step is

```sh
$ git clone --recursive https://github.com/littlevgl/pc_simulator_sdl_eclipse.git
```

Related test code is

```c
1. void lv_test_imgbtn_1(void)
2. {
3.     /* Create an image button and set images for it*/
4.    lv_obj_t * imgbtn1 = lv_imgbtn_create(lv_scr_act(), NULL);
5.    lv_obj_set_pos(imgbtn1, 10, 10);
6.    lv_imgbtn_set_toggle(imgbtn1, true);
7.
8.    LV_IMG_DECLARE(imgbtn_img_1);
9.    LV_IMG_DECLARE(imgbtn_img_2);
10.    LV_IMG_DECLARE(imgbtn_img_3);
11.    LV_IMG_DECLARE(imgbtn_img_4);
12.
13    lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_REL, &imgbtn_img_1);
14.    lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_PR, &imgbtn_img_2);
15.    lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_TGL_REL, &imgbtn_img_3);
16.    lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_TGL_PR, &imgbtn_img_4);
17.
18.    lv_imgbtn_set_action(imgbtn1, LV_BTN_ACTION_CLICK, imgbtn_clicked);
19.
20.    /*Add a label*/
21.    lv_obj_t * label = lv_label_create(imgbtn1, NULL);
22.    lv_label_set_text(label, "Button 1");
23.
24.    /*Copy the image button*/
25.    lv_obj_t * imgbtn2 = lv_imgbtn_create(lv_scr_act(), imgbtn1);
26.    lv_imgbtn_set_state(imgbtn2, LV_BTN_STATE_TGL_REL);
27.    lv_obj_align(imgbtn2, imgbtn1, LV_ALIGN_OUT_RIGHT_MID, 20, 0);
28.
29.    label = lv_label_create(imgbtn2, NULL);
30    lv_label_set_text(label, "Button 2");
31. }
```

Make and run

```sh
$ make; ./demo
```

Result is

[image]

**Bug**: `imgbtn2` is "No data".

Other trials and results

- Trial 1: when L26 is commented, result is the same

- Trial 2: when `lv_imgbtn_set_state(imgbtn1, LV_BTN_STATE_TGL_PR);`, working properly and set state image

- Trial 3: when LittlevGL and lv_examples updated, result is the same

## :camera: Screenshots

Trial 1
[image]

Trial 2
[image]

## :mag: Environment
- Repo: https://github.com/littlevgl/pc_simulator_sdl_eclipse

- LittlevGL: [pc_simulator version: 5.3.0](https://github.com/littlevgl/lvgl/blob/84781c62c2a8291aacdc7abd6ae54e64a2b6b72b). [Master](https://github.com/littlevgl/lvgl/commit/c71047f35975b293362320406c6104b8edcf2b73)

- LV_Examples: [pc_simulator version](https://github.com/littlevgl/lv_examples/tree/2867539419e0af9120f87b6b917db2b2476ef3ff). [Master](https://github.com/littlevgl/lv_examples/commit/85e0d7e22014961d5186fbf2ac4b248ad9400fd9)

embeddedt, 2019-02-12:
@seyyah I think it's caused by the fact that the image source is not copied when `LV_IMGBTN_TILED == 0`:

https://github.com/littlevgl/lvgl/blob/caa2064c90d2b6938133466c3d55ac62fc613624/lv_objx/lv_imgbtn.c#L84-L92

@kisvegabor See #828.

seyyah, 2019-02-12:
@embeddedt i fix minor error. It works :clap: 

:link: Reference: https://github.com/littlevgl/lvgl/pull/828/commits/9ac933fb9e59acf90e73581d008955f466fbf800#diff-d8f3fb8baff1bcf72903cf54d621915aR85

```c
    /*Copy an existing image button*/
    else {
        lv_imgbtn_ext_t * copy_ext = lv_obj_get_ext_attr(copy);
#if LV_IMGBTN_TILED == 0
        memcpy(ext->img_src, copy_ext->img_src, sizeof(ext->img_src));
#else
        memcpy(ext->img_src_left, copy_ext->img_src_left, sizeof(ext->img_src_left));
        memcpy(ext->img_src_mid, copy_ext->img_src_mid, sizeof(ext->img_src_mid));
        memcpy(ext->img_src_right, copy_ext->img_src_right, sizeof(ext->img_src_right));
#endif
```

Test code is

```c
void lv_test_imgbtn_1(void)
{
    /* Create an image button and set images for it*/
    lv_obj_t * imgbtn1 = lv_imgbtn_create(lv_scr_act(), NULL);
    lv_obj_set_pos(imgbtn1, 10, 10);
    lv_imgbtn_set_toggle(imgbtn1, true);

    LV_IMG_DECLARE(imgbtn_img_1);
    LV_IMG_DECLARE(imgbtn_img_2);
    LV_IMG_DECLARE(imgbtn_img_3);
    LV_IMG_DECLARE(imgbtn_img_4);

    lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_REL, &imgbtn_img_1);
    lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_PR, &imgbtn_img_2);
    lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_TGL_REL, &imgbtn_img_3);
    lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_TGL_PR, &imgbtn_img_4);

    lv_imgbtn_set_action(imgbtn1, LV_BTN_ACTION_CLICK, imgbtn_clicked);

    /*Add a label*/
    lv_obj_t * label = lv_label_create(imgbtn1, NULL);
    lv_label_set_text(label, "Button 1");

    /*Copy the image button*/
    lv_obj_t * imgbtn2 = lv_imgbtn_create(lv_scr_act(), imgbtn1);
    lv_imgbtn_set_state(imgbtn2, LV_BTN_STATE_TGL_REL);
    lv_obj_align(imgbtn2, imgbtn1, LV_ALIGN_OUT_RIGHT_MID, 0, 50);

    label = lv_label_create(imgbtn2, NULL);
    lv_label_set_text(label, "Button 2");

    /*Copy the image button*/
    lv_obj_t * imgbtn3 = lv_imgbtn_create(lv_scr_act(), imgbtn1);
    lv_imgbtn_set_state(imgbtn3, LV_BTN_STATE_TGL_PR);
    lv_obj_align(imgbtn3, imgbtn1, LV_ALIGN_OUT_RIGHT_MID, 0, 100);

    label = lv_label_create(imgbtn3, NULL);
    lv_label_set_text(label, "Button 3");

    /*Copy the image button*/
    lv_obj_t * imgbtn4 = lv_imgbtn_create(lv_scr_act(), imgbtn1);
    lv_imgbtn_set_state(imgbtn4, LV_BTN_STATE_PR);
    lv_obj_align(imgbtn4, imgbtn1, LV_ALIGN_OUT_RIGHT_MID, 0, 150);

    label = lv_label_create(imgbtn4, NULL);
    lv_label_set_text(label, "Button 4");
}
```

## :camera: Screenshots
[image]

embeddedt, 2019-02-12:
Thanks for the report!

closed 2019-02-12 by embeddedt

kisvegabor, 2019-02-13:
Thank you for the report and the fix!

PR #828 (2019-02-12): Fix image source not being copied in lv_imgbtn
Fixes #822.