lvgl/lvgl #826: LABEL's Position issue

kanemsn, 2019-02-12:
Hi,

I'm porting LVGL to a stm32F103 with a ST7567(SPI driver), 128*64 resolution;

when I refresh a label, if the x coordinate is 0, it work ok, but I try a x coordinate not 0,  it appear in two position.
example, 
	label1 = lv_label_create(lv_scr_act(), NULL);
	lv_obj_set_pos(label1, 0, 4);
When I call _lv_label_set_text(label1, "date")_ periodically it display in pos (0,4);
But I try:
	label1 = lv_label_create(lv_scr_act(), NULL);
	lv_obj_set_pos(label1, 64, 4);
When I call _lv_label_set_text(label1, "date")_ periodically, it display in pos (0,4) and pos (64, 4)! 
And the update position is pos (0,4)!

I mean I can't put a lable on the  right part of screan to refresh realtime data.
How does this happen? Can't someone help me this issue?

Thank you very much

embeddedt, 2019-02-12:
@kanemsn Please make sure you're not *creating* the label multiple times with `lv_label_create`, but just updating it with `lv_label_set_text`.

If you're not, this might be an issue with your display driver. I could be wrong, but I think the problem isn't within LittlevGL itself since labels are used by pretty much every user of this library in various positions on the screen.

kanemsn, 2019-02-18:
@embeddedt Thank a lot! I had fix this issue, it's the display driver mistake。

Thank you very much！

closed 2019-02-18 by kanemsn