lvgl/lvgl #830: roller object rolling flag

ali-rostami, 2019-02-12:
Hi
I have a roller object which is refreshed every time a new data arrives from uart. -the color of the selected item might be changed based on the received data- as the rate of receiving data is rather high, It doesn't let the user roll the roller. -it rolls a little bit but before getting stable and changing the selected item it gets back because of the refresh command comes after the new data receives.-
I think this can be solved if I can get whether the roller is being rolled or not. somehow a rolling flag. If this flag is set I can prevent the roller from being refreshed.
 so what do you think? Is there any kind of flag for this? or how can I solve this problem?

kisvegabor, 2019-02-13:
How do you refresh the style of roller? With `lv_roller_set_style`?

If yes, and you change only the color and no padding or fonts then you can just modify the style and call `lv_obj_invalidate(roller)`. It just redraws the objects with the new color.

ali-rostami, 2019-02-13:
great, It works. thanks.

closed 2019-02-13 by ali-rostami