lvgl/lvgl #831: How to set selected certain button in list with lv_list_set_btn_selected ?

MichaelVLV, 2019-02-12:
Hi.

I'm trying to set button 2 selected by defaut:
[image]

But selection goes from top always. Please advice what am I missing? 

Code:
```C
static void scr_sel_menu(menu_scr_t group_idx)
{
	lv_obj_t * scr = lv_obj_create(NULL, NULL);
	lv_scr_load(scr);

	lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
	lv_label_set_text(label, "List header");
	lv_obj_align(label, NULL, LV_ALIGN_IN_TOP_MID, 0, 0);

	//Create the list
	lv_obj_t * list = lv_list_create(lv_scr_act(), NULL);
	lv_obj_set_size(list, 160, 98);
	lv_obj_align(list, NULL, LV_ALIGN_IN_TOP_MID, 0, 30);
	lv_list_set_sb_mode(list, LV_SB_MODE_OFF);

	//List button1
	lv_obj_t * list_btn_1 = lv_list_add(list, NULL, "list_btn_1", sel_action);
	lv_obj_set_free_num(list_btn_1, 1);
	lv_btn_set_fit(list_btn_1, false, false);
	lv_obj_set_size(list_btn_1, 120, 20);

	//List button2
	lv_obj_t * list_btn_2 = lv_list_add(list, NULL, "list_btn_2", sel_action);
	lv_obj_set_free_num(list_btn_2, 2);
	lv_btn_set_fit(list_btn_2, false, false);
	lv_obj_set_size(list_btn_2, 120, 20);

	//List button3
	lv_obj_t * list_btn_3 = lv_list_add(list, NULL, "list_btn_3", sel_action);
	lv_obj_set_free_num(list_btn_3, 3);
	lv_btn_set_fit(list_btn_3, false, false);
	lv_obj_set_size(list_btn_3, 120, 20);

	lv_list_set_btn_selected(list, list_btn_2); // Should be selected button2 by default?

	lv_group_add_obj(custom_groups[group_idx], list);
}
```

embeddedt, 2019-02-12:
@MichaelVLV I'm not that familiar with the group system but maybe you need `lv_group_focus_obj`.

MichaelVLV, 2019-02-12:
I tried to replace 
```C
lv_list_set_btn_selected(list, list_btn_2)
```
 with 
```C
lv_group_focus_obj(list_btn_2);
```

but still same - selected button 1 on start.

kisvegabor, 2019-02-12:
There was bug in `lv_list_set_btn_selected()`. I pushed the fix to `master`. It is working now?

MichaelVLV, 2019-02-13:
@kisvegabor yes, now it working. Thanks for quick fix 👍

embeddedt, 2019-02-13:
@MichaelVLV Great! Glad to hear it works now.

closed 2019-02-13 by embeddedt