lvgl/lvgl #832: ddlist opens behind other objects

rrroonn, 2019-02-13:
Hi,

I have created a page to edit some data.  On that page I have multiple fields that are created in the order they appear from top to bottom.  I have a dropdown list for one of the fields.  When the item is clicked, it opens to display the available options, but they are partly obscured by the fields below as it is drawn below the objects created further down the page.

I would expect that a dropdown would draw over the top of other objects. 

I can probably work around this by reordering which objects are created when, but this seems unnecessary complexity.  

Is there a way to correct this behaviour without reordering object creation, please?

thanks,

Ron

```
// ==========================================================
void	relayEdit(uint8_t r)
{
	uint16_t	row=20;

	lv_obj_t *screen = lv_scr_act();
	editing = r;

	page = lv_page_create(lv_scr_act(), NULL); // create a page with screen as parent
    lv_obj_set_size(page, lv_obj_get_width(screen), lv_obj_get_height(screen));

	lv_page_set_sb_mode(page, LV_SB_MODE_AUTO);

// NAME

	lv_obj_t * nameLabel = lv_label_create(page,0);
	lv_label_set_static_text(nameLabel,"Name:");
	lv_obj_set_pos(nameLabel, COL_1, row);
	//
	nameEdit = lv_ta_create(page,0);
	lv_ta_set_text(nameEdit,_Relays[editing].zname);
	lv_ta_set_cursor_type(nameEdit, LV_CURSOR_NONE);
	lv_ta_set_one_line(nameEdit, true);
	lv_ta_set_max_length(nameEdit,MAX_RELAY_NAME);
	lv_obj_set_pos(nameEdit, COL_2, row);
	lv_obj_set_free_num(nameEdit,0);  // flag as number field

	lv_ta_set_action(nameEdit,edit_action);
	row+=ROW_HT+lv_obj_get_height(nameEdit);

// TYPE
	lv_obj_t * typeLabel = lv_label_create(page,0);
	lv_label_set_static_text(typeLabel,"Type:");
	lv_obj_set_pos(typeLabel, COL_1, row);
	//
	typeList = lv_ddlist_create(page, NULL);
	lv_ddlist_set_options(typeList, "toggle\nmomentary");
	lv_ddlist_set_selected(typeList,_Relays[editing].ztype);
	lv_obj_set_pos(typeList, COL_2, row);
	lv_obj_set_free_num(typeList, 1);               /*Set a unique ID*/
	lv_ddlist_set_anim_time(typeList, 0);
	row+=ROW_HT+lv_obj_get_height(typeList);

// TIME

	lv_obj_t * timeLabel = lv_label_create(page,0);
	lv_label_set_static_text(timeLabel,"Seconds:");
	lv_obj_set_pos(timeLabel, COL_1, row);
	//
	timeEdit = lv_ta_create(page,0);
	char	_temp[10];
	itoa(_Relays[editing].zseconds,_temp,10);
	lv_ta_set_text(timeEdit,_temp);
	lv_ta_set_max_length(timeEdit,5);
	lv_ta_set_one_line(timeEdit, true);
	lv_ta_set_cursor_type(timeEdit, LV_CURSOR_NONE);
	lv_obj_set_pos(timeEdit, COL_2, row);
	lv_ta_set_action(timeEdit,edit_action);
	lv_obj_set_free_num(timeEdit,1);  // flag as number field
	row+=ROW_HT+lv_obj_get_height(timeEdit);

// SHOW ON PAGE

// Buttons

	lv_obj_t *	can = lv_btn_create(page,0);
	lv_obj_set_free_num(can,0);
	lv_btn_set_action(can, LV_BTN_ACTION_CLICK, btn_click_action);
	lv_obj_t  * lcan = lv_label_create(can,0);
	lv_label_set_text(lcan,"Cancel");
	lv_obj_align(can,page,LV_ALIGN_IN_BOTTOM_LEFT,25,-10);

	lv_obj_t *	ok  = lv_btn_create(page,0);
	lv_obj_set_free_num(ok,1);
	lv_btn_set_action(ok, LV_BTN_ACTION_CLICK, btn_click_action);
	lv_obj_t  * lok = lv_label_create(ok,0);
	lv_label_set_text(lok,"Ok");
	lv_obj_align(ok,page,LV_ALIGN_IN_BOTTOM_RIGHT,-25,-10);
}
```

kisvegabor, 2019-02-13:
You can use `lv_obj_set_top(ddlist, true);`. It will bring the `ddlist` into the foreground when clicked.

rrroonn, 2019-02-13:
Great,  thank you.

closed 2019-02-13 by rrroonn