lvgl/lvgl #833: Help with containing focus to edited field - keyboard etc

rrroonn, 2019-02-13:
Hi,
I am trying to create a page that allows a user to edit some configuration items.  The page contains multiple objects, including ddlist and textareas.  I use popup keyboards for text areas.

The problem I have struck is that if a use clicks on a text area, this opens a (say alphanum) keyboard.  If the user clicks on another area, this opens another different (say number) keyboard.  The user can click on a ddlist and that opens up - gets very messy very quickly.

Is there a simple way to keep focus on the currently editing object.  For example, if a ddlist is open, then only react to clicks within that object.  Similarly with the keyboard - if a keyboard is open, then only react to clicks/touch within the keyboard?

I hope this makes sense,

thank you,

Ron

kisvegabor, 2019-02-13:
Hi,

There is no ready-to-use solution for this issue. Do you know how is it solved in other GUI libraries?

I can suggest three things:
1. In the `action` of an object (e.g. `ddlist`) apply `lv_obj_set_click(ta, false)` on other objects to prevent their clicking. Note that `ddlist` and `ta` are `page` in the under the hood and you need to use `lv_obj_set_click(lv_page_get_scrl(ta), false)` too.
2. In the `action` of the objects manually close/hide the other objects. Not elegant nor flexible but should work.
3. If something is opened (ddlist, keyboard etc) create a transparent screensized `lv_obj` obejcts. It's purpuse is to absorbe the clicks. Move the "active" elements to the transpraent object with `lv_obj_set_parent`.

rrroonn, 2019-02-13:
This is the first embedded GUI I have used, so my experience is limited.

I do know that other GUIs have the concept of modality - where a window stays on top and eats all input.  I was thinking about this a bit, and I think that there are a couple of solutions.

1.  Implement an input filter that only allows clicks within an active rectangle to pass through.  If a window (such as a keyboard) needs to be modal, it would do a modal open, and create that filter.  It would have some complications such as if the modal window moves but not insurmountable.
2. You could register windows with the input layer such that it knows the starting point for farming out input events.  Eg, a modal window would insert itself so that no prior windows/objects would receive clicks, only its children. Or the input task would trace the stack of windows/objects and work out the highest modal window to allocate events to.

I will look at the workarounds you have proposed.   I have done something like this (3) for keyboards, but not sure how to do this for ddlist.

I do think that the GUI should be responsible for closing the ddlist if there is a click outside its bounds.  Similarly for anything else that the GUI changes in response to clicks - as opposed to the user adding a popup etc.

thanks for your input.

Ron

embeddedt, 2019-02-13:
@rrroonn @kisvegabor I did some experimenting with keeping track of mouse focus. You can test it out with the [`input_block`](https://github.com/littlevgl/lvgl/tree/input_block) branch. I added a new signal called `LV_SIGNAL_PRESS_LEAVE` that is triggered when the user moves to a new object.

One thing I noticed is that we already have the `LV_SIGNAL_FOCUS` and `LV_SIGNAL_DEFOCUS` signals, but they're only used for handling groups at the moment. Since this is `v6.0`, I think we should consider expanding those signals to cover all input devices, not just group handling. This would take care of problems like what @rrroonn pointed out:

> I do think that the GUI should be responsible for closing the ddlist if there is a click outside its bounds.

rrroonn, 2019-02-14:
Sounds like a good plan.  I will test if I get a chance.

kisvegabor, 2019-02-15:
@embeddedt 
> One thing I noticed is that we already have the `LV_SIGNAL_FOCUS` and `LV_SIGNAL_DEFOCUS` signals, but they're only used for handling groups at the moment. Since this is `v6.0`, I think we should [...]

Awesome idea! In `v6.0` we can trigger `LV_ACTION_FOCUS` and `LV_ACTION_DEFOCUS` actions to notify the user. Let's postpone it until the basic structure of the general action handling is added.

Finally, it would result in a similar approach than my 2) [suggestoin](https://github.com/littlevgl/lvgl/issues/833#issuecomment-463215588): "In the `action` of the objects manually close/hide the other objects."

embeddedt, 2019-02-15:
> In `v6.0` we can trigger `LV_ACTION_FOCUS` and `LV_ACTION_DEFOCUS` actions to notify the user.

The problem I see with using actions is that it's manual, i.e., requires the user to do the closing of objects themselves. I was hoping to integrate functionality like this into the objects themselves.

kisvegabor, 2019-02-18:
@rrroonn 
Do you think on `DEFOCUS` automatically close the `ddlist`? If yes, it really can work and would be convenient. 

I've checked the list the existing objects types and as I see only the ddlist is affected.

rrroonn, 2019-02-18:
@kisvegabor yes, I think that's appropriate.

kisvegabor, 2019-03-08:
Can you test it with the dev-6.0 branch? You can close the ddlist in the LV_EVENT_DEFOCUS event.

Here is some info about using the dev-6.0 branch: https://github.com/littlevgl/lvgl/issues/785#issuecomment-470829577

kisvegabor, 2019-03-24:
Due to inactivity, I close this issue. Feel free to reopen it if you have comments.

closed 2019-03-24 by kisvegabor