lvgl/lvgl #835: Switch callback function only called once

harteware07, 2019-02-13:
Hello again :)

i notice a bug in `lv_sw_set_action(sw[i], cb_options);`
The callback function is only called once and then never again! To fix this for now, in the callback function i call ``lv_sw_set_action(btn, cb_options);`` again. then it works. Is the value cleared when callback is called maybe?

kisvegabor, 2019-02-13:
It's working for me

```c
    lv_obj_t * sw = lv_sw_create(lv_scr_act(NULL), NULL);
    lv_sw_set_action(sw, sw_action);
```
```c
lv_res_t sw_action(lv_obj_t * sw)
{
    printf("Switch!\n");
    return LV_RES_OK;
}
```

Is this minimal example works there?

harteware07, 2019-02-13:
yes the minimal example works. however my problem is still happening. i create the switches here:
`	for(uint8_t i = 0; i < SCENE_EDIT_BUTTONS_CNT; i++) {
		sw[i] = lv_sw_create(bg_edit, NULL);
		lv_obj_set_size(sw[i], 42, 22);
		lv_sw_set_style(sw[i], LV_SW_STYLE_BG, &bg_style);
		lv_sw_set_style(sw[i], LV_SW_STYLE_INDIC, &indic_style);
		lv_sw_set_style(sw[i], LV_SW_STYLE_KNOB_ON, &knob_style);
		lv_sw_set_style(sw[i], LV_SW_STYLE_KNOB_OFF, &knob_style);
		lv_obj_set_free_num(sw[i], (1 << i));
		lv_sw_set_action(sw[i], gui_scenes_edit_options);

		lv_obj_align(sw[i], NULL, LV_ALIGN_IN_TOP_LEFT, 405, 55 + (i * labels_y_pad));
	#if LVGL_VERSION_MAJOR == 5 && LVGL_VERSION_MINOR >= 3
		lv_sw_set_anim_time(sw[i], 150);
	#endif
	}`
then in the cb function i have to set it again or it wont work.
`lv_res_t gui_scenes_edit_options(lv_obj_t * btn) {

	uint8_t id = lv_obj_get_free_num(btn);

	if(lv_sw_get_state(btn) == 0) {
		sceneTmp.settings.options |= id;
	} else {
		sceneTmp.settings.options &= ~id;
	}
	lv_sw_set_action(btn, gui_scenes_edit_options);
	return 0;
}`

any ideas?

embeddedt, 2019-02-13:
@harteware07 Can you put a `printf` or a breakpoint in the callback to verify that the callback isn't called a second time?

I tried a minor variation of this code just now and it worked properly.

harteware07, 2019-02-14:
i will do and report back!

harteware07, 2019-02-15:
So i checked and the callback function is only called once, when the slider is first moved.

embeddedt, 2019-02-15:
@harteware07 Hmm. This is odd behavior; I haven't seen it before. Can you create a minimal example that I can try in the `pc_simulator`? It will help me reproduce the issue.

kisvegabor, 2019-03-02:
In the past few days [LittlevGL's Forum](https://forum.littlevgl.com) has launched to discuss questions there. See this [Announcment](https://forum.littlevgl.com/viewtopic.php?f=3&t=5).

If the issue still persists please open a topic for it on the Forum.

embeddedt, 2019-03-07:
Appears to have been reported on the forum: https://forum.littlevgl.com/viewtopic.php?p=138#p138

kisvegabor, 2019-03-24:
I've tested it with `lv_test_group1()` using, mouse, keyboard and encoder but all worked well for me.

embeddedt, 2019-04-04:
I haven't run into the problem during my testing.

@harteware07 Is it still broken for you with the latest `master` branch?

kisvegabor, 2019-04-10:
I close this issue in lack of activity. If someone investigated it feel free to comment here and reopen the issue.

closed 2019-04-10 by kisvegabor

embeddedt, 2019-04-19:
It's been reported again.

https://forum.littlevgl.com/viewtopic.php?f=12&t=98&p=750#p750

With three reports I'm now pretty sure there must be an issue.

kisvegabor, 2019-04-20:
Certainly yes. But we need to able to repoduce it.

kisvegabor, 2019-04-22:
Probably it caused the issue: https://forum.littlevgl.com/viewtopic.php?p=756#p756

harteware07, 2019-04-24:
I will check if it fixes it and report back!

harteware07, 2019-05-21:
I created a simple array of sliding switches. the callback is only called if you click the switches slowly. 
I then added "return LV_RES_OK;" to the end of my callback and that fixed it! was always returning 0 in my callbacks.

embeddedt, 2019-05-21:
@harteware07 Ah, that makes sense. 0 is defined as `LV_RES_INV`, which should only be returned if the object was deleted.