lvgl/lvgl #837: Screen bad SSD1963

autoprog, 2019-02-13:
Hi,
problems scren, LCD 7" with SSD1963, 16bits interface/color.
based it: https://www.elecfreaks.com/wiki/index.php?title=7.0%22_Width_800*480_TFT_LCD_Module

[image]

[image]

infos:
#define LV_COLOR_DEPTH     16 
 ssd1963_cmd(0xf0);            //SET pixel data I/F format=16bit
 ssd1963_data(0x03);
ssd1963_cmd(0x3a);      //RGB pixel format
ssd1963_data(0x50);
 ssd1963_cmd(0x36);     	 //SET address mode=flip vertical
 ssd1963_data(0x00);

thanks,
Carlos.

embeddedt, 2019-02-13:
@autoprog Do the screen dimensions in `lv_conf.h` and your driver match?

autoprog, 2019-02-13:
Hi, yes 800x480.
thanks,
Carlos.

autoprog, 2019-02-14:
Hi,
Solved problems!
write array length.

void lv_par_write_array( uint16_t* data_out, uint16_t len ){
from
for ( i = 0; i < (len+1); i++) {....

to
for ( i = 0; i < len; i++) ....

}

thanks,
Carlos.

closed 2019-02-14 by autoprog

kisvegabor, 2019-02-15:
Great! :)