lvgl/lvgl #838: Layout driving me crazy - maybe I am nuts

rrroonn, 2019-02-14:
I am trying to simply form creation but it is driving me nuts trying to get fields to stay where I put them.  Maybe I have done it wrong, but after hours of trying, I am at a loss.  Assistance appreciated.

This is how the form is created:
```
// =====================================================================================
form_t *formCreate(char *heading, uint16_t col0, uint16_t col1, uint16_t rowht,form_action_t callback)
{
form_t	*f = pvPortMalloc(sizeof(form_t));

	f->callback = callback;
	f->page = lv_win_create(lv_scr_act(), NULL);
	lv_win_set_title(f->page, heading);
	lv_obj_t * close = lv_win_add_btn(f->page, SYMBOL_CLOSE, btn_click_action);
	lv_obj_t * ok 	 = lv_win_add_btn(f->page, SYMBOL_OK,    btn_click_action);

	lv_obj_set_free_ptr(close,f);
	lv_obj_set_free_ptr(ok,f);
	lv_obj_set_free_num(ok,1);

	f->cover = 0;

	lv_obj_set_size(f->page,LV_HOR_RES,LV_VER_RES);
	lv_win_set_layout(f->page,LV_LAYOUT_OFF);

	for (int i=0; i<FORM_MAX_COLS; i++) f->col[i] = 0;
	f->columns = 2;
	f->col[0]=col0;
	f->col[1]=col1;
	f->first_row = FORM_FIRST_ROW;

	f->row_height = rowht;
	f->fcount = 0;
	f->fields = 0;
	f->saved = 0;
	f->old_kb = 0;

	return f;
}
```
On the form I write some text using the following functions:
```
// =====================================================================================
static void At(form_t *form, lv_obj_t * fld, uint16_t row, uint16_t col)
{
	lv_obj_set_pos(fld, form->col[col], form->first_row+(form->row_height*row));
}
// =====================================================================================
void formAtSay(form_t *form, uint16_t row, uint16_t colmn, const char *label)
{
	lv_obj_t * l = lv_label_create(form->page,0);
	lv_label_set_static_text(l,label);
	At(form,l,row,colmn);
}

```
Now add some test code:
```

	form_t * form = formCreate(/*heading*/buff,/*col0*/30,/*col1*/130,/*rowht*/60,/*accept_action*/form_accept);
	formAddColumn(form,/*col2*/400);
	formAddColumn(form,/*col3*/600);

	formAtSay(form,0,0,"R0C0");
	formAtSay(form,0,1,"R0C1");
	formAtSay(form,0,2,"R0C2");
	formAtSay(form,0,3,"R0C3");

	formAtSay(form,1,0,"R1C0");
	formAtSay(form,1,1,"R1C1");
	formAtSay(form,1,2,"R1C2");
	formAtSay(form,1,3,"R1C3");

	formAtSay(form,2,0,"R2C0");
	formAtSay(form,3,1,"R3C1");
	formAtSay(form,4,2,"R4C2");
	formAtSay(form,5,3,"R5C3");
	return;

```
and this is what it looks like:

[image]

Can you see why it is so?

thank you,

Ron

Note, I have tries all different  object types and they all equally drive me nuts.

rrroonn, 2019-02-14:
OK, so its a fit issue.  Added:

```
	lv_cont_set_fit(lv_page_get_scrl(lv_win_get_content(f->page)) ,false,false);
	lv_obj_set_size(f->page,LV_HOR_RES,LV_VER_RES);
	lv_obj_set_size(lv_page_get_scrl(lv_win_get_content(f->page)),LV_HOR_RES,LV_VER_RES);

```
I had previously tried:

`	lv_cont_set_fit(lv_page_get_scrl(f->page), false, false);
` 

Also, maybe its a personal preference, but I think that setting a window size like

`	lv_win_set_size(win,size_x,size_y);
`
should be enough to set a fixed size window.  Having to do this seems excessive:

`lv_obj_set_size(lv_page_get_scrl(lv_win_get_content(win)),x,y);`

Also, you now have to do lots of calculations to try to get the 'content' size correct if you want it to just fill the screen or be a set size.   

Firstly,  have to work out how big the header is - which means grabbing the .ext member and working with that directly.  I honestly think that it should be simpler.

embeddedt, 2019-02-14:
@rrroonn `lv_obj_set_size(win, xsize, ysize)` should work. Can you show a small sample where it doesn't work?

rrroonn, 2019-02-14:
```
form_t *formCreate(char *heading, uint16_t col0, uint16_t col1, uint16_t rowht,form_action_t callback)
{
form_t	*f = pvPortMalloc(sizeof(form_t));

	f->callback = callback;
	f->page = lv_win_create(lv_scr_act(), NULL);
	sWINDOW.body.grad_color = sWINDOW.body.main_color = LV_COLOR_GREEN;

	lv_win_set_style(f->page,LV_PAGE_STYLE_SCRL ,&sWINDOW);
	lv_win_set_style(f->page,LV_PAGE_STYLE_BG ,&sWINDOW_HEADING);

	lv_win_set_title(f->page, heading);
	lv_obj_t * close = lv_win_add_btn(f->page, SYMBOL_CLOSE, btn_click_action);
	lv_obj_t * ok 	 = lv_win_add_btn(f->page, SYMBOL_OK,    btn_click_action);

	f->cover = 0;
	lv_cont_set_fit(lv_page_get_scrl(lv_win_get_content(f->page)) ,false,false);
	lv_obj_set_size(f->page,LV_HOR_RES,LV_VER_RES);
	lv_win_set_layout(f->page,LV_LAYOUT_OFF);
```
produces this (which is too small for all fields):

[image]

```
form_t *formCreate(char *heading, uint16_t col0, uint16_t col1, uint16_t rowht,form_action_t callback)
{
form_t	*f = pvPortMalloc(sizeof(form_t));
	f->callback = callback;
	f->page = lv_win_create(lv_scr_act(), NULL);

	sWINDOW.body.grad_color = sWINDOW.body.main_color = LV_COLOR_GREEN;

	lv_win_set_style(f->page,LV_PAGE_STYLE_SCRL ,&sWINDOW);
	lv_win_set_style(f->page,LV_PAGE_STYLE_BG ,&sWINDOW_HEADING);

	lv_win_set_title(f->page, heading);
	lv_obj_t * close = lv_win_add_btn(f->page, SYMBOL_CLOSE, btn_click_action);
	lv_obj_t * ok 	 = lv_win_add_btn(f->page, SYMBOL_OK,    btn_click_action);

	lv_obj_set_free_ptr(close,f);
	lv_obj_set_free_ptr(ok,f);
	lv_obj_set_free_num(ok,1);

	f->cover = 0;
	lv_cont_set_fit(lv_page_get_scrl(lv_win_get_content(f->page)) ,false,false);
	lv_obj_set_size(f->page,LV_HOR_RES,LV_VER_RES);
	lv_obj_set_size(lv_page_get_scrl(lv_win_get_content(f->page)),LV_HOR_RES,LV_VER_RES);
	lv_win_set_layout(f->page,LV_LAYOUT_OFF);

```
produces:

[image]

Yes, the page is now too big, but all fields can be seen.

You can see in the first example just using `lv_obj_set_size(f->page,LV_HOR_RES,LV_VER_RES);`
 that the content area is too small to fit all the fields. 

Ron
 
Ignore the funky colours - jut me trying to debug the issue.

embeddedt, 2019-02-14:
I did some experimenting. I think there is an issue with the way the window page is being sized. I think the following example should work, but it doesn't:

```c
static void test_win(void)
{
	lv_obj_t * win = lv_win_create(lv_scr_act(), NULL);

	lv_win_set_title(win, "Title");
	lv_obj_t * close = lv_win_add_btn(win, SYMBOL_CLOSE, NULL);
	lv_obj_t * ok 	 = lv_win_add_btn(win, SYMBOL_OK,    NULL);

	lv_obj_set_size(win,LV_HOR_RES,LV_VER_RES);
	lv_win_set_layout(win,LV_LAYOUT_OFF);
	lv_obj_t * label = lv_label_create(win, NULL);
	lv_obj_set_pos(label, 100, 50);
}
```

I can adjust the X coordinate of the label as expected but the Y-coordinate doesn't change no matter what value I put.

kisvegabor, 2019-02-18:
You need to disable to disable the `fit` property to enable explicitly `y` positioning. See the difference:

```c
lv_obj_t * win = lv_win_create(lv_scr_act(), NULL);

    lv_win_set_title(win, "Title");
    lv_obj_t * close = lv_win_add_btn(win, SYMBOL_CLOSE, NULL);
    lv_obj_t * ok    = lv_win_add_btn(win, SYMBOL_OK,    NULL);

    lv_obj_set_size(win,LV_HOR_RES,LV_VER_RES);
    lv_win_set_style(win, LV_WIN_STYLE_BG, &lv_style_pretty);   /*White background*/
    lv_win_set_style(win, LV_WIN_STYLE_CONTENT_BG, &lv_style_transp_tight);
    lv_win_set_style(win, LV_WIN_STYLE_CONTENT_SCRL, &lv_style_plain_color);  /*Blue contetn area*/

    lv_obj_t * label = lv_label_create(win, NULL);
    lv_obj_set_pos(label, 100, 50);
```

[image]

---

```c
    lv_obj_t * win = lv_win_create(lv_scr_act(), NULL);

    lv_win_set_title(win, "Title");
    lv_obj_t * close = lv_win_add_btn(win, SYMBOL_CLOSE, NULL);
    lv_obj_t * ok    = lv_win_add_btn(win, SYMBOL_OK,    NULL);

    lv_obj_set_size(win,LV_HOR_RES,LV_VER_RES);
    lv_win_set_style(win, LV_WIN_STYLE_BG, &lv_style_pretty);  /*White background*/
    lv_win_set_style(win, LV_WIN_STYLE_CONTENT_BG, &lv_style_transp_tight);
    lv_win_set_style(win, LV_WIN_STYLE_CONTENT_SCRL, &lv_style_plain_color);   /*Blue contetn area*/
    lv_obj_t * page = lv_win_get_content(win);
    lv_page_set_scrl_fit(page, false, false);
    lv_page_set_scrl_width(page, LV_HOR_RES);
    lv_page_set_scrl_height(page, LV_VER_RES - 80);
   
    lv_obj_t * label = lv_label_create(win, NULL);
    lv_obj_set_pos(label, 100, 50);
```

[image]

rrroonn, 2019-02-23:
FYI:

To make the content part of the window fit the screen, you need to do some maths:
```
	lv_obj_t *win = lv_win_create(lv_scr_act(), NULL);
	lv_obj_t * page = lv_win_get_content();
	lv_obj_t * scroll = lv_page_get_scrl(page);
	uint16_t padding = 2 * page->style_p->body.padding.inner; 
	lv_obj_set_size(scroll,lv_obj_get_width(page)-padding,lv_obj_get_height(page)-padding);

```

I think it would be useful to have the lv_window include a function something like

```
void  lv_win_maximise_content_area(lv_obj_t *win)
{
  lv_win_ext_t * ext = lv_obj_get_ext_attr(win);
  lv_obj_t * scrl = lv_page_get_scrl(ext->page);
  uint16_t padding = 2 * page->style_p->body.padding.inner; 
  lv_obj_set_size(scrl ,lv_obj_get_width(ext->page)-padding,lv_obj_get_height(ext->page)-padding);
}
```
just a thought.

embeddedt, 2019-02-23:
@kisvegabor @rrroonn I think this is how the window object should ideally behave:

1. The minimum size of the content area should be the size of the window.
2. Once an object is created outside the visible area, then the content area grows to allow scrolling.
3. If that object is later deleted, then the content area resizes down to the size of the window.

I think that the reason @rrroonn ran into problems is because 1 is not true. Am I correct or did I miss something?

rrroonn, 2019-02-23:
Yes, @embeddedt, that is correct. The content area was too small even though the winow itself was full screen size.

kisvegabor, 2019-02-24:
There were always issues because of `fit`: "positioning is not working",  "layout is not working" etc. So based on @embeddedt's suggestion I figured out a concept about how to make `fit` more intuitive and still more versatile.

If introduced 
```c
typedef enum {
    LV_FIT_NONE,         /*Do not change the size automatically*/
    LV_FIT_TIGHT,        /*Involve the children*/
    LV_FIT_FLOOD,        /*Align the size to the parent's edge*/
    LV_FIT_FILL,         /*Align the size to the parent's edge first but if there is an object out of it then involve it*/
}lv_fit_t;
```

* `LV_FIT_TIGHT` replaces teh current fit behavor to involve all children
* `LV_FIT_FLOOD` expands the size to the edge of the parent. I think it will be useful e.g. to make the right side a form's fields to expand to the right side of the parent automatically.
* `LV_FIT_FILL` combination of `LV_FIT_TIGHT` and `LV_FIT_FLOOD`. FLOOD until smaller then parent size, but TIGHT if there is children out of the parent.

The new `fit` modes can be applied to left, right, top, bottom directions separatly by  
`lv_cont_fit4(cont, LV_FIT_TIGHT, LV_FIT_FILL, LV_FIT_NONE, LV_FIT_TIGHT, )`

Or horizontally/vertically with  `lv_cont_fit2(cont, LV_FIT_TIGHT, LV_FIT_FILL)`

Or for all directiony:  `lv_cont_fit(cont, LV_FIT_TIGHT)`

For `lv_page` (the content area of `lv_win`) `lv_cont_set_fit(ext->scrl, LV_FIT_FILL)` is applied to allow expanding in all directions but keep at least the window's size.

It works like this:
[image]

There is one case when the behavior is littlebit strange. It's beacuse the `FLOOD` mode sets the size to the edge of the parent.
[image]

The changes are in the `new_fit` branch.

What do you think about it?

embeddedt, 2019-02-24:
@kisvegabor The concept looks great! I'll test it as soon as I get a chance.

I wouldn't worry about the weird behavior; I've had that problem occur with draggable HTML objects as well. I think it's not a major problem.

kisvegabor, 2019-02-24:
> The concept looks great! I'll test it as soon as I get a chance.
Great!

> I wouldn't worry about the weird behavior; I've had that problem occur with draggable HTML objects as well. I think it's not a major problem.

I agree.

kisvegabor, 2019-03-08:
The new "fit" system seems working so I close this issue.

If you find a big feel free to reopen it!

closed 2019-03-08 by kisvegabor