lvgl/lvgl #839: How to trigger lv_task_handler() only when user interface changes?

DevriesL, 2019-02-14:
In the current reference design, we call lv_task_handler() periodically every few milliseconds, but this waste some cpu resource and not good for battery sensitive devices, if there any method to trigger lv_task_handler only when the lvgl user interface changes? I think maybe we can add a hook in some draw function?

Thanks.

kisvegabor, 2019-02-15:
Have a look the "Task handling" section here: https://docs.littlevgl.com/#Porting

It describes how to sleep / wake-up the MCU.

DevriesL, 2019-02-15:
> Have a look the "Task handling" section here: https://docs.littlevgl.com/#Porting
> 
> It describes how to sleep / wake-up the MCU.

Hello, Thanks for your reply, I think lv_task_handler do the similar things with lv_refr_task, but lv_refr_task is running periodically, should I remove the lv_refr_task and only use lv_task_handler when new event comming?

DevriesL, 2019-02-16:
Hi, I design a method to implement tick-less on our device:
1, let lv_task_handler return the period of active task except indev_proc_task/lv_refr_tasktask, if there is no custom task, return portMAX_DELAY.
2, In FreeRTOS, use xTaskNotify(x,x,x, period of active task) to wait for new event and periodically call lv_task_handler based on the  period of active task.

kisvegabor, 2019-02-18:
Good to hear that you made it work! 

Of course, some customization of the described approach might be required depending on your system.

closed 2019-02-18 by kisvegabor