lvgl/lvgl #842: How to use touchpad with LVGL?

GnUfTw, 2019-02-14:
Hello,
I am using LVGL to display a GUI for an embedded linux application. It is not clear to me what steps are necessary in order to use LVGL with a capacitive touch LCD screen. When I run my application, the GUI shows up fine on the display, but the touch input isn't working. 

I have read the porting documentation and I am adding the touchpad as an input device in main() as follows:
```c
lv_indev_drv_t indev_drv;
lv_indev_drv_init(&indev_drv);
indev_drv.type = LV_INDEV_TYPE_POINTER;
indev_drv.read = ts_input_read;
lv_indev_drv_register(&indev_drv)
```
And the ts_input_read function is defined as follows:
``` c
static lv_coord_t touchpad_x;
static lv_coord_t touchpad_y;

static bool ts_input_read(lv_indev_data_t* data)
{
    data->point.x = touchpad_x;
    data->point.y  touchpad_y;
    data->state = LV_INDEV_STATE_PR;
    return false;
}
```
It's not clear to me where I need to go from here.
- Do I need to create a custom driver for the touchpad I am using?
- Do I need to add more logic to **ts_input_read** to read from the touchpad?
- How do I know if **data->state** should get set to **LV_INDEV_STATE_PR** or **LV_INDEV_STATE_REL**?

Any suggestions would be much appreciated.

embeddedt, 2019-02-14:
@GnUfTw If you're on Linux then your touchscreen might be a standard `/dev/input/eventX` device. In that case there is [a driver](https://github.com/littlevgl/lv_drivers/blob/master/indev/evdev.c) in the [lv_drivers](https://github.com/littlevgl/lv_drivers) repository.

GnUfTw, 2019-02-14:
@embeddedt Do you know if there is a way I can determine if the touchscreen is a standard `/dev/input/eventX` device?

If I run the command `ls /dev/input` I get the following output: `by-path event0 event1 event2 event3 mice mouse0 mouse1 touchscreen0`.

GnUfTw, 2019-02-14:
I have made the following changes to my **lv_drv_conf.h** file:
- `#define USE_EVDEV 1`
- `#define EVDEV_NAME "/dev/input/event2`

Although, it is still not clear to me how to use the **evdev** touchpad driver from within my application. I am left with the following questions:

- Do I need to call `evdev_init()` function from main?
- How am I supposed to use the `evdev_set_file()` function?
- How am I supposed to use the `evdev_read()` function?

embeddedt, 2019-02-14:
@GnUfTw 

1. Call `evdev_init` before you register the touchscreen driver.
2. Don't worry about `evdev_set_file`.
3. Instead of registering `ts_input_read`, register `evdev_read`.

GnUfTw, 2019-02-14:
@embeddedt Thanks for the information. Unfortunately, the touchscreen is still not working. My main function now looks like this:
``` c
#include <stdlib.h>
#include <stdio.h>
#include <unistd.h>
#include <stdbool.h>
#include "lvgl/lvgl.h"
#include "lv_drivers/display/fbdev.h"
#include "lv_drivers/indev/evdev.h"
#include "app/app.h"

int main(void) {
	/*LittlevGL init*/
	lv_init();

	/*Linux frame buffer device init*/
	fbdev_init();

	/*Add a display the LittlevGL using the frame buffer driver*/
	lv_disp_drv_t disp_drv;
	lv_disp_drv_init(&disp_drv);
	disp_drv.disp_flush = fbdev_flush;      /*It flushes the internal graphical buffer to the frame buffer*/
	lv_disp_drv_register(&disp_drv);

	/* Linux input device init */
	evdev_init();

	/* Set up touchpad input device interface */
	lv_indev_drv_t indev_drv;
	lv_indev_drv_init(&indev_drv);
	indev_drv.type = LV_INDEV_TYPE_POINTER;
	indev_drv.read = evdev_read;
	lv_indev_drv_register(&indev_drv);

	App();

	/*Handle LittlevGL tasks (tickless mode)*/
	while(1) {
		lv_tick_inc(5);
		lv_task_handler();
		usleep(5000);
		App_Tick();
	}

	return 0;
}
```
A large portion of the application I am working on has been developed and works with the simulator. I am currently working on porting it to the target system and getting it to work with the actual hardware.

Do you have any suggestions regarding where I should go from here?

embeddedt, 2019-02-14:
You can't have `lv_tick_inc(5);` in the same while loop. It needs to go in a separate thread.

kisvegabor, 2019-02-15:
@GnUfTw I suggest adding `printf`s to the `evdev` driver to see what's going on.

fallstool, 2019-02-15:
Hi:
The following code you can refer to,  It works in my touch LCD screen.
`bool evdev_read(lv_indev_data_t * data)
{
    struct input_event in;

    while(read(evdev_fd, &in, sizeof(struct input_event)) > 0) {
        if(in.code == ABS_MT_POSITION_X) {
        	evdev_root_x = in.value;
        } else if(in.code == ABS_MT_POSITION_Y) {
        	evdev_root_y = in.value;
        } else if(in.code == ABS_MT_TOUCH_MAJOR) {
        	evdev_button = in.value == 0 ? LV_INDEV_STATE_REL : LV_INDEV_STATE_PR;
        }
    }
    
    data->point.x = evdev_root_x;
 	data->point.y = evdev_root_y;
    data->state = evdev_button;
    if(data->point.x < 0)
         data->point.x = 0;
       if(data->point.y < 0)
         data->point.y = 0;
       if(data->point.x >= LV_HOR_RES)
         data->point.x = LV_HOR_RES - 1;
       if(data->point.y >= LV_VER_RES)
         data->point.y = LV_VER_RES - 1;

       return false;
}`

kisvegabor, 2019-02-15:
@fallstool Please use 
` ```c `
` code `
` ``` `
around the code block to add syntax highlighting.

GnUfTw, 2019-02-15:
I have added logic to run `lv_tick_inc()` on a background thread, this is shown below.
``` c
void *tick_thread(void *data)
{
	(void) data;	// Suppress warnings.

	// Call lv_tick_inc() periodically to inform LVGL about how much time has elapsed.
	while (1)
	{
		usleep(5000);
		lv_tick_inc(5);
	}
}

int main(void)
{
	pthread_t lvTickThread;
	int rc;

	/*LittlevGL init*/
	lv_init();

	/*Linux frame buffer device init*/
	fbdev_init();

	/*Add a display the LittlevGL using the frame buffer driver*/
	lv_disp_drv_t disp_drv;
	lv_disp_drv_init(&disp_drv);
	disp_drv.disp_flush = fbdev_flush;      /*It flushes the internal graphical buffer to the frame buffer*/
	lv_disp_drv_register(&disp_drv);

	/* Linux input device init */
	evdev_init();

	/* Set up touchpad input device interface */
	lv_indev_drv_t indev_drv;
	lv_indev_drv_init(&indev_drv);
	indev_drv.type = LV_INDEV_TYPE_POINTER;
	indev_drv.read = evdev_read;
	lv_indev_drv_register(&indev_drv);

	App();

	rc = pthread_create(&lvTickThread, NULL, tick_thread, NULL);

	/*Handle LittlevGL tasks (tickless mode)*/
	while (1)
	{
		lv_task_handler();
		usleep(5000);
		App_Tick();
	}

	return 0;
}
```
This did not fix the issue, touches are still not working. I then  added logging to the `evdev_read()` function. The modified function is shown below.
``` c
bool evdev_read(lv_indev_data_t * data)
{
    struct input_event in;
    printf("[DEBUG] Entered evdev_read() function...\n");

    while(read(evdev_fd, &in, sizeof(struct input_event)) > 0)
    {
        if (in.type == EV_REL)
        {
            printf("[DEBUG] in.type == EV_REL\n");
            if(in.code == REL_X)
            {
            	printf("[DEBUG] in.code == REL_X\n");
		#if EVDEV_SWAP_AXES
			evdev_root_y += in.value;
		#else
			evdev_root_x += in.value;
		#endif
            }
            else if(in.code == REL_Y)
            {
            	printf("[DEBUG] in.code == REL_Y\n");
		#if EVDEV_SWAP_AXES
			evdev_root_x += in.value;
		#else
			evdev_root_y += in.value;
		#endif
            }
        }
        else if(in.type == EV_ABS)
        {
            printf("[DEBUG] in.type == EV_ABS\n");
            if(in.code == ABS_X)
            {
            	printf("[DEBUG] in.code == ABS_X\n");
		#if EVDEV_SWAP_AXES
			evdev_root_y = in.value;
		#else
			evdev_root_x = in.value;
		#endif
            }
            else if(in.code == ABS_Y)
            {
            	printf("[DEBUG] in.code == ABS_Y");
		#if EVDEV_SWAP_AXES
			evdev_root_x = in.value;
		#else
			evdev_root_y = in.value;
		#endif
            }
        }
        else if(in.type == EV_KEY)
        {
            printf("[DEBUG] in.type == EV_KEY\n");
            if(in.code == BTN_MOUSE || in.code == BTN_TOUCH)
            {
            	printf("[DEBUG] in.code == %s\n", in.code == BTN_MOUSE ? "BTN_MOUSE" : "BTN_TOUCH");
                if(in.value == 0)
                    evdev_button = LV_INDEV_STATE_REL;
                else if(in.value == 1)
                    evdev_button = LV_INDEV_STATE_PR;
            }
        }
        printf("[DEBUG] in.value == %d\n", in.value);
    }

    /*Store the collected data*/

#if EVDEV_SCALE
    data->point.x = map(evdev_root_x, 0, EVDEV_SCALE_HOR_RES, 0, LV_HOR_RES);
    data->point.y = map(evdev_root_y, 0, EVDEV_SCALE_VER_RES, 0, LV_VER_RES);
#else
#if EVDEV_CALIBRATE
    data->point.x = map(evdev_root_x, EVDEV_HOR_MIN, EVDEV_HOR_MAX, 0, LV_HOR_RES);
    data->point.y = map(evdev_root_y, EVDEV_VER_MIN, EVDEV_VER_MAX, 0, LV_VER_RES);
#else
    printf("[DEBUG] evdev_root_x: %d\n", evdev_root_x);
    printf("[DEBUG] evdev_root_y: %d\n", evdev_root_y);
    data->point.x = evdev_root_x;
    data->point.y = evdev_root_y;
#endif
#endif

    printf("[DEBUG] evdev_button: %d\n", evdev_button);
    data->state = evdev_button;

    if(data->point.x < 0)
      data->point.x = 0;
    if(data->point.y < 0)
      data->point.y = 0;
    if(data->point.x >= LV_HOR_RES)
      data->point.x = LV_HOR_RES - 1;
    if(data->point.y >= LV_VER_RES)
      data->point.y = LV_VER_RES - 1;

    printf("[DEBUG] Leaving evdev_read() function...\n");
    return false;
}
``` 
When I run my application with the debug print outs (shown above) included in the code, on the target I get the following debug information printed out of the console repeatedly and it never changes, even though I am pressing buttons in my app on the touchscreen throughout this process.
```
[DEBUG] Entered evdev_read() function...
[DEBUG] evdev_root_x: 0
[DEBUG] evdev_root_y: 0
[DEBUG] evdev_button: 0
[DEBUG] Leaving evdev_read() function...
```
So, it seems to me that I am not receiving the input touch events at all. Advice on where to go from here?

embeddedt, 2019-02-15:
@GnUfTw Can you add debug printfs to `evdev_init` to make sure it `open()`s `/dev/input/event2` without errors?

GnUfTw, 2019-02-15:
UPDATE: I have figured out what the issue was. I needed to set `EVDEV_NAME` to `"/dev/input/touchscreen0"` instead of `"/dev/input/event2"`.

To elaborate on why I thought `/dev/input/event2` was the right device. On the target machine I ran the command `evtest` which gave me the following output:
```
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0: 20cc000.snvs:snvs-powerkey
/dev/input/event1: EP0790M09
/dev/input/event2: iMX6UL Touchscreen Controller
/dev/input/event3: keys
```
The platform we are targeting with this application is an  iMX6UL so I just assumed that `event2` was the right input device. After I realized the while loop in `evdev_read` wasn't getting entered, I went back to my target machine and ran `cat /dev/input/event2` and began pressing the touchscreen to find that nothing was being streamed via that input device. I then ran `cat /dev/input/touchscreen0` and began pressing the touchscreen, tons of random characters were printed out to the console. At this point, I changed `EVDEV_NAME` and the application is now working with touch input.

Thanks for the suggestions & help! It's much appreciated.

closed 2019-02-15 by GnUfTw

embeddedt, 2019-02-15:
@GnUfTw I suspected the problem was going to be something along those lines. I'm glad it's working for you now!

Adri1G, 2019-06-26:
>     3. Instead of registering `ts_input_read`, register `evdev_read`.
Hello,

What is the difference between evdev_read and touchpad_read?
@embeddedt  is is not better to use touchpad_read for a touchpad?

evdev_read is locate in lv_drivers/indev/evdev.h and touchpad_read is locate in lvgl/lv_porting/lv_port_indev_templ.c

embeddedt, 2019-06-26:
@Adri1G `evdev_read` is a prebuilt, already working driver for Linux touchscreens.

`touchpad_read` is a template driver designed to assist you in writing your own.