lvgl/lvgl #843: BUG about 'lmeter' and 'gague' obj

as564335sa, 2019-02-15:
Hi,
 The `lmeter` and `gague` obj have a bug, the scale is not regular,like this:
[image]
[image]

kisvegabor, 2019-02-15:
Do you use the latest master branch?

Leo2442926161, 2019-02-15:
Hi kisvegabor

> Do you use the latest master branch?
I download it at the day before yesterday, I saw the file version is v5.3 
 
> Please to simple code the reproduce the issue.
please try these simple code:

void gauge_demo(void)
{
    /*Create a style*/
    static lv_style_t style;
    lv_style_copy(&style, &lv_style_pretty_color);
    style.body.main_color = LV_COLOR_HEX3(0x666);     /*Line color at the beginning*/
    style.body.grad_color =  LV_COLOR_HEX3(0x666);    /*Line color at the end*/
    style.body.padding.hor = 10;                      /*Scale line length*/
    style.body.padding.inner = 8;                    /*Scale label padding*/
    style.body.border.color = LV_COLOR_HEX3(0x333);   /*Needle middle circle color*/
    //style.body.empty = 1;
    style.line.width = 3;
    style.line.color = LV_COLOR_RED;                  /*Line color after the critical value*/
    style.text.color = LV_COLOR_HEX3(0x333);
    style.text.opa = LV_OPA_TRANSP;

    /*Create a gauge*/
    lv_obj_t * gauge1 = lv_gauge_create(lv_scr_act(), NULL);
    lv_gauge_set_scale(gauge1, 360, 12 * 10, 0);
    lv_gauge_set_style(gauge1, &style);
    lv_gauge_set_critical_value(gauge1, 360);
    lv_gauge_set_range(gauge1, 0, 360);
    lv_obj_align(gauge1, NULL, LV_ALIGN_IN_TOP_LEFT, 20, 20);
    lv_obj_set_size(gauge1, 400, 400);
}

kisvegabor, 2019-02-15:
To find the issue I replaced `lv_trigo_sin` with `sin` and used `float` numbers. The result is better but still not perfect:
[image]

I assume that the rest of inaccuracy occurs because the line can be drawn on on integer pixels.

embeddedt, 2019-02-15:
@as564335sa @kisvegabor I'm pretty sure there was a known issue relating to the scale of the pixels between the line meter, but I'm having trouble finding the discussion.

kisvegabor, 2019-02-18:
@embeddedt I also remember something similar.

To make the `lmeter` and `gauge` a little bit better we should improve we should add some extra precision to the `trigo` functions in `lv_math.c`. I think 1/8 degree would be enough. 
@Leo2442926161 would you be interested to add this update.

Once the trigo functions are updated I can update the `lmeter` and `gauge`.

Leo2442926161, 2019-02-18:
Hi kisvegabor,

 I’d like to try it tommrrow

closed 2019-03-05 by as564335sa