lvgl/lvgl #846: Reduce label's padding/border?

hallmatthewp, 2019-02-15:
I'm using LVGL for text labels. My desire is to create and render text, but with as little extra "padding" beyond the width and height of the text, as possible. 

E.g. I have a 240x240 display. I have a image written to the display (using my own image format + display driver, not w/ LVGL) that spans from (0,0) to (239, 199), leaving a 240x40 px empty space at the bottom of the display. I want to create a label and write text into that 240x40 px space ONLY, without clearing any of the original image.

Do I adjust the label's style parameters to accomplish this? Which ones? Or is it something else?

Thanks!

kisvegabor, 2019-02-18:
Hi,

By default, the label's size automatically adjusted to the size of the text so there is no padding.

What you can do to reduce the size of the label is to reduce the `line_space` and `letter_space` in its style or choose a smaller font.

kisvegabor, 2019-03-03:
In the past few days [LittlevGL's Forum](https://forum.littlevgl.com) has launched to discuss questions there. See this [Announcment](https://forum.littlevgl.com/viewtopic.php?f=3&t=5).

If the issue still persists please open a topic for it on the Forum.

closed 2019-03-03 by kisvegabor