lvgl/lvgl #847: LVGL runs Slow

gwolfsystems, 2019-02-15:
Hello! I'm working with this wonderfull library. I'm currently using a STM32F407 Discovery running at max speed. The ILCD has an ILI9341 with SPI interface. I've notice the huge difference between drawing with the ILI9341 and drawing with LVGL. The library does not use DMA, but I have been looking for other libraries and they all wait after the bytes to be send, because Library has to configue D/I and CS. Besides, This library runs very fast.  I attatch you a video showing an image that loads with the library and, after that, LVGL starts. I also attatch you my LCDconf.h. The whole project is made with Atollic Truestudio and using Free RTOS. Is there something that I can do in order to improve speed?
Thanks!

Video:
https://1drv.ms/v/s!Arwn1F2Qb7hskK98V6XbKY5VUriyNA
Whole Project:
https://1drv.ms/f/s!Arwn1F2Qb7hskJosf_ZOwTTdr1AqWA
ILI9341:
https://github.com/martnak/STM32-ILI9341

kisvegabor, 2019-02-18:
Hi,

I'm almost certain that it's the issue of the driver.I saw that you use 1/10 screen sized VDB. LittlevGL first renders the image to that buffer and than sends the result image to your display via `disp_flush`.

So If the driver would be fast and LittlevGL would be slow you seen 10 parts appearing after eachother with some delay. The dealy between the parts would mean that LittlevGL is rendering the next part. 

But in the video, I see that the lines are filled after each other and I can't see the distinct  10 parts.

I can't find your `disp_flush` function on OneDriver. Show it please.

And please test `disp_flush` directly. Just initialize a `lv_color_t colors[LV_HOR_RES * 10]` with `LV_COLOR_RED` and pass it to `disp_flush` to see how fast it is.

kisvegabor, 2019-03-03:
In the past few days [LittlevGL's Forum](https://forum.littlevgl.com) has launched to discuss questions there. See this [Announcment](https://forum.littlevgl.com/viewtopic.php?f=3&t=5).

If the issue still persists please open a topic for it on the Forum.

closed 2019-03-03 by kisvegabor