lvgl/lvgl #848: Raspberry Pi 3 with 7 Inch display Single Tap causes Double Click?

neilabdev, 2019-02-15:
Thanks for the great work!  But as described here in the link below, on the Raspberry pi 3 with 7 inch touch display,  I experience a double tap for a single touch, which causes issues in LittleVGL, which correctly expects a single tap. My presumption is that is the OS Driver, but for the possibility that their is an option I overlooked which buffers the second click, and or verification that it works properly for others on a specific version of the kernel/install, it would be a great help to know so that I can begin with a proper installation/option, etc. Any help would be greatly appreciated! It runs awesome on Mac,Linux & Emscripten, and apparently for others, on Raspberri Pi 3 with 7 inch display. 

https://raspberrypi.stackexchange.com/questions/66817/single-touch-on-touchscreen-results-in-double-click

embeddedt, 2019-02-15:
@neilabdev I'm not sure exactly how the Raspberry Pi touch screen driver works, so this is general information that would apply to any `/dev/input/eventX` device.

If it always sends two events when you only want one, maybe you could do a hack in the driver where you just ignore every other event.

Can you try the `evtest` command and see if two press events fire whenever you touch the screen?

neilabdev, 2019-02-15:
Thanks @embeddet ! My guess was that, because Desktop Environments usually require a double tap to open a folder, etc, it simulates a double tap for a single touch to make it feel more natural. However, for embedded interfaces, they only assume a single tap, thus the dilemma. 

As to buffering the click, yes, I patched my code in the action(s) instead of the driver and it worked well for buttons, but more complex controls requires a bit more effort to get it work properly, as child or parent may get the event which would otherwise be handled internally, instead of your application action, etc.   I like your driver idea better, and didn't consider interfacing at that level, which seems more appropriate. It will be the method taken if 7inch Raspberry pi sends double tap as a matter of course, making such a driver beneficial for others.

 Regarding *evtest*, thanks so much for pointing me in this direction, wasn't aware of this tool. Results are below, and it gives some direction:

```
root@raspberrypi:~# evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	FT5406 memory based driver
Select the device event number [0-0]: 0
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "FT5406 memory based driver"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    426
      Min        0
      Max      800
    Event code 1 (ABS_Y)
      Value    254
      Min        0
      Max      480
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        9
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max      800
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max      480
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
Event: time 1550262915.143806, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 57
Event: time 1550262915.143806, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 432
Event: time 1550262915.143806, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 370
Event: time 1550262915.143806, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1550262915.143806, type 3 (EV_ABS), code 0 (ABS_X), value 432
Event: time 1550262915.143806, type 3 (EV_ABS), code 1 (ABS_Y), value 370
Event: time 1550262915.143806, -------------- SYN_REPORT ------------
Event: time 1550262915.223797, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1550262915.223797, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1550262915.223797, -------------- SYN_REPORT ------------

[TOUCH REGULAR BUTTON HERE]

Event: time 1550262931.583807, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 58
Event: time 1550262931.583807, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 256
Event: time 1550262931.583807, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 411
Event: time 1550262931.583807, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1550262931.583807, type 3 (EV_ABS), code 0 (ABS_X), value 256
Event: time 1550262931.583807, type 3 (EV_ABS), code 1 (ABS_Y), value 411
Event: time 1550262931.583807, -------------- SYN_REPORT ------------
Event: time 1550262931.623804, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1550262931.623804, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1550262931.623804, -------------- SYN_REPORT ------------
[... 33 lines cut ...]
Event: time 1550262949.143802, type 3 (EV_ABS), code 1 (ABS_Y), value 255
Event: time 1550262949.143802, -------------- SYN_REPORT ------------
Event: time 1550262949.183829, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 453
Event: time 1550262949.183829, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 257
Event: time 1550262949.183829, type 3 (EV_ABS), code 0 (ABS_X), value 453
Event: time 1550262949.183829, type 3 (EV_ABS), code 1 (ABS_Y), value 257
Event: time 1550262949.183829, -------------- SYN_REPORT ------------
Event: time 1550262949.223829, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 461
Event: time 1550262949.223829, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 260
Event: time 1550262949.223829, type 3 (EV_ABS), code 0 (ABS_X), value 461
Event: time 1550262949.223829, type 3 (EV_ABS), code 1 (ABS_Y), value 260
Event: time 1550262949.223829, -------------- SYN_REPORT ------------
Event: time 1550262949.263800, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 472
Event: time 1550262949.263800, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 262
Event: time 1550262949.263800, type 3 (EV_ABS), code 0 (ABS_X), value 472
Event: time 1550262949.263800, type 3 (EV_ABS), code 1 (ABS_Y), value 262
Event: time 1550262949.263800, -------------- SYN_REPORT ------------
Event: time 1550262949.303827, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 482
Event: time 1550262949.303827, type 3 (EV_ABS), code 0 (ABS_X), value 482
Event: time 1550262949.303827, -------------- SYN_REPORT ------------
Event: time 1550262949.343824, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1550262949.343824, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1550262949.343824, -------------- SYN_REPORT ------------

```

neilabdev, 2019-02-15:
Single Touch on Button

```
Event: time 1550263900.393810, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 62
Event: time 1550263900.393810, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 433
Event: time 1550263900.393810, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 390
Event: time 1550263900.393810, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1550263900.393810, type 3 (EV_ABS), code 0 (ABS_X), value 433
Event: time 1550263900.393810, type 3 (EV_ABS), code 1 (ABS_Y), value 390
Event: time 1550263900.393810, -------------- SYN_REPORT ------------
Event: time 1550263900.473796, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1550263900.473796, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1550263900.473796, -------------- SYN_REPORT ------------

```

embeddedt, 2019-02-15:
@neilabdev 

> My guess was that, because Desktop Environments usually require a double tap to open a folder, etc, it simulates a double tap for a single touch to make it feel more natural. However, for embedded interfaces, they only assume a single tap, thus the dilemma.

That would make sense.

> Event: time 1550263900.393810, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 62
> Event: time 1550263900.393810, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 433
> Event: time 1550263900.393810, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 390
> Event: time 1550263900.393810, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1 **pressed?**
> Event: time 1550263900.393810, type 3 (EV_ABS), code 0 (ABS_X), value 433
> Event: time 1550263900.393810, type 3 (EV_ABS), code 1 (ABS_Y), value 390
> Event: time 1550263900.393810, -------------- SYN_REPORT ------------
> Event: time 1550263900.473796, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
> Event: time 1550263900.473796, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0 **released?**
> Event: time 1550263900.473796, -------------- SYN_REPORT ------------

Are you sure these aren't the press and release events?

neilabdev, 2019-02-16:
@embeddedt 

After some research, yes I think you are correct. Which from looking at it, there are no duplicate events :(  Wherefore, I will have to enable some logging to figure out where and why the action handler is called twice. Perhaps there is a misconfiguration on my part, that I hope once found, will at least help someone else in the future avoid this very issue. Will keep you posted. Thx.

neilabdev, 2019-02-16:
@embeddedt  You help save the day with the *evtest* tool. My hypothesis deriving from the link above, where someone unrelated to LittleVGL was also experiencing the sytmtom of double clicks, made me presume that double clicks was the underlying issue, as *action* handlers were being called twice. However, *evtest* proved this hypothesis incorrect, and a nap help me look at the code a second time and realize what it was, which I will document below.

ISSUE:

Handler registered twice [THE END]

ORIGIN:

After refactoring my project from the pc_simular code and wanting to run in multiple devices via #ifdef this, #ifdef that, I noticed I actually call code similar to the following multiple times, but only on the pi:

```c

void touch_init() {
#if USE_EVDEV
    evdev_init();
    lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read = evdev_read;
    lv_indev_drv_register(&indev_drv);  //THIS IS CALLED TWICE
#endif
}

```

Wherefore, while the *indev_drv* will go away when the function ends, its registration via *lv_indev_drv_register* lives on to follow me on github :)  

Closing this one. While unlikely for someone else will make that same mistake, if they do, double registration is the issue. Thanks again!

closed 2019-02-16 by neilabdev

embeddedt, 2019-02-16:
@neilabdev Interesting. It's good you have it working now!