lvgl/lvgl #849: What is the role/function of LV_GC_ROOT?

Shining10, 2019-02-16:
hi，What is the role/function of LV_GC_ROOT? What is the purpose of increasing this?

embeddedt, 2019-02-16:
@Shining10 It was introduced to handle the garbage collector in the MicroPython binding.

From [#776](https://github.com/littlevgl/lvgl/issues/776#issuecomment-459147032):

> A _gc root_ is a global variable that **nothing is referring to it**. That means that there is no dynamically allocated memory block or stack frame which contains a pointer to it.
> gc must be aware of all gc roots to work correctly. For example, if it doesn't know about `indev_list`, it will not identify anything pointing to the first element of the linked list and will deallocate it on the next gc collection cycle.
> 
> I tried to identify all gc roots in lvgl project (I hope I caught them all).
> This is required because lvgl on Micropython uses the Micropython dynamic memory allocation functions, which means that every block of memory allocated with `malloc` is subject to garbage collection.

Shining10, 2019-02-18:
@embeddedt  thanks

closed 2019-02-18 by Shining10