lvgl/lvgl #850: Extend clickability to base object

rrroonn, 2019-02-16:
I have seen previous comments that this is not useful, however, there is value in having an object respond to user input even if it doesn't have a visual feedback mechanism.  For example, I can long-press a label object that represents a capability and trigger an edit function.  Ok, you could manipulate a button to create the same effect, but at the cost of significantly more memory and complexity.

For my personal use, I required a grid of 24 objects to represent switches - each showing a state and each having a button.  That required a label on a button on a button with a label.  Each one of these took 568 bytes.  24 x 568 bytes = nearly 14k of heap.

I took my own advice and created a bastardised label (labelx) with this code in the signal function.
```

    if ((sign == LV_SIGNAL_PRESSED)||(sign == LV_SIGNAL_RELEASED)||(sign == LV_SIGNAL_LONG_PRESS)||
    		(sign == LV_SIGNAL_LONG_PRESS_REP) )
	{
         if(ext->action) {
              res = (ext->action)(label, sign);
         }
}
```
along with an `_add_action()` function this allowed me to have the labelx react to a long-press without requiring a button.  The callback gets to see all events and selectively handle them.

```
static lv_res_t base_action(lv_obj_t * base, uint16_t event)
{
    uint8_t id = lv_obj_get_free_num(base);

    if (event==LV_SIGNAL_LONG_PRESS)
    {
		editing = base;
		switchEdit(id, whenDone);
    }

    return LV_RES_OK; /*Return OK if the button is not deleted*/
}

```
It would be better to add this functionality at the base object level though.

It would not be that hard to implement and I think it would be worth the effort.

thanks,

Ron

embeddedt, 2019-02-16:
@rrroonn As part of `v6.0` the action handler logic is being moved to `lv_obj`. Other object types can declare their own actions, but we will have a set of standardized actions for things like input, focusing, etc.

kisvegabor, 2019-02-18:
As this will be solved in `v6.0` I close this issue.

FYI,  we discuss `v6.0` related things here: https://github.com/littlevgl/lvgl/issues/785

closed 2019-02-18 by kisvegabor