lvgl/lvgl #851: Suggestion for toggle button and buttons in general

rrroonn, 2019-02-16:
Hi,

You currently have a toggle button.  May I suggest that you extend the functionality so that the button shows a different label depending on the toggle state.  I would not suggest creating a separate label as it would cost significant memory.

I understand that the current approach to having a label on a button is to create a button, and then create a label on the button.  Each can be separately styled and laid out and this approach offers a lot of flexibility.  However, it does make using it costly in memory and complexity.

Personally, I would like to see a simple button that has a `char *label;` for the label.  It can be used or not used depending on if it is initialised.  

Having created this button variation, then maybe I can create a button like this:

`	lv_obj_t *btn = lv_btn_create_simple(parent, copy, "Press Me",LV_ALIGN_CENTER);
`

If I have a toggle button:

```
lv_obj_t *btn = lv_btn_create_simple(parent, copy, "Off|On",LV_ALIGN_CENTER);
lv_btn_set_toggle(btn, true);
```
Or even a non-toggle button that shows different label if being pressed:

```	
lv_obj_t *btn = lv_btn_create_simple(parent, copy, "Not pressed|Pressed",LV_ALIGN_CENTER);
```

The GUI would update the label depending on the user interaction.

for your consideration.

thanks,

Ron

embeddedt, 2019-02-16:
@rrroonn How about registering an action callback and calling `lv_label_set_text` inside it?

rrroonn, 2019-02-16:
> @rrroonn How about registering an action callback and calling `lv_label_set_text` inside it?

@embeddedt, yes that would work but adds complexity that could be hidden by the GUI.

lvgl is a great piece of work and very powerful but it does make getting a GUI application harder than it could be.  The app that I am working on has many screens with lots of objects and each part of each screen takes a fair bit of effort.  Personally, I would like a GUI to make this as simple as possible.

For example, by creating a' form' overlay, I have condensed a GUI screen design down from over a hundred lines down to:

```
 // ==========================================================
static bool form_accept(form_t *f, bool post)
{
	// return false to keep form open
	if (post && finished) finished(zone);
	return true;
}
// ==========================================================
void	zoneEdit(uint8_t z, void (*callback)(uint16_t))
{
char buff[60];

	zone = z;
	finished = callback;

	sprintf(buff,"Editing Zone %d",z);

	form = formCreate(/*heading*/buff,/*col0*/30,/*col1*/130,/*rowht*/60,/*accept_action*/form_accept);
	formAddColumn(form,/*col2*/400);
	// NAME
	formAtSayGet(form,0,0,"Name", _Zones[z].zname, MAX_ZONE_NAME);
	// ENABLED
	formAtSayGetYesNo(form,0,2,"Enabled", &_Zones[z].zenabled);
	// TYPE
	_type = formAtSayGetOption(form,1,0,"Type","Instant\nEntry\nSmoke\nUtility", &_Zones[z].ztype);
	// AREA
	_area = formAtSayGetOption(form,2,0,"Area","Bypass\nAway\nHome", &_Zones[z].zarea);

}
```
When using a "form" for user input, this hides a huge amount of complexity and makes it quick and easy to add a new form.

In my opinion, the more that the GUI can do to make building the application easier, the better the experience for the developer.  So, yes, the developer _could_ add an action to the button and then check the state of the button and change the label if needed - but I think that is complexity that _could_ be taken care of by the GUI.  

It is just my opinion and I understand that it might not be high on anyone's priority list. 

Thank you for your ongoing effort.

Ron

embeddedt, 2019-02-18:
@kisvegabor Can you comment on this?

kisvegabor, 2019-02-19:
@rrroonn 

**About the toggle button with label**
I was also considering having a button which has a "built-in" label. However, the current `lv_btn` is a very generic object. A button with a label should be a new object type, like `lv_btnl`.

**About the form**
I found that in GUI projects I never need the exact same thing twice.  Even the basic buttons' appearance and requirements vary among each GUI designs. 
That's why I usually I have a `gui` folder where I add `lvgl` and files like `gui_btn.c`, `gui_form.c`, `gui_slider.c` which are built from `lvgl` objects but has the interface I really need and hides the rest.   

So IMO, letting you build up the form you need (instead of having a "hardcoded" form obejct) gives the highest flexibility. However, at least an example to see how to put things together would be helpful for the users to get started.

closed 2019-03-19 by stale[bot]

stephenrjohnson, 2020-01-04:
@embeddedt the event only has access to the btn  and event how do you get to the label from the btn?

kisvegabor, 2020-01-04:
If the label is the only children of the button you can use `lv_obj_get_child(btn, NULL);`.

embeddedt, 2020-01-05:
Or, you can attach a pointer to the label to the button using [object user data](https://docs.littlevgl.com/en/html/object-types/obj.html#_CPPv420lv_obj_set_user_dataP8lv_obj_t18lv_obj_user_data_t).