lvgl/lvgl #852: lv_obj_set_size() not working on labels

harteware07, 2019-02-16:
I noticed, that the lv_obj_set_size() function does not alter the size of a label.
Here is a minimal example:
`		lv_obj_t *label_liberation = lv_label_create(bg_testing, NULL);
		lv_label_set_static_text(label_liberation, "Liberation 123 4");
		lv_obj_set_style(label_liberation, &style_txt_liberation[i]);
		lv_label_set_body_draw(label_liberation, true);
		lv_obj_set_size(label_liberation, 300, 63); // really big in x
		lv_obj_set_pos(label_liberation, 10, 10);`

The label gets the default size. Now i changed the function `lv_obj_set_size()` and added this:
`    ori.x2 = obj->coords.x1 + w - 1;
    ori.y2 = obj->coords.y1 + h - 1;`
With this modification the label size gets changed. So I think maybe a bug in the signaling function?

embeddedt, 2019-02-16:
From the documentation (https://docs.littlevgl.com/#Label):

> The size of the label object can be automatically expanded to the text size or the text can be manipulated according to several long mode policies:

> ```
>    LV_LABEL_LONG_EXPAND: Expand the object size to the text size
>    LV_LABEL_LONG_BREAK: Keep the object width, break (wrap) the too long lines and expand the object height
>    LV_LABEL_LONG_DOTS: Keep the object size, break the text and write dots in the last line
>    LV_LABEL_LONG_SCROLL: Expand the object size and scroll the text on the parent (move the label object)
>    LV_LABEL_LONG_ROLL: Keep the size and roll just the text (not the object)
> ```

I think using `lv_label_set_long_mode(iconLabel[i], LV_LABEL_LONG_DOTS);` should help. Basically your problem is that the label is automatically sizing itself to accommodate the text.

kisvegabor, 2019-03-03:
In the past few days [LittlevGL's Forum](https://forum.littlevgl.com) has launched to discuss questions there. See this [Announcment](https://forum.littlevgl.com/viewtopic.php?f=3&t=5).

If the issue still persists please open a topic for it on the Forum.

closed 2019-03-03 by kisvegabor