lvgl/lvgl #853: lv_obj_set_free_ptr() There seems to be something wrong with it.

3038922, 2019-02-16:
Ubuntu1804 gcc8.2.0
g++
```
static lv_res_t ssaveAction(lv_obj_t *btn)
{
    int *tempData = static_cast<int *>(lv_obj_get_free_ptr(btn));
    std::cout << "tempData:" << *tempData << std::endl;

    return LV_RES_OK;
}
```
```
    lv_obj_t *aaa = lv_btn_create(displayObj[OBJ_BTNM_SON], nullptr);
    int b = 9900;
    lv_obj_set_free_ptr(aaa, &b);
    lv_btn_set_action(aaa, LV_BTN_ACTION_CLICK, saveAction);
```
[image]

embeddedt, 2019-02-16:
@3038922 Is `b` a local or global variable? It might be going out of scope.

3038922, 2019-02-16:
> @3038922 Is `b` a local or global variable? It might be going out of scope.
b is local.it just a test .
I wanted to pass in a json* class.

embeddedt, 2019-02-16:
@3038922 Please try a global variable and see if the problem persists.

3038922, 2019-02-16:
[image]
[image]
I have so many buttons using `lv_obj_set_free_ptr'. They are all right.

I just wanted to create a save button. Then I found some strange problems.

3038922, 2019-02-16:
int vvvv = 100;
static lv_res_t bbbb(lv_obj_t *btn)
{
    int *aaa = static_cast<int *>(lv_obj_get_free_ptr(btn));
    std::cout << "aaa:" << *aaa << std::endl;
    return LV_RES_OK;
}
void UserDisplay::createPidTest(lv_obj_t *parent)
{
    createUserTask(TASK_OTHER, pidTask, 100, "PID测试");
    if (displayObj[OBJ_BTNM_SON] == nullptr)
        displayObj[OBJ_BTNM_SON] = lv_tabview_create(parent, nullptr);
    lv_obj_t *asdf = lv_btn_create(displayObj[OBJ_BTNM_SON], nullptr);
    lv_obj_set_free_ptr(asdf, &vvvv);
    lv_btn_set_action(asdf, LV_BTN_ACTION_CLICK, bbbb);
}
[image]

Why do I use `lv_create_btn'`Everything is OK. Is it so strange to use `lv_create_btn'?

embeddedt, 2019-02-16:
@3038922 Please describe exactly what the issue is. What behavior are you expecting, and what results are you getting instead?

3038922, 2019-02-16:
> @3038922 Please describe exactly what the issue is. What behavior are you expecting, and what results are you getting instead?
It's my own C language foundation is not solid. The problem of changing to global variables has been solved. Thank you.

closed 2019-02-16 by 3038922