lvgl/lvgl #855: Help with FT6206 or STMPE610 drivers

pekalicious, 2019-02-17:
Hi,

I have an ESP32 and would like to use the [Adafruit 2.8'' TFT Capacitive touchscreen](https://www.adafruit.com/product/2090) with LittlevGL. Unfortunately, that screen has a FT6206 Capacitive Chip. Can somebody help me with a driver for that chip?

I also own an [Adafruit 2.4'' TFT Resistive touchscreen](https://www.adafruit.com/product/3315). That one uses a STMPE610 Resistive Chip, which is also not available for LittlevGL. I am able to run the hello world example by commenting out the touch related code.

Any help would be really appreciated.

Thank you.

embeddedt, 2019-02-17:
See https://docs.littlevgl.com/#Porting (scroll down to the input-related section). You can adapt an existing driver to work with LittlevGL.

closed 2019-02-18 by pekalicious

paulpv, 2019-09-17:
@pekalicious Did you ever get this working?

embeddedt, 2019-09-17:
The link I posted is broken. Here's the updated version: https://docs.littlevgl.com/en/html/porting/index.html

paulpv, 2019-09-17:
Thanks! I got up and running fairly easily within a few hours of posting that.

momosh13, 2019-10-06:
Hi 
I have a same situation, 2.4" TFT with FT6206 Capacitive Touch, and i like to use it with lv_arduino, example " ESP32_TFT_eSPI_Slider ", please help me with this project, or maybe you have some working code so i can use it.
Thank you

kisvegabor, 2019-10-07:
Let's continue this on the [Forum](https://forumtest.littlevgl.com/) to make it visible for a larger community. You can simply log in with your GitHub account.