lvgl/lvgl #856: display part of img source

ali-rostami, 2019-02-17:
Hi 
I have an img object with 4 states for displaying the strength of a signal.
when there is no signal I should display a question mark. **How can I display a character with existing fonts, as the source of an img object?**
In the next 3 states, I should display a part of an image:
state 2: lower one-third of the image.
state 3: lower two-thirds of the image.
state 4: the whole of the image.
by changing the size of the img object it is possible to display for example upper one-third of the image. but now how can I do this?

seyyah, 2019-02-17:
This issue maybe related with "Using images as resource file #706"

ali-rostami, 2019-02-18:
thanks, @seyyah. 
so It seems there isn't an offset parameter for img objects. The idea @kisvegabor offered there seems good.

kisvegabor, 2019-02-19:
@ali-rostami Adding the `offset` parameter should be simple. If you are interested in adding this feature, please reopen this issue and discuss the details.

ali-rostami, 2019-02-19:
OK, let's do it :)
adding this to the img object or introduce a new object? I think It's better to not introducing a new one.
anyway, we should add size property too, should we?

kisvegabor, 2019-02-19:
> I think It's better to not introducing a new one

I agree, only the `offset` parameters should be added to `lv_img_ext`.

> anyway, we should add size property too, shouldn't we?

You can already set the size with `lv_obj_set_size()` to crop the image.

ali-rostami, 2019-02-23:
sorry for late comment.
I got to this line:
```map_fp(coords, mask, img_data, opa, chroma_keyed, alpha_byte, style->image.color, style->image.intense);```
 in lv_img_draw_core function. I think this is the good place we can use the offset parameter. mabey by chnaging like this:
```
/*The pixel size in byte is different if an alpha byte is added too*/
	uint8_t px_size_byte = alpha_byte ? LV_IMG_PX_SIZE_ALPHA_BYTE : sizeof(lv_color_t);
        map_fp(coords, mask, img_data + (px_size_byte * img_px_ofs), opa, chroma_keyed, alpha_byte, style->image.color, style->image.intense);
```
what do you think?

kisvegabor, 2019-02-24:
No problem!

I think it can be added to `lv_img_design` too. It'd be more convenient because the offset value of the image object is available there but in `map_fp` (already no objects at this level only drawing.

ali-rostami, 2019-02-28:
yeah, we don't have any access to ext->offset of the object, but ```map_fp``` is after the decoding -```lv_img_decoder_open```- and I have no idea how we can use the offset before it. AFAIK the offset is the skipped pixels of the final image -raw image-.

kisvegabor, 2019-02-28:
I mean the `offset` can be added here: https://github.com/littlevgl/lvgl/blob/master/lv_objx/lv_img.c#L326
When the object's coordinates loaded, they can be shifted by an offset.

puzrin, 2019-02-28:
> I agree, only the `offset` parameters should be added to `lv_img_ext`.

While i find sprites a bit strange idea in "modern world", `offset` param in api looks "unnatural".

Initially, sprites were invented due different kinds of limitations, not actual now. For example, to reduce http connections before http2 invented.

kisvegabor, 2019-02-28:
I also think it's a reasonable feature because it's an easy way to:
- create simple gif-like animations
- organize related images into one image (fewer resources -> easier update)

In addition, it was already asked several times so it seems people need it. :)

puzrin, 2019-02-28:
I do not object against feature in general, but implementing it via `offset` looks suspicious. For compare, signatures of canvas's `drawImage()`:

https://developer.mozilla.org/ru/docs/Web/API/CanvasRenderingContext2D/drawImage

That's in JS, but you can find apropriate C methods in browser sources.

kisvegabor, 2019-03-01:
@puzrin 
Canvas is a different thing. `lv_img` is the most basic object to display an image.
Using this offset feature a higher level object type can be created. E.g. `lv_slideshow` where you can set the image coordinates and the library will show the images with `lv_slideshow_set_image(obj, 4)`

I

ali-rostami, 2019-03-02:
Hmm, this is the where I added the offset in ```lv_img_design``` function:
```
if(ext->src_type == LV_IMG_SRC_FILE || ext->src_type == LV_IMG_SRC_VARIABLE) {        	
            coords.x1 -= ext->offset.x;
            coords.y1 -= ext->offset.y;

            LV_LOG_TRACE("lv_img_design: start to draw image");
            lv_area_t cords_tmp;
            cords_tmp.y1 = coords.y1;
            cords_tmp.y2 = coords.y1 + ext->h - 1;

            for(; cords_tmp.y1 < coords.y2; cords_tmp.y1 += ext->h, cords_tmp.y2 += ext->h) {
                cords_tmp.x1 = coords.x1;
                cords_tmp.x2 = coords.x1 + ext->w - 1;
                for(; cords_tmp.x1 < coords.x2; cords_tmp.x1 += ext->w, cords_tmp.x2 += ext->w) {
                    lv_draw_img(&cords_tmp, mask, ext->src, style, opa_scale);
                }
            }
        }
```
seems it works for me. what do you think?
ps: unfortunately I couldn't get to this change by understanding the code and this is done by somehow trial and error. for example, at first I expected this would work:
```
            coords.x1 += ext->offset.x;
            coords.y1 += ext->offset.y;
```
so would you please explain more how this function works? thanks.

ali-rostami, 2019-03-04:
no advice?! shall I send the PR?

kisvegabor, 2019-03-04:
> no advice?! shall I send the PR?
Sorry, somehow I didn't notice your comment.  

So the difference is only the sign, right? It's correct to use `-` because `x = 50` offset means the image should start on the 50th pixel. So you need to move left.

> shall I send the PR?

Yes, please send a PR.

kisvegabor, 2019-03-08:
The PR is merged so I close this issue.

Thank you very much for this improvement!

closed 2019-03-08 by kisvegabor

ali-rostami, 2019-03-09:
Thank you, dear Gabor, for providing this opportunity :)