lvgl/lvgl #857: Text Area Keyboard input is slow.

chaos4284, 2019-02-18:
I current developed little Designer(https://github.com/chaos4284/lvgl_designer)
I want to ask a few questions.

kisvegabor, 2019-02-18:
I might look slow because the letters are added when you release the key. It's not the "natural" way. You can tweak the `keyboard` driver to emulate a press and release on key press.

```c
bool keyboard_read(lv_indev_data_t * data)
{
    data->state = state;
    data->key = keycode_to_ascii(last_key);

    /* Tell there is more data when pressed. The next data will the releases state. */
    state = LV_INDEV_STATE_REL;
    return data->state == LV_INDEV_STATE_PR ? true : false;
}
```

As it really looks more natural I'll add this working mechanism. 

Is it work better there?

chaos4284, 2019-02-18:
I will test it after work.

chaos4284, 2019-02-18:
Code is well operate. Thanks.

embeddedt, 2019-02-18:
@chaos4284 I would recommend using submodules for `lvgl`, `lv_drivers`, and `lv_examples`. It would make the library easier to update.

chaos4284, 2019-02-19:
I reopend current issue.
When typing keyboad input,
When typing keyboard input quickly, it is not as smooth as typing in the search box on the Internet browser, and some characters are missing.

kisvegabor, 2019-02-19:
Try to lower `LV_INDEV_READ_PERIOD` in `lv_conf.h`.

chaos4284, 2019-02-25:
No Effect.  The problem is still the same.

kisvegabor, 2019-02-25:
Please send a video about how it looks like now.

chaos4284, 2019-02-26:
Here is the source code at the https://github.com/chaos4284/lvgl_designer.

It exist total three screen.

From the left, It is  component screen, draw screen, property screen.

If you click the button component on the component screen and then click on the Draw screen, the textarea appears on the Property screen.

After selecting textarea with MOUSE, enter  the value with the keyboard.

kisvegabor, 2019-02-28:
I tested you project and I have some tips:
- Call `lv_tick_inc()` in a task. Similarly to the original PC simulator example
- Set `SDL_REFR_PERIOD` to 10 in `monitor.c`
- Set `LV_INDEV_READ_PERIOD` to `10` in `lv_conf.h`

Still not perfect but much better for me

chaos4284, 2019-02-28:
When I modified A to a value of 10, then I confirmed Key Input  are smooth. 
Thanks.

kisvegabor, 2019-02-28:
Great! As LittlevGL polls the key presses  (and not event-driven) I think that all we can do.

closed 2019-02-28 by kisvegabor