lvgl/lvgl #858: Have LV_INDEV_TYPE_KEYPAD automatically swap between PREV/NEXT and LEFT/RIGHT based on context

rajkosto, 2019-02-18:
I have screen with no touchpad and 4 buttons, i have assigned them functions BACK(ESC), OK(ENTER), LEFT, RIGHT
Now, with ENCODER, it automatically swaps between LEFT/RIGHT and PREV/NEXT functions depending on context of group/object it's on (if its a slider, it moves it LEFT/RIGHT, if youre selecting buttons, it goes PREV/NEXT), request this feature be added to KEYPAD input type (as an option) because making my KEYPAD an ENCODER would be a hack (would have to track number of presses and would lose hold capability) and i would not be able to use the BACK button in that case.

Also, when using tab view with button navigation, is it possible to avoid the entire page being highlighted all the time ?

embeddedt, 2019-02-19:
@kisvegabor I'm not familiar with the group-related things. Can you comment on this?

kisvegabor, 2019-02-19:
It seems possible to add "encoder-like" behavior to the `KEYPAD` but it's not so simple. Hopefully, you can achieve your goal with an `ENCODER` device too.

> would have to track number of presses 

How do you mean it?

> and would lose hold capability

if "hold" means long-press action of `lv_btn` then it is supported by the `ENCODER` too

How do you handle the ESC button right now?

rajkosto, 2019-02-20:
No, by "hold" i mean i hold the right button and it keeps increasing the slider value when its focused and in edit mode on it.

I'm working on some stuff but it involves changing the indrv KEYPAD handling code which im not pleased about

kisvegabor, 2019-02-21:
> No, by "hold" i mean i hold the right button and it keeps increasing the slider value when its focused and in edit mode on it.

You can still do it manually by emulating encoder turn if the button is long pressed. It's a little bit hacky though. 

> I'm working on some stuff but it involves changing the indrv KEYPAD handling code which im not pleased about

I've cheked the `indev_keypad_proc` to see how diffucult would be to implement it but didn't see a clear and generic way.

I suggest trying customize `indev_keypad_proc` or `indev_encoder_proc` according to your specific needs.

closed 2019-03-21 by stale[bot]