lvgl/lvgl #859: Maintainers' topic

kisvegabor, 2019-02-18:
**Let's discuss here the maintance related things here.**

One thing I'd like to discuss is that currently, I get notifications about all the new issues. As the library getting more popular it means a lot of notifications. I'd like to
- answer the questions
- make decisions about new features when required
- implement new features

However, I can't handle all three things together.  So, I think it would be the best if I changed my notifications policy to be notified if I'm participating in a conversation or explicitly tagged (as @kisvegabor). It'd allow dealing with things where my assistance is really required and I could focus on higher level things.

Let me know what do think or if you know a better solution.  
@embeddedt I'm very curious about your opinion. Do you think it'd work?

chaos4284, 2019-02-18:
I want to make board related three topic.

amirgon, 2019-02-18:
I think that GitHub issues are not the right place for questions, especially not user questions about how to use the library, or even questions relate to difficulties using the library.   

Instead, I suggest using a forum.  

This is very common practice. For example, on Micropython, questions are answered on https://forum.micropython.org/ . If someone opens an issue on https://github.com/micropython/micropython for asking a question, he is politely asked to move the question to the forum, here is [one example](https://github.com/micropython/micropython/issues/4497) for that.

kisvegabor, 2019-02-18:
@amirgon Having a forum to separate the development from user questions sounds very well to me and it would solve this issue.   
It seems easy to to set-up a forum with [PhpBB](https://www.phpbb.com/).

Let's wait a little to hear other opinions too.

embeddedt, 2019-02-18:
@kisvegabor I agree that it would be impossible to handle all three of those tasks, especially as the library's popularity increases further.

I like @amirgon's suggestion, but I'm not so keen on the idea of using a forum. Maybe, instead of a forum, we could create a separate "meta" repository that contains no code, but allows us to keep the questions in a different spot.

The advantage of this setup is that if someone then comes to the main `lvgl` repository and asks a question, GitHub allows it to be transferred to the "meta" repository without having to close it and tell the user to make a forum account and ask their question again.

@amirgon What do you think about that?

amirgon, 2019-02-18:
> Maybe, instead of a forum, we could create a separate "meta" repository that contains no code, but allows us to keep the questions in a different spot

@embeddedt This is an unusual idea. There are many projects that have their own external forums, but I haven't seen any example of using a "meta" GitHub repo as a forum.  
You know something, even GitHub themselves manage [their own support forum](https://github.community/) as an external forum and not as a repo...  

Do you have an an example of a project that is managing its forum as a GitHub repo?  

Personally I feel that GitHub "Issues" are not a natural "forum platform". It misses some features, such as creating groups and conversations, advanced search etc. It also doesn't look like a question/answer forum from user perspective.

embeddedt, 2019-02-18:
@amirgon 

> Do you have an an example of a project that is managing its forum as a GitHub repo?

No.

> Personally I feel that GitHub "Issues" are not a natural "forum platform". It misses some features, such as creating groups and conversations, permissions, management, advanced search etc.

Can I ask why you think those would be valuable features? What do you think should be improved in the current setup?

fpw, 2019-02-18:
I also think an external forum would be the best approach. There would be support for:

- having sub-forums, e.g. for sharing ports to common platforms, showing implemented projects, ...
- making posts sticky so they are always on top, e.g. an FAQ post. In GitHub issues, the latest is always on top if I'm not mistaken, so users would have to actively search the "FAQ issue"
- having moderators that move topics

embeddedt, 2019-02-18:
@fpw

> There would be support for:
> 
>     * having sub-forums, e.g. for sharing ports to common platforms, showing implemented projects, ...

That is an advantage that forums have.
 
>     * making posts sticky so they are always on top, e.g. an FAQ post. In GitHub issues, the latest is always on top if I'm not mistaken, so users would have to actively search the "FAQ issue"

You can pin issues on GitHub now. See https://help.github.com/articles/pinning-an-issue-to-your-repository/.

>     * having moderators that move topics

I assume this would tie in with the sub-forum idea above.

amirgon, 2019-02-18:
> Can I ask why you think those would be valuable features? What do you think should be improved in the current setup?

@embeddedt Yes.

Here is some explanation [why you must not ask questions on Github issues](https://medium.com/@methane/why-you-must-not-ask-questions-on-github-issues-51d741d83fde).

Someone already suggested adding a [Discussion Tab in GitHub](https://github.com/dear-github/dear-github/issues/44) instead of using the issue tracker.  
Someone else has even built a [Discussions forum, explicitly for for GitHub](https://gitq.com/).  

It even looks like GitHub added some feature called "[Team Discussions](https://github.blog/2017-11-20-introducing-team-discussions/)" but I can't find it anywhere, maybe it's not available publicly.

Let's have a look at @kisvegabor suggestion, [phpbb](https://www.phpbb.com/community/).  
It's not necessarily the best option (and neither the most modern one) but it's start.

- **Landing Page**: There is value to the design and the perception of the forum "landing" page. Look at the [front page](https://www.phpbb.com/community/) - forums are grouped together by subject ("General", "Support Forums", "Extensions Forums" etc.). It's a convenient "landing page", easy to navigate further where we want. You can also add some useful links for documentation, nice images etc.  
A user doesn't have to look for an "issues" tab of some repo who's purpose is to be a forum for another repo.
- Forum is more **personalize**. At the side of every post you can see details about the user, such as name, country, photo and even **a way to contact the user directly**. Did you ever tried to contact a GitHub user directly? If the user doesn't add his email to his profile, you might find this is a non-trivial task. Most forums allow direct communication between members even when email is not public.
- **Search** by keywords, in specific sub-forum, only in topic, in topic and text etc.  
You can filter posts by "unanswered" posts, "Unread" posts, "New posts" etc.
- **Maintenance** : Forums maintainers can move posts between forums, delete user messages and even block or ban users. As a user you can see the list of all your posts on all forums, all the responses you received etc.

kisvegabor, 2019-02-19:
Thank you guys for highlighting these important things! 

@amirgon's link about [why you must not ask questions on Github issues](https://medium.com/@methane/why-you-must-not-ask-questions-on-github-issues-51d741d83fde) summarizes well why a forum is really important.

As an experiment, I created a 
PhpBB forum: https://littlevgl.com/forum/index.php
GitQ forum: https://gitq.com/littlevgl/lvgl

According to my current experience:

**PhpBB**
Cons:
- Really looks oldfashioned

Pros:
- Hosted on littelvgl.com
- Feature reach

**GitQ**
Cons:
- Can't attach files
- Not hosted on littelvgl.com

Pros:
- Looks modern and user-friendly
- Fewer features (compared to PhpBB)
- Integrated with GitHub (has the similar advantages as @embeddedt's suggestion about the meta repo)

It seems one of them should be fine. However we should make a long-term decision. Changing the forum one year later would be very bad. I light of that, I think the PhpBB scales better with the growing community which is very important.

Please give feedback. You can register on the PhpBB forum to see if it's working.

---

I've jus changed my notification policy to see how it's working. So please tag me if you want my review.

seyyah, 2019-02-19:
We prefer the discourse [1].

> A platform for community discussion. Free, open, simple. https://www.discourse.org

## :clipboard: Reference
1. https://github.com/discourse/discourse

amirgon, 2019-02-19:
@kisvegabor @seyyah 

Other notable open source forums software:

- [NodeBB](https://nodebb.org/)
- [Flarum](https://flarum.org/)

kisvegabor, 2019-02-19:
These look great but `discourse` and `NodeBB` are not free and `Flarum` is still beta.

amirgon, 2019-02-19:
> These look great but `discourse` and `NodeBB` are not free 

@kisvegabor Incorrect.  
[NodeBB is open source](https://github.com/NodeBB/NodeBB) under GPL-3.0. [Discourse](https://github.com/discourse/discourse) is also open source. Both are available on GitHub.

kisvegabor, 2019-02-19:
@amirgon 
Upps, really. Seeing the "Pricing" menu confused me. After a quick overview, it wasn't obvious to me how can I install them on a server where I have only FTP access. Still need to explore them deeper.

embeddedt, 2019-02-19:
@kisvegabor I liked how easily I was able to log in and move my question to GitQ. However, it doesn't look like it implements sub-forums the way phpBB and other forums do.

kisvegabor, 2019-02-19:
It really integrates well! 
@embeddedt I added you as administrator to see how permissions work.

amirgon, 2019-02-19:
@kisvegabor @embeddedt  
[GitQ](https://github.com/GitQLab/gitq.com) is not a live project. 4 stars in GitHub, last commit 2 years ago...
Looks more like a proof-of-concept thing than a real project.

embeddedt, 2019-02-19:
@amirgon I think that's because the repository only has a barebones `README.md` file in it. I would assume that the actual code is somewhere else.

kisvegabor, 2019-02-19:
I also didn't find the "real" repo nor any references which states that it's a long term supported project.

amirgon, 2019-02-19:
@embeddedt you are right.
One more thing, I think it could be risky to use a forum that is hosted by someone else (which is not Google or Amazon). Who guarantees it will be there tomorrow?  
I think a better option would be self hosting a forum.
In case of GitQ, it seems you can't access the actual code of GitQ, so you can't host it yourself.

kisvegabor, 2019-02-19:
@amirgon I'm also afraid of it. It should have a more solid background to trust it in the long term.

amirgon, 2019-02-19:
Added a [question on GitQ](https://gitq.com/GitQLab/gitq.com/topics/11/where-is-gitq-source-code)

kisvegabor, 2019-02-19:
Similar topic: do you have experience with Gitter or Slack? Can we use them well in our case?

embeddedt, 2019-02-19:
@kisvegabor Gitter is more of a live chat system, not a community forum. I'm not that familiar with Slack.

kisvegabor, 2019-02-19:
I ment besides the forum, not instead of it.

amirgon, 2019-02-19:
@kisvegabor I'm not sure having both forum and chat is a good idea.  
Suppose I'm searching for some topic, do I need to search both forum and chat history? You split the knowledge between two systems and there is no clear line what topic should go where.   
Whoever answers the questions would also need to split his attention between the two systems.

kisvegabor, 2019-02-20:
@amirgon I had the same thoughts.  Thank you!

So let's wait a few days for GitQ. If nothing will happen I think we should choose a self-hosted forum. Do you agree?

amirgon, 2019-02-20:
> So let's wait a few days for GitQ. If nothing will happen I think we should choose a self-hosted forum. Do you agree?

@kisvegabor Would you like to evaluate [NodeBB](https://github.com/NodeBB/NodeBB)  and [Discourse](https://github.com/discourse/discourse) in the meanwhile?

kisvegabor, 2019-02-20:
Yes, I will check if it is possible to install them via FTP without ssh.

amirgon, 2019-02-20:
> Yes, I will check if it is possible to install them via FTP without ssh.

@kisvegabor Why can't you use ssh?

kisvegabor, 2019-02-20:
littlevgl.com is hosted on a simple web server (not a Virtual Server) where I can only upload the files via FTP.

seyyah, 2019-02-20:
`discourse` requires a Virtual Server :(

amirgon, 2019-02-20:
> So let's wait a few days for GitQ

So GitQ maintainer [recommends we use Discource](https://gitq.com/GitQLab/gitq.com/topics/11/where-is-gitq-source-code/2) :man_facepalming: 

> littlevgl.com is hosted on a simple web server (not a Virtual Server) where I can only upload the files via FTP.

@kisvegabor Did you consider migrating to a VM you have full control of? As LittlevGL grows and becomes more mature I think it would make sense to have good hosting infrastructure.

kisvegabor, 2019-02-20:
> So GitQ maintainer [recommends we use Discource](https://gitq.com/GitQLab/gitq.com/topics/11/where-is-gitq-source-code/2)

In light of that, we should leave GitQ.

> Did you consider migrating to a VM you have full control of? As LittlevGL grows and becomes more mature I think it would make sense to have good hosting infrastructure.

I was considering but I'm trying to keep things as simple as possible. I just don't want to deal with server related things too. :)  Fortunately, if something turns out later which really requires a Virtual Server then the whole page can be moved without any problem.

For the [PhpBB forum](https://littlevgl.com/forum/index.php) I set the same them which is used on [micropython's forum](https://forum.micropython.org/) and it look well enough (at least for me).

The README of **NodeBB** starts like this:
> NodeBB Forum Software is powered by Node.js and built on either a Redis or MongoDB database. 

So it seems I can't use it with the MySQL DB of my server.

So finally the PhpBB forum seems to be the best choice.

amirgon, 2019-02-20:
> For the [PhpBB forum](https://littlevgl.com/forum/index.php) I set the same them which is used on [micropython's forum](https://forum.micropython.org/) and it look well enough (at least for me).

I registered to [LittlevGL forum](https://littlevgl.com/forum/index.php). There are now 2 members! :wink:

kisvegabor, 2019-02-20:
> I registered to [LittlevGL forum](https://littlevgl.com/forum/index.php). There are now 2 members!

Cool! :sunglasses: 

> Are you sure PhpBB has all the features you would need?

 * MarkDown syntax: Havn't find a good extension
 * Monospace code: Works, but no syntax highlight
 * Syntax highlighting: No
 * Preview of the post: I see no problem with that
 * Mention other forum users in post: I made it work with an extension. I mentioned you in a test post. You need to type 3 letters after `@` to show the drop down list.

kisvegabor, 2019-02-20:
I moved the forum to the `forum` subdomain: https://forum.littlevgl.com

The subforums works well: https://forum.littlevgl.com/viewforum.php?f=7

amirgon, 2019-02-20:
> * MarkDown syntax: Havn't find a good extension
> * Monospace code: Works, but no syntax highlight
> * Syntax highlighting: No

@kisvegabor 
Markdown and syntax highlight are important, in my opinion. See how useful they are on GitHub issues.  
I understand that you don't want to deal with hosting infrastructure right now, just take into account that when you start a forum you are **locked in**. It will be much harder to migrate to a more modern forum in the future. (although not impossible. With some effort it's possible to [migrate from PhpBB to Discourse](https://meta.discourse.org/t/importing-from-phpbb3/30810), but it's definitely much harder than starting fresh on Discourse)
 
> * I mentioned you in a test post

I sent you a reply, but it seems that you need to approve every forum post before it's published.
Do you really want to do that?

kisvegabor, 2019-02-20:
> Markdown and syntax highlight are important, in my opinion. See how useful they are on GitHub issues.
I understand that you don't want to deal with hosting infrastructure right now, just take into account that when you start a forum you are **locked in**. It will be much harder to migrate to a more modern forum in the future. (although not impossible. With some effort it's possible to [migrate from PhpBB to Discourse](https://meta.discourse.org/t/importing-from-phpbb3/30810), but it's definitely much harder than starting fresh on Discourse)

I agree, something more "GitHub-ish" would be better.  Does discourse have the mentioned features? (Markdown, C syntax highlight etc?) @seyyah if you are using it, can you tell some words about it?

I've checked [this Discourse forum](https://forum.talksurf.com/) as an example. It seems not so well organized like a PhpBB forum.  Although it looks better :)

> I sent you a reply, but it seems that you need to approve every forum post before it's published.
Do you really want to do that?

I updated the settings. Should work now.

amirgon, 2019-02-21:
> I agree, something more "GitHub-ish" would be better. Does discourse have the mentioned features? (Markdown, C syntax highlight etc?)

- Discourse supports [mentions](https://meta.discourse.org/t/what-are-mentions/32788).
- Discourse supports [Markdown](https://meta.discourse.org/t/post-format-reference-documentation/19197/2) as well as BBCode and HTML
- Discourse supports [GitHub style fenced code](https://help.github.com/en/articles/creating-and-highlighting-code-blocks), but it uses [highlight.js](http://softwaremaniacs.org/soft/highlight/en/) for syntax highlighting, which is different and less feature-rich than GitHub so color highlighting looks a bit different than GitHub.

@kisvegabor Do you know Slant? It's a community driven comparison website, where you can compare various software tools.  
I'm not saying it's objective. It may be skewed by some specific users, but still, have a look at a [comparison between PhpB and Discourse](https://www.slant.co/versus/2789/17561/~discourse_vs_phpbb).
Quoting:

> When comparing Discourse vs phpBB, the Slant community recommends Discourse for most people. In the question“What are the best web forum software packages?” Discourse is ranked **1st** while phpBB is ranked **10th**.

It's worth noting that Discourse had a total of ~100 reviewer while phpBB had only 12. At least it looks like Discourse is a lot more popular these days.

seyyah, 2019-02-21:
> I agree, something more "GitHub-ish" would be better. Does discourse have the mentioned features? (Markdown, C syntax highlight etc?) @seyyah if you are using it, can you tell some words about it?

I didn't use the discourse but as far as I know, Discourse is very popular.

We use the DigitalOcean droplets as a VM. DigitalOcean supports one-click apps. Discourse is in the one-click app list [1].

> One-click install and deploy Discourse to an SSD cloud server in 55 seconds [1].

One-click app installation is very simple process.

Discourse require Rails etc. skills/experience. I've experience with them, but not Discourse.

[image]

The biggest problem is extra server cost and server/system maintenance.

> Discourse requires at least 2GB of RAM [2]

We choose a plan with enough memory during creation. So minimum droplet cost is $10/mo.

[image]

I support/help server/system maintenance.

DigitalOcean support also open source project (Open Source Sponsorships) [4]. I contact with DigitalOcean Open Source Sponsorships for LittlevGL + Discourse (Ticket #3526851).

I agree with @amirgon feature lists.

## :clipboard: Reference
1. https://www.digitalocean.com/products/one-click-apps/discourse/
2. https://www.digitalocean.com/docs/one-clicks/discourse/
3. https://www.digitalocean.com/community/tutorials/how-to-install-discourse-on-ubuntu-16-04
4. https://www.digitalocean.com/open-source/

amirgon, 2019-02-21:
> * Monospace code: Works, but no syntax highlight

@kisvegabor a quick google search for "phpbb highlight.js":

- https://github.com/s9e/phpbb-ext-highlighter
- https://github.com/OxCom/phpbbch

kisvegabor, 2019-02-21:
@seyyah 
> I didn't use the discourse but as far as I know, Discourse is very popular.

From [this comment](https://github.com/littlevgl/lvgl/issues/859#issuecomment-465009047), I thought you are using it.

> We choose a plan with enough memory during creation. So minimum droplet cost is $10/mo.

I'd consider the $5 droplet the but the > $10/mo is not what I want to pay for running a free project :)

> DigitalOcean support also open source project (Open Source Sponsorships) [4]. I contact with DigitalOcean Open Source Sponsorships for LittlevGL + Discourse (Ticket #3526851).

I got the mail about the ticket.Thank you very much for writing! I'm curiously waiting what they will say.

@amirgon 
> a quick google search for "phpbb highlight.js":

I'll try them, thank you!

puzrin, 2019-02-21:
IMHO for new start discourse has no comparable alternatives (if you are not going to develop your own forum software as i do). It's a project of stackoverflow's owner.

muhammetasan, 2019-02-21:
How about Zulip? Users can join with GitHub accounts . As far as I know it has integration to GitHub issue tracker or topics. It is free and open source.

puzrin, 2019-02-22:
I have one more question. Why not create just a separate repo, only for "low-tech questions"?

The root of problem is well known: number of issues, not related to project development, become too high. But why should those be in other form (chat, forum), another domain and so on? Does separate github tracker has limitations for such kind of questions?

embeddedt, 2019-02-22:
@puzrin 

@amirgon doesn't think that GitHub issues should be used for questions.

https://github.com/littlevgl/lvgl/issues/859#issuecomment-464789426
https://github.com/littlevgl/lvgl/issues/859#issuecomment-464890776

puzrin, 2019-02-22:
@embeddedt Thanks. I missed this posts.

I'd like to say, selection of place for userland questions is a search of balance between max information throughput and simplicity.

- IMO lvgl is too small for separate forum.
- As user, i don't like been pushed to register somewhere, if that's more complicated than "login via github".

My current projects are still ok with gitter (minimal possible choice). If that's not enougth, i'd try separate repo or zulip. If still not enougth - discourse.

embeddedt, 2019-02-22:
@puzrin These were my thoughts as well but @amirgon appears to be suggesting that the current setup is too disorganized and makes it hard for readers to find information.

> * As user, i don't like been pushed to register somewhere, if that's more complicated than "login via github".

I don't like it either. :smiley:

kisvegabor, 2019-02-22:
@muhammetasan I've just registered to Zulin to see how it works. It seems more like an advanced chat service than a forum. Is that correct?

@puzrin 
LittlevGL really might be small for a dedicated forum but it seems growing. And it's better to create the right platforms now. 

Here are my thoughts about it:
- On a forum, announcements can be created. E.g. when I realized that lvgl was a [top trending repository](https://github.com/littlevgl/lvgl/issues/771) on Github I opened an issue to share it with you. However, having an "issue" is quite stupid. It'd be better the open a thread in the forum.
- We can add sub-forums for Blog posts, Projects etc. ([PhpBB](https://forum.littlevgl.com/index.php) does a very good job at cleanly organize topics.)
- GitHub might be an also unfamiliar place for some people.  A forum might seem more natural to ask questions.
- I hope a forum will help to make users more active. Here, on GitHub, there are developers to answer questions. So a user can feel: "Why answer a question here? Somebody will tell it soon and probably he knows better." IMO on a user-oriented place, users are more encouraged to share their thoughts.
- It'd be good to see that there are 13 open issues which are really releavant. 
- @amirgon shared [this like](https://medium.com/@methane/why-you-must-not-ask-questions-on-github-issues-51d741d83fde). It worth have look at it.

> As user, i don't like been pushed to register somewhere, if that's more complicated than "login via GitHub".

I added a social login plugin to PhpBB. Now you can log in with your GitHub account. Please try it.

I still need to try the syntax highlight plugins suggested by @amirgon. 

Here the current features of the PhpBB forum:
- well organized topics
- easy login (Google, Facebook, GitHub, Reddit etc)
- syntax highlight (assuming it will work)
- email notifications
- mention system
- free
- self-hosted

kisvegabor, 2019-02-22:
I've fixed some issues with the social login.

puzrin, 2019-02-22:
IMO "lvgl forum" is not the place where one could return every evening to chat with friends about weather. Because lvgl is not religion, hobby or life style :).

Let's split 2 tasks:

1. stop messing tracker
2. create self-sufficient website community

May be (1) without (2) sounds impolite, but that's still technically independent  task. And it's logically correct to analyse "how to implement it with minimal efforts".

(2) without returning traffic can rely only on new users from gihtub & search. Project should be really big to keep forum "in not dead state" for such conditions.

If you still believe into success - i'd suggest reconsider 'discourse' as forum. It's more modern & familiar.

> I added a social login plugin to PhpBB. Now you can log in with your GitHub account. Please try it.

After activation by email, i expected to be on forum main page, not at littlevgl.com. And i expected to be logged in (had to click login link again)

kisvegabor, 2019-02-22:
> IMO "lvgl forum" is not the place where one could return every evening to chat with friends about weather. Because lvgl is not religion, hobby or life style :).

I'm sure there are events which should be shared with the community. Maybe my "top trending example" wasn't the best: Another example for this when LittlevGL became officially supported on ESP32.

> without returning traffic can rely only on new users from gihtub & search. Project should be really big to keep forum "in not dead state" for such conditions.

It's a key question. I'd be curious other's opinion about it.

muhammetasan, 2019-02-22:
Yes Zulip is like chat. An advanced version of Slack and free. discourse seems so similar but it is not free. @kisvegabor

amirgon, 2019-02-22:
> > without returning traffic can rely only on new users from gihtub & search. Project should be really big to keep forum "in not dead state" for such conditions.
> 
> It's a key question. I'd be curious other's opinion too about it.

@kisvegabor let's try to count.
How many issues, today, are in fact questions that belong to a forum instead of GitHub issues? How many such issues do you get every week?  
Once you have a forum, you could ask (politely) on each such GitHub issue - to move this question to the forum, and close the issue.  
This would be the **minimal** number of new forum messages you get every week.  
In such case, would you consider it "in not dead state"?

puzrin, 2019-02-22:
> I'm sure there are events which should be shared with the community. Maybe my "top trending example" wasn't the best: Another example for this when LittlevGL became officially supported on ESP32.

As user of many github projects, i'm interested in unified way to consume news. 

- look `milestones` tab for big steps
- subscrube to important threads.

As maintainer of multiple repos, i tries to minimize maintenance efforts:

- publish self-explaining demos on gh-pages
- use in-page JS instead of server-side scripts
- route "uninteresting" things to gitter (enought for my case)

> Once you have a forum, you could ask (politely) on each such GitHub issue - to move this question to the forum, and close the issue.

https://github.com/eslint/eslint/tree/master/.github - github allows to politely reduce unwanted issues. Try to start creating issue/pr there.

kisvegabor, 2019-02-24:
@amirgon Good approach. Based on the current rate of the issues (more forum related issues are open every day) I think the forum won't be dead.

@puzrin The issue template sounds a good idea as well.

I made `MarkDown` work on PhpBB. Syntax highlight is still not working with it but I'll try to solve it too.

With this, I don't see to much advantage of `Discourse` to `PhpBB` (except the more modern look). However, `PhpBB` works for free (no Virtual Server), already working, and already know (almost) everything we discussed. I'll improve its appearance a little according to the design rules of 2019.

kisvegabor, 2019-02-27:
I officially started the [Forum](https://forum.littlevgl.com/index.php) and opened GitHub which can be referenced when a Forum-related issue is opened: https://github.com/littlevgl/lvgl/issues/906

Thank you very much for all the good ideas, links, and researches!

If you are interested in being a moderator on the Forum please write a comment here.

As this discussion became quite long I renamed the topic to "LittlevGL Forum" and I close it now.

closed 2019-02-27 by kisvegabor

amirgon, 2019-02-27:
@kisvegabor It looks very nice! especially after you improved the css.
Any plans to enable the Syntax Highlight plugin?

I suggest opening a "Micropython" section on the forum, I can moderate it, if you want.

[12 more replies]