lvgl/lvgl #860: how to use lv_tick_inc with timer_create

MShahrukh0, 2019-02-18:
Hello I am using LV_INDEV_TYPE_KEYPAD type but sometime when i click button its function skipped, so i am using lv_tick_inc(5); and usleep(50); as follows
`while(1)
	{
		#if 1
		count++;
		if(!(count%50))
		{
			if(!(count%5000))
			{	
			#if 0	
			printf("time 10 loop %d ms\n",((clock()-start)*1000)/CLOCKS_PER_SEC);
			start=clock();	
			#endif
			//count = 0;
			}
			lv_tick_inc(5);
		}
		#endif
		
		lv_task_handler();
		usleep(50);
	}`
here I am using count to call lv_tick_inc after every 2500us , up to some extent my keypad is working fine but still it skip keys value sometime, in lvgl documentation it is written to use lv_tick_inc in a timer.
So does it solve my problem after using timer ? if yes then please help me how to implement c timer?

embeddedt, 2019-02-18:
`lv_tick_inc` expects the parameter to be passed in milliseconds. You should use `usleep(5*1000)` to sleep for 5 milliseconds, not 50 microseconds.

It's a good idea to use `lv_tick_inc` in a timer, but how you implement that is specific to your platform (i.e. PC simulator, STM32).

chaos4284, 2019-02-19:
You refer of coide (https://github.com/chaos4284/lvgl_designer/blob/master/libTimer/timer.c and https://github.com/chaos4284/lvgl_designer/blob/master/libTimer/main.c) If your system is based on linux.

MShahrukh0, 2019-02-19:
@chaos4284 your code is somehow similar to that concept of count which i am already using in my code, although I use your method but see no positive changes.
@embeddedt what difference it will do if i use lv_tick_inc in a timer ? I am working on linux platform, my purpose is to achieve that condition where I click any button and don't skip its function even a single time in 100 clicks.
Can you suggest me something ?

embeddedt, 2019-02-19:
@MShahrukh0 As I mentioned above, at the moment you're calling `lv_tick_inc` way too frequently (every 50 milliseconds instead of 5 milliseconds). That will cause problems in many places.

The reason for using a timer is to make sure that the tick is called every 5 milliseconds regardless of how long it takes `lv_task_handler` to execute.

MShahrukh0, 2019-02-20:
@embeddedt now I am calling lv_tick_inc every 5ms but still key pressed and released detected but function of that key get miss, may be this happening because of time fluctuations in clicking button though the fluctuation is in ms or us , that's why I need a solution to get my key function done if evdev detect the key press event, can you help?

embeddedt, 2019-02-20:
@kisvegabor Do you have any ideas here? I'm not sure what the problem is.

kisvegabor, 2019-02-21:
@embeddedt Neither am I.

So you are sure that `evdev` gets your keys but LittlevGL doesn't call the action?

> now I am calling lv_tick_inc every 5ms

Do you call `lv_tick_inc(5)` in every 5 ms, right?

MShahrukh0, 2019-02-22:
@kisvegabor yes I am calling it in every 5ms,
now somehow I achieve what I was wanted , here i am attaching my evdev.c code kindly look into that and tell me is this way good for an application?
[evdev.txt](https://github.com/littlevgl/lvgl/files/2892338/evdev.txt)

embeddedt, 2019-02-22:
@MShahrukh0 I don't think you should be dealing with objects inside your driver. The driver's job should only be to return the coordinates and press state.

MShahrukh0, 2019-02-23:
@embeddedt but that was not serving my purpose and function getting skip but now if event read detect key press than function is perform with 100% success rate.
If You can suggest me other better way to achieve the same than please help me out

embeddedt, 2019-02-23:
@MShahrukh0 Can you reproduce this issue in the PC simulator? Otherwise I don't think I can help you further.

MShahrukh0, 2019-02-25:
@embeddedt ok thank you because I am not using pc simulator and till now there is no  bug or error found in using my method.

kisvegabor, 2019-03-03:
Maybe it's related: https://github.com/littlevgl/lvgl/issues/857

---

In the past few days [LittlevGL's Forum](https://forum.littlevgl.com) has launched to discuss questions there. See this [Announcment](https://forum.littlevgl.com/viewtopic.php?f=3&t=5).

If the issue still persists please open a topic for it on the Forum.

closed 2019-03-03 by kisvegabor