lvgl/lvgl #861: Led Blink [ HardFault_Handler ]

autoprog, 2019-02-18:
Hi,
my test SSD1963, 800x480, Lv5.3.
infos:
 LV_VDB_SIZE     (42 * 1024), color 16bits,

Win_800x480 = ok
Chart_780x320 =  ( update = 100ms) = ok
Led blink = failure.....init ok, blinks x...stoped with(  HardFault_Handler)
reset processor.
led size(10,10)
lv_led_off(led1); lv_led_on(led1); or  lv_led_toggle(led1);

if case problems the memory size.
how to set memory internal address?.
my memory banks
0x00000000 = 64k
0x20000000 = 64k
0x20200000 = 128k

thanks,
Carlos.

embeddedt, 2019-02-18:
@autoprog 

> how to set memory internal address?

https://github.com/littlevgl/lvgl/blob/62195b7cc7c9c9dc643985c11f4f4de6571b530c/lv_conf_templ.h#L24

autoprog, 2019-02-18:
Hi,
not solved,
see:
#define LV_MEM_SIZE    (128U * 1024U)   
#define LV_VDB_SIZE         (128U * 1024U) 
#define LV_VDB_ADR    0x80000000  sram

problems no is not ram size.
with led blink timer 50~200ms = error.

thanks,
Carlos

embeddedt, 2019-02-19:
@autoprog Does the LED work at all? Please describe exactly what you've tried and what isn't working.

autoprog, 2019-02-19:
Hi,
I do update new Lvg 5.3 02/19/19.
yesterday led blink x, will ( HardFault_Handler).

my test  today:  led not blink (It seems to me that it burned, kkkkkk),excuse me.

see code ( in thread freertos):
   lv_init();
   ssd1963_init();
      vTaskDelay(1000);
   PRINTF("\r\n LCD inilializado ok .\r\n");
//Register the display drier in LittlevGL
		                lv_disp_drv_t disp_drv;
		                lv_disp_drv_init(&disp_drv);
		                disp_drv.disp_flush = ssd1963_flush;
		                disp_drv.disp_fill = ssd1963_fill;
		                disp_drv.disp_map = ssd1963_map;
		                lv_disp_drv_register(&disp_drv);
      //--------------------------------------------------------
         //sysmon_create();
                               static lv_style_t style_led;
		                lv_style_copy(&style_led, &lv_style_pretty_color);
		                style_led.body.radius = LV_RADIUS_CIRCLE;
		                style_led.body.main_color = LV_COLOR_MAKE(0xb5, 0x0f, 0x04);
		                style_led.body.grad_color = LV_COLOR_MAKE(0x50, 0x07, 0x02);
		                style_led.body.border.color = LV_COLOR_MAKE(0xfa, 0x0f, 0x00);
		                style_led.body.border.width = 3;
		                style_led.body.border.opa = LV_OPA_30;
		                style_led.body.shadow.color = LV_COLOR_MAKE(0xb5, 0x0f, 0x04);
		                style_led.body.shadow.width = 10;
  	                   
                               lv_obj_t * led1  = lv_led_create(lv_scr_act(), NULL);
		                lv_obj_set_style(led1, &style_led);
		                lv_obj_align(led1, NULL, LV_ALIGN_IN_BOTTOM_RIGHT, -50, -50);
		                lv_led_on(led1);
		             

print led ok, will   ( HardFault_Handler), with/without Sysmon.

                              while(1){
  		       vTaskDelay(5);
  		      lv_task_handler();

		      if(++ t > 100/5){              // sample Sysmon
		          	 t =0;
		                                                       // or	sysmon_task(NULL);

							                      	 // lv_led_toggle(led1);
							                      	//  lv_led_toggle(led2);
							  		         //  lv_led_on(led1);
                       		                PRINTF("Dbug sysmon running \r\n");

		                }
	   }
//--------------------------

thanks,
Carlos.

embeddedt, 2019-02-19:
@kisvegabor Do you have any ideas?

kisvegabor, 2019-02-20:
I tested your code in the PC simulator and it worked.

1. Which MCU do you use?
2. Can you debug the MCU? If yes, please find where the hard fault occurs. If no, use `printf`s inside the lvgl functions to narrow the issue.

autoprog, 2019-02-20:
Hi,
my MCU RT1021 (RT1020-EVK, NXP). Eclipse.
see:  led example not modifier.
[image]
othes runnig ok, Sysmon...

how to print Log in my printf  Tera term?

thanks,
Carlos.

kisvegabor, 2019-02-21:
> how to print Log in my printf Tera term?

I don't know your system but even a UART would do the job to see where the hardfault occurs.

Can you debug the MCU? It'd be much easier that way.

embeddedt, 2019-02-21:
@kisvegabor I'm pretty sure he's using a debugger; I can see a backtrace in the top right corner of his screenshot.

autoprog, 2019-02-21:
Hi,
see other test ( your canvas_2 ) simple retangle.

void lv_test_canvas_2(void)
 {
     lv_color_t bg_color = LV_COLOR_MAKE(0xec, 0xf0, 0xf1);
     lv_color_t fill_color = LV_COLOR_MAKE(0xbc, 0xa0, 0xa1);
     static lv_color_t cbuf[sizeof(lv_color_t) * 100 * 100];
     uint32_t i;
     for(i = 0; i < sizeof(cbuf) / sizeof(cbuf[0]); i++) {
         cbuf[i] = bg_color;
     }
    lv_obj_t * canvas = lv_canvas_create(lv_scr_act(), NULL);
    lv_canvas_set_buffer(canvas, cbuf, 100, 100, LV_IMG_CF_TRUE_COLOR);   here ok
     lv_point_t points3[4] = {
       {.x=10, .y=10},
       {.x=50, .y=10},
       {.x=50, .y=50},
       {.x=10, .y=50}
     };

     lv_canvas_draw_rect(canvas, points3, LV_COLOR_RED); here ok, running forever...

    lv_canvas_fill_polygon(canvas, points3, 4, LV_COLOR_RED, fill_color); <----reset system
 }
//------------------------------------------------------------
see debug:
[image]

note vars it ok value.

I use Debud printf onboard (usb cdc com) receiver pc tera Term.
PRINTF("Hello world! \r\n");

thanks,
Carlos.

kisvegabor, 2019-02-22:
The LED is a more simple object, so please debug it in the same way to sing the offending line.

autoprog, 2019-02-22:
Hi,
see test with Leds.

[image]

now use memory internal 64k.
canvas2 fill pollygon error.
Sysmon ok.

thanks.
Carlos.

seyyah, 2019-02-22:
> static lv_color_t cbuf[sizeof(lv_color_t) * 100 * 100];

@autoprog please try to run/compile and report for `lv_test_canvas_2`'s orginal version. The piece of code you share contains some changes.

https://github.com/littlevgl/lv_examples/blob/9a40c22dab372b960acf16b961c1b5b24d1c4b8a/lv_tests/lv_test_objx/lv_test_canvas/lv_test_canvas.c#L108-L114

autoprog, 2019-02-22:
Hi,
testd: static 
lv_color_t cbuf[sizeof(lv_color_t) * 100 * 100]; and 100 *80
lv_canvas_set_buffer(canvas, cbuf, 100, 80 or 100, LV_IMG_CF_TRUE_COLOR);

memory size not is, test 64k and 128k.
memory size not is.
can it other problems here.
thanks,
Carlos.

seyyah, 2019-02-22:
Do you receive an error message when you run only the original `lv_test_canvas_2` test code in your configuration?

Can you create a minimal example that I can try in the `pc_simulator`? It will help me reproduce the issue.

kisvegabor, 2019-02-24:
In the LED example please try to narrow the place of the issue with the debugger. You should find the most "atomic" line where it halts (not a function)

autoprog, 2019-02-25:
hi,
see
lv_refr_area_with_vdb();
lv_refr_area_with_vdb(const lv_area_t * area_p)
in lv_refr_area_part_vdb(area_p);
disp flush  started
active->driver.disp_flush: x1:0,y1:0, x2:799,y2:162, corlor:2147483648

see debug log: win + chat ok

[log_win_chart ok.txt](https://github.com/littlevgl/lvgl/files/2902465/log_win_chart.ok.txt)

log: led error
[LOG_led.txt](https://github.com/littlevgl/lvgl/files/2902473/LOG_led.txt)

thanks,
Carlos.

autoprog, 2019-02-27:
hi,
new test,

code: https://github.com/littlevgl/lvgl/issues/885 = ok

code for comparison:

  //-------------------------not run
/*                                                 
                                                    static lv_style_t style_led;
		                lv_style_copy(&style_led, &lv_style_pretty_color);
		                style_led.body.radius = LV_RADIUS_CIRCLE;
		                style_led.body.main_color = LV_COLOR_MAKE(0xb5, 0x0f, 0x04);
		                style_led.body.grad_color = LV_COLOR_MAKE(0x50, 0x07, 0x02);
		                style_led.body.border.color = LV_COLOR_MAKE(0xfa, 0x0f, 0x00);
		                style_led.body.border.width = 3;
		                style_led.body.border.opa = LV_OPA_30;
		                style_led.body.shadow.color = LV_COLOR_MAKE(0xb5, 0x0f, 0x04);
		                style_led.body.shadow.width = 10;

		                lv_obj_t * sensor1_led  = lv_led_create(lv_scr_act(), NULL);
		                lv_obj_set_style(sensor1_led, &style_led);
		                lv_obj_align(sensor1_led, NULL, LV_ALIGN_IN_TOP_MID, 40, 40);
		                lv_led_on(sensor1_led);

		                lv_obj_t * sensor2_led  = lv_led_create(lv_scr_act(), sensor1_led);
		                lv_obj_align(sensor2_led, sensor1_led, LV_ALIGN_OUT_BOTTOM_MID, 0, 40);
		                lv_led_set_bright(sensor2_led, 190);

		                lv_obj_t * sensor3_led  = lv_led_create(lv_scr_act(), sensor1_led);
		                lv_obj_align(sensor3_led, sensor2_led, LV_ALIGN_OUT_BOTTOM_MID, 0, 40);
		                lv_led_off(sensor3_led);

		                lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
		                lv_label_set_text(label, "LED On");
		                lv_obj_align(label, sensor1_led, LV_ALIGN_OUT_LEFT_MID, -40, 0);

		                label = lv_label_create(lv_scr_act(), NULL);
		                lv_label_set_text(label, "LED half");
		                lv_obj_align(label, sensor2_led, LV_ALIGN_OUT_LEFT_MID, -40, 0);

		                label = lv_label_create(lv_scr_act(), NULL);
		                lv_label_set_text(label, "LED Off");
		                lv_obj_align(label, sensor3_led, LV_ALIGN_OUT_LEFT_MID, -40, 0);
*/
      //------------------------- running ok

		                int led_size = 24;
		                static lv_style_t style_speedbox_label;
		                lv_style_copy(&style_speedbox_label, &lv_style_pretty);
		                style_speedbox_label.text.font = &lv_font_dejavu_20;

		                	static lv_style_t style_led;
		                	lv_style_copy(&style_led, &lv_style_pretty_color);
		                	style_led.body.radius =  12;
		                	style_led.body.main_color = LV_COLOR_MAKE(0xff, 0x3f, 0x3f);
		                	style_led.body.grad_color = LV_COLOR_MAKE(0xa0, 0x50, 0x50);
		                	style_led.body.border.color = LV_COLOR_MAKE(0xfa, 0x0f, 0x00);
		                	style_led.body.border.width = 2;

		                	lv_obj_t * sensor1_label = lv_label_create(lv_scr_act(), NULL);
		                	lv_obj_align(sensor1_label, NULL, LV_ALIGN_IN_TOP_LEFT, 0, 10);
		                	lv_label_set_text(sensor1_label, "Sensor 1");
		                        lv_label_set_align(sensor1_label, LV_LABEL_ALIGN_CENTER);
		                        lv_label_set_style(sensor1_label, &style_speedbox_label);

		                	lv_obj_t * sensor1_led = lv_led_create(lv_scr_act(), NULL);
		                	lv_obj_set_size(sensor1_led, led_size, led_size);
		                        lv_obj_align(sensor1_led, sensor1_label, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
		                	lv_obj_set_style(sensor1_led, &style_led);
[... 40 lines cut ...]
		                	lv_label_set_text(sensor3_status, "OK");
		                   lv_label_set_align(sensor3_status, LV_LABEL_ALIGN_CENTER);
		                   lv_label_set_style(sensor3_status, &style_speedbox_label);
       //--------------------------------------------------------------------

i=0;
               while(1){
                  vTaskDelay(5);
	 lv_task_handler();
           
	      if(++ t > 100/5){
         	 t =0;

	if( ++i >=3){i=0; }
                  switch(i){
	case 0: lv_led_toggle(sensor1_led); break;
	case 1: lv_led_toggle(sensor2_led); break;
	case 2: lv_led_toggle(sensor3_led); break;
	}
             }
         }
//----------------------------------------------------------------------------

thanks,
Carlos.

kisvegabor, 2019-02-28:
Form this it seesm the issue is in your driver:
```c
lv_refr_area_with_vdb();
lv_refr_area_with_vdb(const lv_area_t * area_p)
in lv_refr_area_part_vdb(area_p);
disp flush started
active->driver.disp_flush: x1:0,y1:0, x2:799,y2:162, corlor:2147483648
```

Send you `lv_conf.h` please.

Send you `lv_conf.h`

kisvegabor, 2019-02-28:
And please use shorter example codes with only the really required parts.

autoprog, 2019-02-28:
Hi Gabor,
code source test, led ok
use Eclipse(by NXP).

thanks,
Carlos.

kisvegabor, 2019-03-01:
I don't understand, what happened?

closed 2019-03-01 by autoprog