lvgl/lvgl #862: Colouring issue

sahilkhanna, 2019-02-18:
I was trying out the hello world example and was messing around withe colors. I found out that the white label looked a bit dim on a black background. So I changed the background color to white. 
In theory, I shouldn't be able to see the text but I could see a faded text (greyish). Is there a way to force complete white color on the texts?

embeddedt, 2019-02-19:
@sahilkhanna Try setting the text color in the style as well (`style.text.color`).

sahilkhanna, 2019-02-19:
@embeddedt That's how I am setting the color. So I set the color of background as -

```
    lv_style_scr.body.main_color = LV_COLOR_WHITE;
    lv_style_scr.body.grad_color = LV_COLOR_WHITE;
```
and then label color as

`     style.text.color = LV_COLOR_WHITE;`

embeddedt, 2019-02-19:
@sahilkhanna Maybe it's caused by antialiasing. Can you turn `LV_ANTIALIAS` off and see if that works?

sahilkhanna, 2019-02-19:
@embeddedt I tried that as well but same result. For reference this is how it looks - white background and white text.
[image]

embeddedt, 2019-02-19:
@sahilkhanna Can you reproduce it on the PC simulator?

sahilkhanna, 2019-02-20:
@embeddedt I haven't used the simulator yet but I'll try and post back here. Thanks for the help.

kisvegabor, 2019-03-02:
In the past few days [LittlevGL's Forum](https://forum.littlevgl.com) has launched to discuss questions there. See this [Announcment](https://forum.littlevgl.com/viewtopic.php?f=3&t=5).

If the issue still persists please open a topic for it on the Forum.

closed 2019-03-02 by kisvegabor