lvgl/lvgl #864: ST7920 and NOKIA LCD 5110 Monochome and Stm32F030C8 (64kb, 8kb RAM)

NickNifontov, 2019-02-19:
I plan to use ST7920 128x64 monochrome LCD at Stm32F030C8 CPU. Have anybody a positive/negative experience?

8kb RAM and 64kb flash will be enoough for monochrome theme at 128x64 resolution with minimum object at scene?

Also plan use a Lite version a Nokia LCD 5110 (84x84, PCD8544 driver). 

Nokia LCD 5110 -
[https://www.aliexpress.com/item/Free-shipping-1pcs-New-Module-Blue-backlight-84-48-84x84-LCD-adapter-PCB-for-Nokia/32526061324.html?spm=2114.search0604.3.2.6e95623cUMX51d&s=p&ws_ab_test=searchweb0_0,searchweb201602_6_10065_10068_319_10059_10884_317_10887_10696_321_322_10084_453_10083_454_10103_10618_10307_537_536_10902,searchweb201603_59,ppcSwitch_0&algo_expid=e7f4aac2-486a-4b88-befc-921857f9a2f5-0&algo_pvid=e7f4aac2-486a-4b88-befc-921857f9a2f5]

ST7920 - [https://www.aliexpress.com/item/Free-shipping-LCD-Board-12864-LCD12864-5V-display-With-Chinese-word-stock-with-backlight-12864-5V/32524858507.html?spm=a2g0v.10010108.1000001.7.633679c7USm51o&ws_ab_test=searchweb0_0,searchweb201602_6_10065_10068_319_10059_10884_317_10887_10696_321_322_10084_453_10083_454_10103_10618_10307_537_536_10902,searchweb201603_59,ppcSwitch_0&algo_expid=dad4230d-85a9-4e18-a175-eac9eb01639d-0&algo_pvid=dad4230d-85a9-4e18-a175-eac9eb01639d]

embeddedt, 2019-02-19:
@NickNifontov Personally, I'd advise you to get a microcontroller with at least 128 KB of Flash and maybe 16 kilobytes of RAM.

The absolute bare minimum requirements to fit LittlevGL itself, according to the README, are 64 KB of Flash and 10 KB of RAM, but you'll need more space to fit your application as well and I've never been able to get it that small.

I think @kisvegabor has tried it on a wider variety of platforms than I have so he can probably give you more exact information.

NickNifontov, 2019-02-19:
cool

i'll try stm32f103cbt6 (128kb flash, 20kb RAM)

kisvegabor, 2019-02-20:
I agree with @embeddedt. You should choose a little bit more powerful MCU to use LittlevGL.

I updated your post to make links really links. Try to edit your comment to see how it works.

closed 2019-02-20 by kisvegabor