lvgl/lvgl #867: Concept of modality

manison, 2019-02-19:
I'm trying to show a modal message box and realized that there is (probably) no concept of modality built into the lvgl at all. 

Here is the example:

```c
static lv_res_t on_btn_click(lv_obj_t *o)
{
    lv_ta_set_text(lv_obj_get_free_ptr(o), "I told you not to click me.");
    return LV_RES_OK;
}

void test_modality(void)
{
    lv_group_t *g = lv_group_create();
    lv_indev_set_group(kbd_indev, g);
    lv_group_set_wrap(g, true);

    lv_obj_t *ta = lv_ta_create(lv_scr_act(), NULL);
    lv_obj_align(ta, NULL, LV_ALIGN_CENTER, -50, 0);
    lv_group_add_obj(g, ta);

    lv_obj_t *btn = lv_btn_create(lv_scr_act(), NULL);
    lv_obj_align(btn, ta, LV_ALIGN_OUT_RIGHT_TOP, 8, 0);
    lv_obj_set_free_ptr(btn, ta);
    lv_btn_set_action(btn, LV_BTN_ACTION_CLICK, on_btn_click);
    lv_group_add_obj(g, btn);

    lv_obj_t *l = lv_label_create(btn, NULL);
    lv_label_set_text(l, "Don't click!");

    lv_obj_t *mbox = lv_mbox_create(lv_scr_act(), NULL);
    lv_mbox_set_text(mbox, "I want to be modal :-(");

    static const char * btns[] = { "OK", "" };
    lv_mbox_add_btns(mbox, btns, NULL);
}
```
[image]

You can see that in spite of the message box is showing you can use controls beneath it (type in the text area or click the button).

First I tried to resolve this like some other GUI framework do it (pseudocode):

```
bool was_disabled = is_parent_disabled(parent);
disable(parent);
lv_mbox_create(parent, NULL);
...mbox is showing...
if (!was_disabled) enable(parent);
```
But I quickly found that there is no (easy) way to disable lvgl object (prevent it from receiving input).

Then I came up with an idea, that I can put a transparent layer in between the message box and other controls that would eat all mouse and keyboard input. And this "blanket" can be even transcluent so I have a super cool visual effect indicating the controls beneath the message box are inaccessible at the moment.

```c
static lv_res_t mbox_action(lv_obj_t *btnm, const char *txt)
{
    lv_obj_t *mbox = lv_mbox_get_from_btn(btnm);
    lv_obj_t *c = lv_obj_get_parent(mbox);
    lv_obj_del(c);
    return LV_RES_INV;
}

void test_modality(void)
{
    ...
    lv_obj_t *c = lv_cont_create(lv_scr_act(), NULL);
    static lv_style_t style = { 0 };
    style.body.main_color = style.body.grad_color = LV_COLOR_WHITE;
    style.body.opa = LV_OPA_30;
    lv_cont_set_style(c, &style);
    lv_obj_set_size(c, LV_HOR_RES, LV_VER_RES);
    lv_cont_set_style(c, &style);
    lv_obj_t *mbox = lv_mbox_create(c, NULL);
    ...
    lv_mbox_set_action(mbox, mbox_action);
}
```
[image]

Last thing I'm struggling with is a proper keyboard navigation. I would like to move focus to the message box (its first button). The user should be able to use Enter (or Space) key to press the button. She should be able to navigate between buttons if there is more than one. It should not be possible to focus control beneath the message box (until the message box is closed). I must admit I don't understand groups (yet). Should I add message box to the group? Or should I create another group for the message box? Should I use `lv_group_focus_freeze` somehow?

Further discussion:
* Should lvgl support concept of modality out of the box?
* Should there be a possibility to disable input for specific object(s)?
* What control is suitable for the "blanket"? Right now I'm using container but I'm not sure if it is the right one.

embeddedt, 2019-02-19:
@kisvegabor This refers to the architecture so I think you should comment here.

kisvegabor, 2019-02-20:
You are on the right track:
- Creating a screen-sized object to absorb the clicks is a good way to solve this. Alternatively, you can use `lv_layer_top()`. It gives a layer which exists on all screen on top of the normal screens. You can create objects here, change its style etc.
- To `freeze` a group `lv_group_focus_freeze` should be used:
```c
lv_group_focus_obj(mbox);
lv_group_focus_freeze(g, true);
```

I've just tested and updated the [lv_tutorial_keyboard](https://github.com/littlevgl/lv_examples/blob/master/lv_tutorial/10_keyboard/lv_tutorial_keyboard.c) example. It shows the above described things.

You can run the example like `lv_tutorial_keyboard(NULL);` This way you will have an emulated keyboard with a few buttons but you can pass your KEYPAD input device too.

manison, 2019-02-20:
Thanks, this works fine.

closed 2019-02-20 by manison