lvgl/lvgl #868: Basic Hello World does not work?

fovea1959, 2019-02-19:
I am trying to port lvgl to run with an ili9341, connected to an ESP32 running under Mongoose OS.

I have an existing library for working the ILI9341 that is working.

My lv_conf.h just sets the LV_HOR_RES (320), LV_VER_RES (240) and LV_COLOR_DEPTH (16).

I have my program set up to display some stuff on the screen using the "native" API, set up my lv_disp_drv_t, setup a timer callback to call lv_task_handler() every 50ms, and then set a timer callback to display "hello world" one second in the figure.

I can see in my custom disp_flush where it clears the display at the beginning, but disp_flush is never called again. I can't figure out why not?

```
#include "mgos.h"
#include "mgos_ili9341.h"
#include "fonts/FreeSansBold12pt7b.h"
#include "lvgl/lvgl.h"

/*
call this from init to verify display
*/

void xxxx(void) {
  mgos_ili9341_set_fgcolor(0, 0, 0);
  mgos_ili9341_fillRect(0, 0, 320, 240);  // clear screeb
  mgos_ili9341_set_fgcolor(0, 0, 0xff);
  mgos_ili9341_drawRect(0, 0, 320, 240);  // Draw a rectangle
  mgos_ili9341_set_fgcolor(0x0ff, 0xff, 0);
  mgos_ili9341_drawRect(1, 1, 320-2, 240-2);  // Draw a rectangle
  mgos_ili9341_set_fgcolor(0xff, 0, 0);          // Red
  //mgos_ili9341_set_window(20, 30, 219, 59);      // 200x30 pixels
  mgos_ili9341_drawRoundRect(0, 0, 200, 30, 8);  // Draw a rounded rectangle
  mgos_ili9341_set_fgcolor(0, 0xff, 0);          // Green
  mgos_ili9341_set_font(&FreeSansBold12pt7b);     // Set font
  mgos_ili9341_print(5, 5, "Hello World");
}

static void my_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t* color_p)
{
    LOG(LL_INFO, ("** my_disp_flush x=%d %d y=%d %d", x1, x2, y1, y2));

    int32_t x;
    int32_t y;
    for(y = y1; y <= y2; y++) {
        for(x = x1; x <= x2; x++) {
          mgos_ili9341_set_fgcolor (color_p->red, color_p->green, color_p->blue);
          mgos_ili9341_drawPixel(x, y);
          color_p++;
        }
    }
    lv_flush_ready();
}

static void disp_timer_cb(void *arg) {
  LOG(LL_INFO, ("** lv_task_handler() in"));
  lv_task_handler();
  //LOG(LL_INFO, ("** lv_task_handler() out"));

  (void) arg;
}

static void hello_timer_cb(void *arg) {
  LOG(LL_INFO, ("** hello_timer_cb() in"));

  /*Create a Label on the currently active screen*/
  lv_obj_t * label1 =  lv_label_create(lv_scr_act(), NULL);

  /*Modify the Label's text*/
  lv_label_set_text(label1, "Hello world!");

  /* Align the Label to the center
   * NULL means align on parent (which is the screen now)
   * 0, 0 at the end means an x, y offset after alignment*/
  lv_obj_align(label1, NULL, LV_ALIGN_CENTER, 0, 0);

  LOG(LL_INFO, ("** hello_timer_cb() out"));

  (void) arg;
}

lv_disp_drv_t disp_drv;

enum mgos_app_init_result mgos_app_init(void) {
  LOG(LL_INFO, ("** init started"));
  xxxx();

  lv_init();

  lv_disp_drv_init(&disp_drv);
  disp_drv.disp_flush = my_disp_flush;
  lv_disp_drv_register(&disp_drv);

  mgos_set_timer(50, MGOS_TIMER_REPEAT, disp_timer_cb, NULL);

  mgos_set_timer(1000, MGOS_TIMER_REPEAT, hello_timer_cb, NULL);

  LOG(LL_INFO, ("** init done"));

  return MGOS_APP_INIT_SUCCESS;
}
```

Am I doing something basically wrong?

fovea1959, 2019-02-19:
...missing call to lv_tick_inc...

closed 2019-02-19 by fovea1959

embeddedt, 2019-02-19:
@fovea1959 I would also advise you to call `lv_task_handler` more frequently, maybe every 10 ms.