lvgl/lvgl #869: lv_group_remove_obj issue

Masi2017, 2019-02-19:
Hi All,
I've upgraded my lvlg from v5.1 to v5.3 . Now I have issue with lv_group_remove_obj . when I delete and object and run lv_obj_del system crashes and I figure out "break" statement in lv_group_remove_obj causes the issue . When I remove "break" statement it works fine rather than that it crashes. I am guessing this happens when you have an object and then you add another two objects . If you remove the parent of the last two added objects this happens. I should mention that I am using external keypad so I add objects to the group.

kisvegabor, 2019-02-20:
Hi,

Please share a very minimal code to reproduce the issue.

Masi2017, 2019-02-25:
Hi @kisvegabor 
I found what causes the issue. I created a btn and added to a group, then to create the second btn, I copied the previous btn ( second_btn = lv_btn_create(parent, prev_btn) ) , and then added the new btn to the group while since this had been copied from previous btn it had been already added to the group .

kisvegabor, 2019-02-25:
Oh, then it's the same issue as this: https://github.com/littlevgl/lvgl/issues/810

I'll fix it soon!
Let's continue on the other issue.

closed 2019-02-25 by kisvegabor